zibous/ha-watermeter #3: Question on NANO-CUL (mbus) 868 Mhz

to-masz, 2021-04-21:
@zibous, [README.md](https://github.com/zibous/ha-watermeter/blob/master/README.md) says
>https://www.smart-home-komponente.de/nano-cul/nano-cul-868-extra/
Please not, that you do need the NANO-CUL (mbus) 868 Mhz for `wmbusmeters` **not for** `***\*FHEM\****`!! . smart-home-komponente.net can flash the firmware for `wmbusmeters`. 

Does it mean it won't work out of the box? There is no additional info in [wmbusmeters-nano_cul.md](https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md) besides:
>nano CUL 868 mit Externer Magnetfußantenne , see: https://www.smart-home-komponente.de/. 

I would like to order the dedicated device for wmbus and I have no idea whether it is enough to order whatever "nano CUL 868" (even from Amazon) or there is a need to flash it with custom firmware to make it work with wmbusmeters and your code.

zibous, 2021-04-21:
@to-masz 
if you order the NANO-CUL from amazon it is flashed with the FHEM Firmware.

You can get the WMBUS Firmware from smart-home-komponente.de and you have to flash the wbus
firmeware but cvester has problems...

see: https://community.home-assistant.io/t/watermeter-diehl-izar-rc-i-r4-how-to-do/211129/19

I ordered the Nano-cul with the WMBUS firmeware  for wmbusmeters, so that I didn't have to flash the firmware. I use this on a RASPBERRY PI3B.

cvjensen, 2021-04-21:
I've just ordered this USB: https://shop.imst.de/wireless-modules/usb-radio-products/10/im871a-usb-wireless-m-bus-usb-adapter-868-mhz recommended from Frederik:

https://github.com/weetmuts/wmbusmeters/issues/279

So yes @zibous is right - I got a nano cul for sale. I can't recommend it :D 

No joke aside I have to test the im871a device to "divide and conquer" where my problems reside. 
Maybe it's the cul device , maybe its my raspberry or .... I pray not ... its the water meter ..

zibous, 2021-04-21:
@cvjensen 

I'm curious how the im871a works. Actually not much more expensive than the NANO CUL .. Wish you the best of luck with testing and hopefully it is not the water meter

to-masz, 2021-04-23:
Ok, so (after reading all the threads) it looks like nano-cul does not support C1, C2 mode of the WMBUS protocol. @cvjensen I would recommend you to check with smart-home-komponente if that is the case. There are slight differences between modes in communication - according to https://radiocrafts.com/technologies/wireless-m-bus-technology-overview/. But maybe it is only about adjusting something in the firmware. 

I think about being lazy and ask smart-home-komponente to flash nano-cul for me to support T1,T2 mode of WMBUS 😉 im871a looks more flexible with better driver/software support, but I cannot find it available for less than 70EUR+

closed 2021-11-05 by zibous

nobodyguy, 2022-12-06:
> I think about being lazy and ask smart-home-komponente to flash nano-cul for me to support T1,T2 mode of WMBUS wink im871a looks more flexible with better driver/software support, but I cannot find it available for less than 70EUR+
For anyone looking for an affordable and robust solution, I've designed an USB stick with iM871A module that costs around 45€. Feel free to drop me a message.
With the latest firmware, it supports combined C/T mode apart from the standalone C, T, S and R modes.