zibous/ha-watermeter #4: Error str vs int

druss67, 2021-05-25:
Hi Zibous,
thanks for your great work, I was interested to have my water flow meter visible on my domoticz, but no skills to do, you did so thanks for this.
I am near to success but I have this error about conversion or calculation:
`pi@PiPortable:~ $ sudo /opt/ha-watermeter/app.py
2021-05-25 07:23:28,524 - INFO: ✔︎ Watermeter data application start (__main__:65)
2021-05-25 07:23:28,525 - DEBUG: ✔︎ Watermeter MQTT Client 192.168.15.207 (__main__:62)
2021-05-25 07:23:28,526 - INFO: ✔︎ Watermeter connect to MQTT Client 192.168.15.207, tele/IZAR/LWT (__main__:65)
2021-05-25 07:23:28,528 - ERROR: An unhandled exception occured: '<=' not supported between instances of 'str' and 'int'. Traceback: [<FrameSummary file /opt/ha-watermeter/app.py, line 151 in <module>>] (__main__:84)
`
My hardawre is a RPI3, I use Raspbian Buster, python version is 2.7.16

I am not an expert in Python, how can I can solve this issue, please?

zibous, 2021-05-25:
@druss67 

> My hardawre is a RPI3, I use Raspbian Buster, python version is 2.7.16

The python app needs **python version 3**

```python
#!/usr/bin/env python3# -*- coding: utf-8 -*-
```

You only have 2 options:

- Upgrade Debian on Raspberry to python version 3.
- Using it's native MQTT interface Domoticz and make your own calculations app

But a fresh Debian 10 installation has both Python 2 and Python 3 installed by default. 

You can verify the specific versions of each by appending --version after typing python or python3 in the shell.
```bash
 python3 --version

## test 
 /usr/bin/python3 /opt/ha-watermeter/app.py
```
But if you haven't installed it then you can install it afterwards.

druss67, 2021-05-25:
Hello Peter,

Thanks for your quick answer.
Here below the result of your test.

pi@PiPortable:~ $ sudo /usr/bin/python3 /opt/ha-watermeter/app.py
2021-05-25 13:03:16,373 - INFO: ✔︎ Watermeter data application start
(__main__:65)
2021-05-25 13:03:16,374 - DEBUG: ✔︎ Watermeter MQTT Client 192.168.15.207
(__main__:62)
2021-05-25 13:03:16,376 - INFO: ✔︎ Watermeter connect to MQTT Client
192.168.15.207, tele/IZAR/LWT (__main__:65)
2021-05-25 13:03:16,378 - ERROR: An unhandled exception occured: '<=' not
supported between instances of 'str' and 'int'. Traceback: [<FrameSummary
file /opt/ha-watermeter/app.py, line 151 in <module>>] (__main__:65)

zibous, 2021-05-25:
@druss67 
Ok fine, the python3 is working.

But it looks like, that you have problems with the MQTT Broker Connection.
The application can not connect to the MQTT Broker with the defined User / Password ...

Check the MQTT Brocker
```bash
mosquitto -v
mosquitto version 1.5.7 starting
Using default config.
Opening ipv4 listen socket on port 1883.
````

Try to get the message for WMUSMeters
```bash
mosquitto_sub -h 192.168.15.207 -p 1883   -u 'YOURUSERNAME' -P 'YOURPASSWORD' -v -t 'tele/wasser/verbrauch'
```
Maybe you have also to change the MQTT Message Topic 'tele/wasser/verbrauch'

Or you can use the MQTT-Explorer to see the message...
http://mqtt-explorer.com

druss67, 2021-05-26:
Hello Peter,

the broker is working.
pi@PiPortable:~ $ mosquitto -v
1621964358: mosquitto version 1.5.7 starting
1621964358: Using default config.
1621964358: Opening ipv4 listen socket on port 1883.
1621964358: Error: Address already in use
pi@PiPortable:~ $ 

Thank you for the trick about MQTT-Explorer, I need to check deeply, and I try to understand how it works, because I am not familiar with.

I will do some test/check and I will come back to you if needed.

Thanks for your great support!
Best regards,

druss67

druss67, 2021-05-26:
Hi Peter,

I did some progress. Broker seems work. When I send command:
`/usr/bin/mosquitto_pub -h localhost -t domoticz/in/conso -m "{\"idx\":7,\"svalue\":\"999\"}" -u "pi" -P "raspberry"`

I see in Domoticz the value sent.

[image]

But when I use the following command I have this error syntax:
`pi@PiPortable:/opt/ha-watermeter $ sudo ./app.py
Traceback (most recent call last):
  File "./app.py", line 25, in <module>
    from lib import logger, calc
  File "/opt/ha-watermeter/lib/__init__.py", line 7, in <module>
    import lib.logger as logger
  File "/opt/ha-watermeter/lib/logger.py", line 6, in <module>
    from config import *
  File "/opt/ha-watermeter/config.py", line 36
    MQTT_HOST = 192.168.15.207  # mqtt brocker host
                         ^
SyntaxError: invalid syntax
pi@PiPortable:/opt/ha-watermeter $ `

My config file:
`# all for the mqtt brocker
MQTT_HOST = 192.168.15.207  # mqtt brocker host
MQTT_PORT = 1883  # mqtt brocker port
MQTT_APPID = 'pi'  # mqtt brocker user name
MQTT_PASSWORD = 'raspberry'  # mqtt brocker user password
MQTT_KEEPALIVE = 60  # mqtt brocker keep alive time

# all topic settings
MQTT_TOPIC = 'domoticz/in'  # listen to this wmbusmeters topic

# all application publish topics
MQTT_TOPIC_NAME = 'IZAR'
MQTT_PUBLISH_TOPIC = 'tele/' + MQTT_TOPIC_NAME + '/wasser'  # used for publishing new data
MQTT_STATE_TOPIC = 'tele/' + MQTT_TOPIC_NAME + '/wasser/status'  # mqtt state topic
MQTT_AVAILABILITY_TOPIC = 'tele/' + MQTT_TOPIC_NAME + '/LWT'  # mqtt availability topic Online | Offline
MQTT_ENABLE_LOGGING = False  # enable | disable debug logging mqtt`

zibous, 2021-05-27:
@druss67 

> I did some progress. Broker seems work. When I send command:
> `/usr/bin/mosquitto_pub -h localhost -t domoticz/in/conso -m "{\"idx\":7,\"svalue\":\"999\"}" -u "pi" -P "raspberry"`

0.k

Change the **`MQTT_HOST = '192.168.15.207'`** 
```ìni
> My config file:
> `# all for the mqtt brocker
> MQTT_HOST = '192.168.15.207' # mqtt brocker host
> MQTT_PORT = 1883 # mqtt brocker port
> MQTT_APPID = 'pi' # mqtt brocker user name
> MQTT_PASSWORD = 'raspberry' # mqtt brocker user password
```

Maybe u run the app with python2 ?
```bash
     pi@PiPortable:/opt/ha-watermeter $ sudo ./app.py
```

1. First check if wmbusmeters is working:
   WMBUSMETERS sends the recorded data as MQTT topic `tele/wasser/verbrauch`. 
   You should also be able to query this directly in domoticz, or with MQTT Explorer or with the console.

   If WMBUSMETERS works, the app.py listens to the `MQTT topic` and calculates the further data (current consumption, consumption / hour ....) and sends the results as `MQTT_PUBLISH_TOPIC` (see config).

   ## MQTT_TOPIC = 'tele/wasser/verbrauch'  # listen to this wmbusmeters topic

  ```bash 
       mosquitto_sub -h 192.168.15.207 -p 1883   -u 'pi' -P 'raspberry' -v -t 'tele/wasser/verbrauch'
       ## or
       mosquitto_sub -h localhost -p 1883   -u 'pi' -P 'raspberry' -v -t 'tele/wasser/verbrauch'
   ```

2. Try to run the python app with python3 !!:

   ```bash
     pi@PiPortable: su -
     pi@PiPortable: cd /opt/ha-watermeter
     pi@PiPortable:/opt/python3 app.py
  ```
If everything goes well, then you should see the results

zibous, 2021-05-27:
@druss67 
TIP:
Install lnav `sudo apt install lnav`and you can see the syslog messages.
```bash
 pi@PiPortable: su -
 pi@PiPortable: apt install lnav -y
 pi@PiPortable: lnav
```

[image]

My NanoCul not only sees my water clock, but also everyone in my area 👍  ....

druss67, 2021-05-27:
Hi Peter,
I have modified  the shortcut for python with this command `ln -s /usr/bin/python3 /usr/bin/python` I assume it runs with python3 but i can test with python3 command in the terminal.
I agree the problem is more related to WMBUSMETERS, I will investigate on that direction this evening.

Thanks for the tips about lnav.
Good day
Druss67

zibous, 2021-05-27:
@druss67 
I use wmbusmeters **version: 0.9.34**. It looks like that newer versions has problems to work with NANO-CUL (mbus) 868 Mhz.
So try to use the version  0.9.34:

```bash
su -
wget https://github.com/weetmuts/wmbusmeters/archive/refs/tags/0.9.34.zip
or
wget https://github.com/weetmuts/wmbusmeters/archive/refs/tags/0.9.34.tar.gz
```
see: wmbusmeters version: 0.9.34 with NANO-CUL (mbus) 868 Mhz
https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md

druss67, 2021-05-27:
Hi Peter,

Something coming UP in mind, I use a Nooelec hardware with a realtek RTL2823U. I hope it is supported, I will check after business day.

Thanks

zibous, 2021-05-27:
@druss67 
Oh.. i have installed Re4son-v7:
> Linux rb3b 4.14.93-Re4son-v7+ #1 SMP Thu Jan 24 12:44:19 AEDT 2019 armv7l 

for the realtek RTL2823U Testcase.

see: https://re4son-kernel.com/re4son-pi-kernel/

But be careful, the other applications may not support this. 

Make a Backup with `raspibackup`, then you can simply go back in case something goes wrong.
see: https://www.linux-tips-and-tricks.de/en/backup/

But the Realtek RTL2823U application is very memory and processor intensive.
see: https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md

Thanks for the coffee 🥇

druss67, 2021-05-27:
Hi Peter,

I did some test but no result.
Regarding what you have mentioned about ressources of the RTL-SDR, I prefer to stop my investigation, because I have noticed that my raspberry freeze sometimes then for my Domoticz server I don't want to have this kind of problems. So I have  ordered a nanocul and I will redo a fresh install of my raspberry for testing before to go live on production.
I will follow your howto, and I will come back to you if needed.
Thanks for your great support Peter!

Druss67

zibous, 2021-05-27:
@druss67 

> Regarding what you have mentioned about ressources of the RTL-SDR, I prefer to stop my investigation, because I have noticed that my raspberry freeze sometimes then for my Domoticz server I don't want to have this kind of problems.
Yes, the RTL-SDR need a lot of memory and cpu. This is why i switched to the NanoCUL.

Be careful if you order the NanoCul, be shure that the right firmware (wmbusmeter aka wmbus) is flashed there,
otherwise you have to reflash the NanoCul.  In my case, I explicitly ordered the NanoCul from the supplier (https://www.smart-home-komponente.de/nano-cul/nano-cul-868-extern/) so that it works with WMBUSMETERS !

For me the nanoCul works with wmbusmeters version: 0.9.34.

The alternative would be a second simple Raspberry, which is only used as a watermeter data receiver, mqtt brocker ... 
You can always use it ...

druss67, 2021-05-27:
I bought from eBay, at the shop you mentioned. It seems identical as yours.
So, I will give you feedback after installation.

Yes, probably I will keep a second raspberry for this activity.

I wish you a good evening Peter.

zibous, 2021-06-02:
@druss67 
Hi,
Raspberry Zero - wmbusmeters with rtl-sdr is working for me.

see: https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-with-rtl-sdr.md

druss67, 2021-06-02:
Hi Peter,

Nice, and as usual well detailed how to ,👍
I am still waiting my nanocul, hope I Will receive it for this weekend.
First step will be to install on my pi3, and after perhaps I will test on a nano pi 😉

Good day,
Druss67

druss67, 2021-06-08:
Hi Peter,

Just to inform you that I received my nanocul and raspberry Zero.
As you mentioned I had to update firmware, but everything went well.

I install following this howto [https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md](url)

here below the status. It eems working correctly. I stop for today. I restart my test tomorrow. I will keep you in touch.

`pi@PiZero:~ $ wmbusmeters --listento=t1 --debug --logtelegrams /dev/ttyUSB0 diehl izar 73203568 NOKEY
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) number of meters: 1
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary "3F2028F8F8787A20697320756E6B6E6F776E2920557365206F6E65206F662042206220432065204620472069204B206C204D20522054207420562057205820780D0A"
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serial) registered regular callback 0 DEVICE_CUL every 60 seconds
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(config) meter izar link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "diehl" "izar" "73203568" not-encrypted
(wmbus) set timeout DEVICE_CUL to "0" with expected activity ""
(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: none
(serial) waiting for stop
`

druss67, 2021-06-13:
Hello Peter,

I come back with Pi zero and Nanocul.
Following the test you did, I was able to have the result than you.
`i@PiZero:~ $ lsusb
Bus 001 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@PiZero:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Jun  9 20:41 usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -> ../../ttyUSB0
pi@PiZero:~ $ `
It seems nano cul is not recognize.
It seems I have succeed to update firmware because in Hterm upload went well.
I will continue to investigate, but if you have an idea, you will be welcomed...
[image]

zibous, 2021-06-13:
@druss67 
Sorry that I’m only getting in touch now, but I have received a new heating system and am installing sensors, ESP32 EMS system ...

Are you sure it is the correct firmware?
I don't have Windows and so I can't use HTERM, but I get a different message on
   `ls -l / dev / serial / by-id`
```bash
root@rb3b:~# ls -l /dev/serial/by-id
lrwxrwxrwx 1 root root 13 Mai 15 14:17 usb-SHK_NANO_CUL_868-if00-port0 -> ../../ttyUSB0
```

druss67, 2021-06-16:
Hello Peter,
no problem, I have also other topic to work on. I am going to add my brand new solar panels to Domoticz.
Going back to nanocul, I have uploaded the firmware using Xloader in a virtual W10 from my Manjaro. Only way for me to access to a Windows machine.
Hterm is just a way to check, and it works from linux.
This morning I did a try with a firmware coming from here `https://github.com/smarthomeagentur/culfw1/releases` .
No success, here the results:
`pi@PiZero:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Jun 13 18:09 usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -> ../../ttyUSB0
pi@PiZero:~ $ wmbusmeters --listento=t1 --debug --logtelegrams /dev/ttyUSB0 diehl izar 73203568 NOKEY
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) number of meters: 1
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
No wmbus device found! Exiting!
pi@PiZero:~ $ 
`
I suppose the upload ran uncorrectly, I will check deeply later on...

zibous, 2021-06-16:
@druss67 

Do you use 'CUL_V3_MBUS.hex' ?

Which version wmbusmeters ?  I use 0.9.34. I do not upgrade to the new version, because i am not shure 
that the new version is working.

`received binary "" (im871a) are you there?`... this is not o.k.

druss67, 2021-07-03:
Hello Peter,

I am back with a brand new nanocul with mbus firmware and a pizero. I want to restart from zero, so which howto must I use? This one will be okay? [https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-with-rtl-sdr.md]

zibous, 2021-07-03:
@druss67 
wmbusmeters-with-rtl-sdr.md - Turns your **Realtek RTL2832 based DVB dongle** into a SDR receiver.
For **nanocul** use: https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md

druss67, 2021-07-04:
Hi Peter,
I don't understand why I haven't the same results as yours, with the same hardware, see below...
`
pi@raspberrypi:~ $ lsusb
Bus 001 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Jul  4 08:47 usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -> ../../ttyUSB0
pi@raspberrypi:~ $ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Jul  4 08:47 /dev/ttyUSB0
pi@raspberrypi:~ $ dmesg | grep ttyUSB0
[   19.573852] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0
pi@raspberrypi:~ $ wmbusmeters --listento=t1 /dev/ttyUSB0
Wmbus devices found but no meters supplied. You must supply which link modes to listen to. Eg. auto:c1
`

zibous, 2021-07-04:
@druss67 

Do you use the wmbusmeters  version: 0.9.34 ?

Try:  
```bash
service wmbusmeters stop
wmbusmeters --listento=t1 --debug --verbose /dev/ttyUSB0
````

In my case:
```
(wmbusmeters) version: 0.9.34
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary "3F2028F8F8787A20697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046206620472069204B206C204D20522054207420562057205820780D0A"
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: t1
(serial) waiting for stop
(serial) received binary "6231393434413531313738303734333433343431383439314341323231313830303133334344373632463539"
(cul) checkCULFrame "b1944A511780743434418491CA2211800133CD762F59"
(cul) no eol found yet, partial frame
(serial) received binary "43354435423731413646424444374532450D0A"
(cul) checkCULFrame "b1944A511780743434418491CA2211800133CD762F59C5D5B71A6FBDD7E2E<0D><0A>"
(wmbus) trimming frame A "1944A511780743434418491CA2211800133CD762F59C5D5B71A6FBDD7E2E"
(wmbus) ff a dll crc 0-9 491c ok
(wmbus) ff a dll crc mid 12-27 7e2e ok
(wmbus) trimmed 4 crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed  frame A "1944A511780743434418A2211800133CD762F59C5D5B71A6FBDD"
(cul) received full T1 frame
(wmbus) parseDLL @0 26
(wmbus) parseELL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) DLL L=19 C=44 (from meter SND_NR) M=11a5 (DME) A=43430778 VER=44 TYPE=18 (Pressure meter)
(telegram) TPL CI=a2
Received telegram from: 43430778
          manufacturer: (DME) DIEHL Metering, Germany
           device type: Pressure meter
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: a511 dll-mfct (DME)
(wmbus) 04: 78074343 dll-id (43430778)
(wmbus) 08: 44 dll-version
(wmbus) 09: 18 dll-type (Pressure meter)
(wmbus) 0a: a2 tpl-ci-field (Mfct specific)
(wmbus) "telegram=||1944A511780743434418A2211800133CD762F59C5D5B71A6FBDD|+6"
... more lines
```

druss67, 2021-07-04:
Yes, I have downloaded manually.

the result, strange no version shown.

`pi@raspberrypi:~ $ wmbusmeters --listento=t1 --debug --verbose /dev/ttyUSB0
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary "3F2028F8787A20697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046206620472069204B206C204D20522054207420562057205820780D0A"
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: t1
(serial) waiting for stop
^C(serial) stopping manager
(serial) event loop stopped!
(serial) closing 1 devices
(serial) no devices working
(serial) no devices working
(serialtty) closed /dev/ttyUSB0
`

zibous, 2021-07-04:
@druss67 

Try to execute the wmbusmeters as sudo, because i think you run the test as pi user.

```bash
pi@raspberrypi:~ su - 
Password: *****
wmbusmeters --listento=t1 --debug --verbose /dev/ttyUSB0
```

druss67, 2021-07-04:
still no version but seems better...

`pi@raspberrypi:~ $ sudo wmbusmeters --listento=t1 --debug --verbose /dev/ttyUSB0
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary "3F2028F8F8787A20697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046206620472069204B206C204D20522054207420562057205820780D0A"
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: t1
(serial) waiting for stop
(serial) received binary "62314534344135313136383335323037"
(cul) checkCULFrame "b1E44A5116835207"
(cul) no eol found yet, partial frame
(serial) received binary "33374230373230324337414133303031303035323234393032383538313934414337374631343239344445394633463641344444433739413642310D0A"
(cul) checkCULFrame "b1E44A511683520737B07202C7AA3001005224902858194AC77F14294DE9F3F6A4DDC79A6B1<0D><0A>"
(wmbus) trimming frame A "1E44A511683520737B07202C7AA3001005224902858194AC77F14294DE9F3F6A4DDC79A6B1"
(wmbus) ff a dll crc 0-9 202c ok
(wmbus) ff a dll crc mid 12-27 de9f ok
(wmbus) ff a dll crc final 30-34 a6b1 ok
(wmbus) trimmed 6 crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed  frame A "1E44A511683520737B077AA3001005224902858194AC77F142943F6A4DDC79"
(cul) received full T1 frame
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
Received telegram from: 73203568
          manufacturer: (DME) DIEHL Metering, Germany
           device type: Water meter
(wmbus) 00: 1e length (30 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: a511 dll-mfct (DME)
(wmbus) 04: 68352073 dll-id (73203568)
(wmbus) 08: 7b dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: a3 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
(wmbus) "telegram=||1E44A511683520737B077AA3001005224902858194AC77F142943F6A4DDC79|+43"
`

zibous, 2021-07-05:
@druss67 
Yes, it looks like that you have not used any time the root user.
Build wmbusmeters , add services .... as root user otherwise you get problems with permissions.

https://linuxize.com/post/su-command-in-linux/

Is this your watermeter `Received telegram from: 73203568` ??

```bash
wmbusmeters --listento=t1 --debug --logtelegrams /dev/ttyUSB0 diehl izar 73203568 NOKEY
```

druss67, 2021-07-05:
Hi Peter,

Yes, I will recreate a clean installation with su.
Yes, 73203568 is my watermeter.
Thanks for your great support Peter!

druss67, 2021-07-06:
Hello Peter,

I receive data but not redeable, so what should I do? Is it due to the fact that I haven't the key to decrypt, so what can I do?
`pi@raspberrypi:~ $ wmbusmeters --listento=t1 --debug --logtelegrams /dev/ttyUSB0 diehl izar 73203568 NOKEY
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) number of meters: 1
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary "3F2028F8F8787A20697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046206620472069204B206C204D20522054207420562057205820780D0A"
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) meter izar link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "diehl" "izar" "73203568" not-encrypted
(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: t1
(serial) waiting for stop
(serial) received binary "6231453434413531313638333532303733374230373230324337413538303031303035364535343735354637453839313041"
(cul) checkCULFrame "b1E44A511683520737B07202C7A580010056E54755F7E8910A"
(cul) no eol found yet, partial frame
(serial) received binary "323242304337443435433345303836454235463042314442430D0A"
(cul) checkCULFrame "b1E44A511683520737B07202C7A580010056E54755F7E8910A22B0C7D45C3E086EB5F0B1DBC<0D><0A>"
(wmbus) trimming frame A "1E44A511683520737B07202C7A580010056E54755F7E8910A22B0C7D45C3E086EB5F0B1DBC"
(wmbus) ff a dll crc 0-9 202c ok
(wmbus) ff a dll crc mid 12-27 45c3 ok
(wmbus) ff a dll crc final 30-34 1dbc ok
(wmbus) trimmed 6 crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed  frame A "1E44A511683520737B077A580010056E54755F7E8910A22B0C7DE086EB5F0B"
(cul) received full T1 frame
(wmbus) parseDLL @0 31
(meter) diehl: for me? 73203568
(meter) diehl: probably not for me since received media 0x07 does not match
(meter) diehl: yes for me
(meter) diehl 73203568 "1E44A511683520737B077A580010056E54755F7E8910A22B0C7DE086EB5F0B"
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(wmbus) decrypted content failed check, did you use the correct decryption key? Ignoring telegram.
(wmbus) telegram ignored by all configured meters!
`

zibous, 2021-07-06:
Try
```bash
pi@raspberrypi:~ su -
Password: *****
wmbusmeters --listento=t1 --debug --logtelegrams /dev/ttyUSB0
```

druss67, 2021-07-07:
I don't know who encrypt the communication I receive. In fact it seems there is no way for me to have it decrypted.
So there is no more interest to maintain a Pi to read something I can't read. :-(
 

`pi@raspberrypi:~ $ sudo wmbusmeters --listento=t1 --debug --logtelegrams /dev/ttyUSB0

(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(detect) "/dev/ttyUSB0" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyUSB0) sent "FF0B00F4"
(serial) received binary "000A"
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial) received binary ""
(im871a) are you there?
(serial /dev/ttyUSB0) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(serial /dev/ttyUSB0) sent "2D0D0A"
(serial) received binary "3F2028F8F8787A20697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046206620472069204B206C204D20522054207420562057205820780D0A"
(cul) are you there?
(serial /dev/ttyUSB0) sent "560A0D"
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
CUL answered: V 1.67 nanoCUL868
(serialtty) closed /dev/ttyUSB0
(cul) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5830310A0D"
(config) listen to link modes: t1
(serial) waiting for stop
(serial) received binary "6231453434413531313638333532303733374230373230"
(cul) checkCULFrame "b1E44A511683520737B0720"
(cul) no eol found yet, partial frame
(serial) received binary "324337414238303031303035443043354239343139433935374139333332423844433635424331423138343146364544303444360D0A"
(cul) checkCULFrame "b1E44A511683520737B07202C7AB8001005D0C5B9419C957A9332B8DC65BC1B1841F6ED04D6<0D><0A>"
(wmbus) trimming frame A "1E44A511683520737B07202C7AB8001005D0C5B9419C957A9332B8DC65BC1B1841F6ED04D6"
(wmbus) ff a dll crc 0-9 202c ok
(wmbus) ff a dll crc mid 12-27 65bc ok
(wmbus) ff a dll crc final 30-34 04d6 ok
(wmbus) trimmed 6 crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed  frame A "1E44A511683520737B077AB8001005D0C5B9419C957A9332B8DC1B1841F6ED"
(cul) received full T1 frame
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
Received telegram from: 73203568
          manufacturer: (DME) DIEHL Metering, Germany
           device type: Water meter
(wmbus) 00: 1e length (30 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: a511 dll-mfct (DME)
(wmbus) 04: 68352073 dll-id (73203568)
(wmbus) 08: 7b dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
[... 45 lines cut ...]
(wmbus) ff a dll crc mid 30-45 4d52 ok
(wmbus) ff a dll crc mid 48-63 1027 ok
(wmbus) ff a dll crc final 66-70 725b ok
(wmbus) trimmed 10 crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed  frame A "3E44A511683520737B077AB80030055D89DE0900AED3882C573E2D686C1588B86DE6CB1AC59B2516880089D97B7E81D964C30A2894394FE2E4987126719A21"
(cul) received full T1 frame
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
Received telegram from: 73203568
          manufacturer: (DME) DIEHL Metering, Germany
           device type: Water meter
(wmbus) 00: 3e length (62 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: a511 dll-mfct (DME)
(wmbus) 04: 68352073 dll-id (73203568)
(wmbus) 08: 7b dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: b8 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 3005 tpl-cfg 0530 (AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
(wmbus) "telegram=||3E44A511683520737B077AB80030055D89DE0900AED3882C573E2D686C1588B86DE6CB1AC59B2516880089D97B7E81D964C30A2894394FE2E4987126719A21|+74"
`

druss67, 2021-07-07:
Anyway, thanks for the support you gave me, Peter!
greatly appreciated.

zibous, 2021-07-07:
@druss67 
it looks like that the telegram is o.k
Try to enable MQTT and check the payload.

druss67, 2021-07-07:
Command doesn't work. 

`pi@raspberrypi:~ $ sudo wmbusmeters --format=json auto:t1 MyWater izar 73203568 NOKEY
You cannot have a suffix appended to auto.
`
MQTT enabled but nothing on the payload.

zibous, 2021-07-08:
The telegram from wmbusmeters looks good to me. I think it's the configuration and to determine this you have to be more systematic:

1. Run all as root: su-
2. Check wmbusmeters - look ok
3. Is MQTT working: Testing send and receive data
     https://randomnerdtutorials.com/testing-mosquitto-broker-and-client-on-raspbbery-pi/
4. Check system configuration
    ```ini
    /etc/wmbusmeters.conf
    /etc/wmbusmeters.d/diehl
    ```
5. Check Installation and Service is working
6. Start service and check MQTT Payload

Check System configuration
----------------------------------
```ini
/etc/wmbusmeters.conf 
/usr/bin/wmbusmeters 
/usr/sbin/wmbusmetersd 
/etc/systemd/system/wmbusmeters.service 
/etc/logrotate.d/wmbusmeters

directories:
/etc/wmbusmeters.d 
/var/log/wmbusmeters/meter_readings
```

Issues see: https://github.com/weetmuts/wmbusmeters/issues

druss67, 2021-07-08:
All okay except in system configuration, /etc/systemd/system/wmbusmeters.service is not present.

if I check after a restart I have wmbusmeters running.

`pi@raspberrypi:~ $ ps aux | grep wmbusmeters
pi         885  0.0  0.4   7332  2084 pts/0    S+   17:50   0:00 grep --color=auto wmbusmeters
`

closed 2021-07-10 by druss67