zibous/ha-watermeter #6: ERROR: Error on calculation (lib.calculator:71)

cvjensen, 2021-12-16:
Hi again zibous , 

Remember me ? (Some time ago I had troubles collecting data from my multical 21's wireless mbus)
Fredrik (wmbusmeters creator) never found out why I couldn't get the data, so I ended up buying my own Multical 21.
The multical 21 is not installed yet because I will make sure I'm actually able to get the data into HA before I do the installation.

When I start ha-watermeter I'm getting the following output in console:

**2021-12-16 10:04:16,404 - ERROR: Error on calculation (lib.calculator:71)**

```
pi@raspberrypi:/opt/ha-watermeter $ sudo python3 app.py
2021-12-16 10:04:05,691 - INFO: ✔︎ Watermeter data application start (__main__:65)
2021-12-16 10:04:05,692 - DEBUG: ✔︎ Watermeter MQTT Client localhost (__main__:62)
2021-12-16 10:04:05,695 - INFO: ✔︎ Watermeter connect to MQTT Client localhost, tele/wasser/connection (__main__:65)
2021-12-16 10:04:05,703 - DEBUG: ✔︎ localhost connected OK, subscribe to topic wmbusmeters/77634443 (__main__:62)
2021-12-16 10:04:05,707 - DEBUG: ✔︎ Message Id 1 published. (__main__:62)
2021-12-16 10:04:16,388 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-16 10:04:16,390 - INFO: ✔︎ MQTT Publish state topic tele/wasser/wasser/status (__main__:65)
2021-12-16 10:04:16,391 - DEBUG: Init calculator (lib.calculator:62)
2021-12-16 10:04:16,392 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-16 10:04:16,393 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-16 10:04:16,394 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-16 10:04:16,395 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-16 10:04:16,404 - ERROR: Error on calculation (lib.calculator:71)
2021-12-16 10:04:16,405 - DEBUG: ✔︎ Message Id 3 published. (__main__:62)
2021-12-16 10:04:47,532 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-16 10:04:47,534 - INFO: ✔︎ MQTT Publish state topic tele/wasser/wasser/status (__main__:65)
2021-12-16 10:04:47,535 - DEBUG: Init calculator (lib.calculator:62)
2021-12-16 10:04:47,536 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-16 10:04:47,536 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-16 10:04:47,537 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-16 10:04:47,538 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-16 10:04:47,538 - ERROR: Error on calculation (lib.calculator:71)
```

Wmbusmeters is running and its publishing the data like this:
```
{
  "media": "cold water",
  "meter": "multical21",
  "name": "Multical21",
  "id": "77634443",
  "total_m3": 0.094,
  "target_m3": 0.094,
  "max_flow_m3h": 0,
  "flow_temperature_c": 127,
  "external_temperature_c": 4,
  "current_status": "DRY",
  "time_dry": "22-31 days",
  "time_reversed": "",
  "time_leaking": "",
  "time_bursting": "",
  "timestamp": "2021-12-16T09:11:51Z",
  "device": "im871a[00102807]",
  "rssi_dbm": -80
}
```

config.py looks like this: 
```
pi@raspberrypi:/opt/ha-watermeter $ diff config.py config.py.dist | grep -vi password
22,23c22,23
< SMARTMETER_ID = '77634443'  # internal used smartmeter identification
< SMARTMETER_NAME = 'Multical21'  # Name of the smartmeter
---
> SMARTMETER_ID = '1234556'  # internal used smartmeter identification
> SMARTMETER_NAME = 'Wasserzähler Haus'  # Name of the smartmeter
38,39c38,39
< MQTT_APPID = 'mqtt'  # mqtt brocker user name
---
> MQTT_APPID = 'pi'  # mqtt brocker user name
44c44
< MQTT_TOPIC = 'wmbusmeters/77634443'  # listen to this wmbusmeters topic
---
> MQTT_TOPIC = 'tele/wasser/verbrauch'  # listen to this wmbusmeters topic
47c47
< MQTT_TOPIC_NAME = 'multical21'
---
> MQTT_TOPIC_NAME = 'IZAR'
```

[image]

Any clue ?

zibous, 2021-12-16:
@cvjensen 
```
2021-12-16 10:04:47,536 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-16 10:04:47,537 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
```

I think you have problems with permissions, because the previous are not present.

The error is on:
```
            # the last periode data
            self.cd['last_total']['month'] = lastmonth_dts.strftime(DATEFORMAT_MONTH)
            self.cd['last_total']['month_m3'] = float(self.payload['last_month_total_m3'])
```

try to run the python script as root or set the permission for user `pi`

```
pi@raspberrypi: su -
password:

root@raspberrypi: cd /opt/ha-watermeter
root@raspberrypi:  python3 app.py

```

cvjensen, 2021-12-16:
Just tried as root and I still get the error:

```
pi@raspberrypi:/opt/ha-watermeter $ sudo -i
root@raspberrypi:~# cd /opt/ha-watermeter/
root@raspberrypi:/opt/ha-watermeter# ls
app.py     config.py.dist  docs         lib      logs         README.md         service.template
config.py  data            __init__.py  LICENSE  __pycache__  requirements.txt  tests
root@raspberrypi:/opt/ha-watermeter# python3 app.py
2021-12-16 10:54:34,791 - INFO: ✔︎ Watermeter data application start (__main__:65)
2021-12-16 10:54:34,792 - DEBUG: ✔︎ Watermeter MQTT Client localhost (__main__:62)
2021-12-16 10:54:34,793 - INFO: ✔︎ Watermeter connect to MQTT Client localhost, tele/multical21/LWT (__main__:65)
2021-12-16 10:54:34,803 - DEBUG: ✔︎ localhost connected OK, subscribe to topic wmbusmeters/77634443 (__main__:62)
2021-12-16 10:54:34,810 - DEBUG: ✔︎ Message Id 1 published. (__main__:62)
2021-12-16 10:57:55,369 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-16 10:57:55,369 - INFO: ✔︎ MQTT Publish state topic tele/multical21/wasser/status (__main__:65)
2021-12-16 10:57:55,370 - DEBUG: Init calculator (lib.calculator:62)
2021-12-16 10:57:55,370 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-16 10:57:55,371 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-16 10:57:55,371 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-16 10:57:55,371 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-16 10:57:55,376 - ERROR: Error on calculation (lib.calculator:71)
2021-12-16 10:57:55,377 - DEBUG: ✔︎ Message Id 3 published. (__main__:62)
2021-12-16 10:58:11,237 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-16 10:58:11,238 - INFO: ✔︎ MQTT Publish state topic tele/multical21/wasser/status (__main__:65)
2021-12-16 10:58:11,238 - DEBUG: Init calculator (lib.calculator:62)
2021-12-16 10:58:11,239 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-16 10:58:11,239 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-16 10:58:11,239 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-16 10:58:11,240 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-16 10:58:11,240 - ERROR: Error on calculation (lib.calculator:71)
2021-12-16 10:58:11,241 - DEBUG: ✔︎ Message Id 4 published. (__main__:62)

```

zibous, 2021-12-16:
@cvjensen 
Can you see the file `/opt/ha-watermeter/data/77634443.json` and do this have content ?

cvjensen, 2021-12-16:
No that file does not exist . 

```
pi@raspberrypi:~ $ ls -l  /opt/ha-watermeter/data/
totalt 16
-rw-r--r-- 1 root root 3493 dec 15 18:24 43430778-day-report.csv
-rw-r--r-- 1 root root  900 dec 15 18:24 43430778.json
-rw-r--r-- 1 root root 3060 dec 15 18:24 43430778-report.csv
-rw-r--r-- 1 root root  570 dec 15 18:24 default_data.json

```

So I gues it cant load the 'payload' because the file does not exist ?

zibous, 2021-12-16:
@cvjensen 
> So I gues it cant load the 'payload' because the file does not exist ?

No, the `payload` are the values from the mqtt topic. 

[image]

You can try to comment out the missing payload values:
```
line 59:  # lastmonth_dts = datetime.strptime(self.payload['last_month_measure_date'], DATEFORMAT_DAY)

line 70:  # self.cd['last_total']['month'] = lastmonth_dts.strftime(DATEFORMAT_MONTH)
line 71:  # self.cd['last_total']['month_m3'] = float(self.payload['last_month_total_m3'])
             # current / prev m3
line 74:    if(self.cd['last_total']['month_m3'] > 0.00):
line 75:        self.cd['ratio'] = round(float(meterM3 / self.cd['last_total']['month_m3']), 2)
```

cvjensen, 2021-12-16:
I dont know whats going on ...

Both are commented out.

```
root@raspberrypi:/opt/ha-watermeter# cat -n lib/calculator.py | grep 70
    70              # self.cd['last_total']['month'] = lastmonth_dts.strftime(DATEFORMAT_MONTH)

root@raspberrypi:/opt/ha-watermeter# cat -n lib/calculator.py | grep 71
    71              # self.cd['last_total']['month_m3'] = float(self.payload['last_month_total_m3'])

```

still...

```
root@raspberrypi:/opt/ha-watermeter# python3 app.py
2021-12-16 16:26:20,169 - INFO: ✔︎ Watermeter data application start (__main__:65)
2021-12-16 16:26:20,170 - DEBUG: ✔︎ Watermeter MQTT Client localhost (__main__:62)
2021-12-16 16:26:20,171 - INFO: ✔︎ Watermeter connect to MQTT Client localhost, tele/multical21/LWT (__main__:65)
2021-12-16 16:26:20,197 - DEBUG: ✔︎ localhost connected OK, subscribe to topic wmbusmeters/77634443 (__main__:62)
2021-12-16 16:26:20,201 - DEBUG: ✔︎ Message Id 1 published. (__main__:62)
2021-12-16 16:28:51,038 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-16 16:28:51,039 - INFO: ✔︎ MQTT Publish state topic tele/multical21/wasser/status (__main__:65)
2021-12-16 16:28:51,040 - DEBUG: Init calculator (lib.calculator:62)
2021-12-16 16:28:51,040 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-16 16:28:51,041 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-16 16:28:51,042 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-16 16:28:51,042 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-16 16:28:51,049 - ERROR: Error on calculation (lib.calculator:71)
2021-12-16 16:28:51,054 - DEBUG: ✔︎ Message Id 3 published. (__main__:62)

```

zibous, 2021-12-16:
@cvjensen 

o.k try to get more informations about the error:

```
line 165:        except BaseException as e:
line 166:            log.error(f"Error on calculation: {str(e)}")
line 167:            return False
```

i think the problem is the timestamp.

If so, change the `"timestamp": now.strftime(DATEFORMAT_CURRENT),` to `"timestamp": now.strftime(DATEFORMAT_TIMESTAMP)` (config.py):

```json
# data model smartmeter
WATERMETER_DATA = {
    "name": SMARTMETER_NAME,
    "device": "watermeter",
    "deviceid": SMARTMETER_ID,
    "date": "",
    "time": "",
    "total_m3": 0.00,
    "ratio": 0.00,
    "m3": {
        "current": 0.000,
        "hour": 0.000,
        "day": 0.000,
        "month": 0.000,
        "year": 0.000
    },
    "liter": {
        "current": 0.000,
        "hour": 0.000,
        "day": 0.000,
        "month": 0.000,
        "year": 0.000
    },
    "last_total": {
        "hour": "2020-14-09",
        "hour_m3": 0.000,
        "day": "2020-07-14",
        "day_m3": 0.000,
        "month": "2020-07",
        "month_m3": 0.000,
        "year": "2019-12-31",
        "year_m3": 117.730
    },
    "alarm": "No alarm",
    "last_alarm": "No alarm",
    "periode": now.strftime(DATEFORMAT_DAY),
    "month": now.strftime(DATEFORMAT_MONTH),
    "year": now.strftime(DATEFORMAT_YEAR),
    "timestamp": now.strftime(DATEFORMAT_TIMESTAMP),
    "last_update": now.strftime(DATEFORMAT_CURRENT),
    "elapsed_time": 0,
    "icon": "mdi:billboard",
    "unit_of_measurement": u"\u33A5",
    "version": APPS_VERSION,
    "attribution": "Data provided by Peter Siebler"
}
```

cvjensen, 2021-12-17:
I've tried to change what you said . Still getting the error:

```
pi@raspberrypi:/opt/ha-watermeter $ python3 app.py
2021-12-17 17:37:02,120 - INFO: ✔︎ Watermeter data application start (__main__:65)
2021-12-17 17:37:02,121 - DEBUG: ✔︎ Watermeter MQTT Client localhost (__main__:62)
2021-12-17 17:37:02,122 - INFO: ✔︎ Watermeter connect to MQTT Client localhost, tele/multical21/LWT (__main__:65)
2021-12-17 17:37:02,136 - DEBUG: ✔︎ localhost connected OK, subscribe to topic wmbusmeters/77634443 (__main__:62)
2021-12-17 17:37:02,144 - DEBUG: ✔︎ Message Id 1 published. (__main__:62)
2021-12-17 17:37:11,623 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-17 17:37:11,623 - INFO: ✔︎ MQTT Publish state topic tele/multical21/wasser/status (__main__:65)
2021-12-17 17:37:11,624 - DEBUG: Init calculator (lib.calculator:62)
2021-12-17 17:37:11,625 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-17 17:37:11,625 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-17 17:37:11,625 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-17 17:37:11,625 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-17 17:37:11,631 - ERROR: Error on calculation (lib.calculator:71)

```

Why is it that I'm still getting the error ":71" when I try to comment out that line ? I dont get it ..

zibous, 2021-12-17:
@cvjensen 
You can try the new version.

- config.py.dist
- lib/calculator.py

Checks the payload attributes and shows more info on logging.

cvjensen, 2021-12-18:
So now we have this:

```
pi@raspberrypi:/opt/ha-watermeter $ python3 app.py
2021-12-18 10:31:55,713 - INFO: ✔︎ Watermeter data application start (__main__:65)
2021-12-18 10:31:55,713 - DEBUG: ✔︎ Watermeter MQTT Client localhost (__main__:62)
2021-12-18 10:31:55,715 - INFO: ✔︎ Watermeter connect to MQTT Client localhost, tele/multical21/LWT (__main__:65)
2021-12-18 10:31:55,726 - DEBUG: ✔︎ localhost connected OK, subscribe to topic wmbusmeters/77634443 (__main__:62)
2021-12-18 10:31:55,729 - DEBUG: ✔︎ Message Id 1 published. (__main__:62)
2021-12-18 10:32:05,321 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-18 10:32:05,322 - INFO: ✔︎ MQTT Publish state topic tele/multical21/wasser/status (__main__:65)
2021-12-18 10:32:05,324 - DEBUG: Init calculator (lib.calculator:62)
2021-12-18 10:32:05,325 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-18 10:32:05,326 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-18 10:32:05,327 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-18 10:32:05,328 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-18 10:32:05,347 - DEBUG: Missing key last_month_measure_date - used default value 2021-12-18T10:32:05Z (lib.calculator:62)
2021-12-18 10:32:05,350 - ERROR: Error on calculation: lib.calculator,  line: 79, message: unconverted data remains: T10:32:05Z (lib.calculator:71)
```

Just fyi:

```
pi@raspberrypi:/opt/ha-watermeter $ ls data/
43430778-day-report.csv  43430778.json  43430778-report.csv  default_data.json
```

zibous, 2021-12-18:
@cvjensen 
Sorry copy and past error.

[image]

Fix is on github:

```
line79: lastmonth_dts = datetime.strptime(self.__checkPayload__('last_month_measure_date', now.strftime(DATEFORMAT_DAY)), DATEFORMAT_DAY)
```

cvjensen, 2021-12-18:
Here we go - no errors!

```
pi@raspberrypi:/opt/ha-watermeter $ python3 app.py
2021-12-18 16:41:21,120 - INFO: ✔︎ Watermeter data application start (__main__:65)
2021-12-18 16:41:21,121 - DEBUG: ✔︎ Watermeter MQTT Client localhost (__main__:62)
2021-12-18 16:41:21,123 - INFO: ✔︎ Watermeter connect to MQTT Client localhost, tele/multical21/LWT (__main__:65)
2021-12-18 16:41:21,146 - DEBUG: ✔︎ localhost connected OK, subscribe to topic wmbusmeters/77634443 (__main__:62)
2021-12-18 16:41:21,153 - DEBUG: ✔︎ Message Id 1 published. (__main__:62)
2021-12-18 16:41:25,704 - INFO: ✔︎ MQTT Message received topic wmbusmeters/77634443 (__main__:65)
2021-12-18 16:41:25,705 - INFO: ✔︎ MQTT Publish state topic tele/multical21/wasser/status (__main__:65)
2021-12-18 16:41:25,705 - DEBUG: Init calculator (lib.calculator:62)
2021-12-18 16:41:25,706 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
2021-12-18 16:41:25,706 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/77634443.json (lib.calculator:65)
2021-12-18 16:41:25,707 - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator:62)
2021-12-18 16:41:25,708 - INFO: ✔︎ Start calculation (lib.calculator:65)
2021-12-18 16:41:25,716 - DEBUG: Missing key last_month_measure_date - used default value 2021-12-18 (lib.calculator:62)
2021-12-18 16:41:25,718 - DEBUG: Missing key last_month_total_m3 - used default value 0.0 (lib.calculator:62)
2021-12-18 16:41:25,718 - INFO: ✔︎ new data calculation (lib.calculator:65)
2021-12-18 16:41:25,719 - DEBUG: update periodes and timestamps (lib.calculator:62)
2021-12-18 16:41:25,719 - DEBUG: update water leak status (lib.calculator:62)
2021-12-18 16:41:25,719 - DEBUG: Missing key current_alarms - used default value no alarm (lib.calculator:62)
2021-12-18 16:41:25,719 - DEBUG: Missing key previous_alarms - used default value no alarm (lib.calculator:62)
2021-12-18 16:41:25,720 - INFO: ✔︎ Publish the new data tele/multical21/wasser (lib.calculator:65)
2021-12-18 16:41:25,721 - DEBUG: ✔︎ saved current data to /opt/ha-watermeter/data/77634443.json (lib.calculator:62)
2021-12-18 16:41:25,721 - DEBUG: ✔︎ Message Id 3 published. (__main__:62)
2021-12-18 16:41:25,722 - DEBUG: ✔︎ Message Id 4 published. (__main__:62)

```

Now I have a data/77634443.json file that looks like this:

```
{
  "name": "Multical21",
  "device": "watermeter",
  "deviceid": "77634443",
  "date": "2021-12-18",
  "time": "15:41:56",
  "total_m3": 0.094,
  "ratio": 0,
  "m3": {
    "current": 0,
    "hour": 0.094,
    "day": 0.094,
    "month": 0.094,
    "year": -117.636
  },
  "liter": {
    "current": 0,
    "hour": 94,
    "day": 94,
    "month": 94,
    "year": -117636
  },
  "last_total": {
    "hour": "2020-14-09",
    "hour_m3": 0,
    "day": "2020-07-14",
    "day_m3": 0,
    "month": "2021-12",
    "month_m3": 0,
    "year": "2019-12-31",
    "year_m3": 117.73
  },
  "wmbusmeter": {
    "media": "cold water",
    "meter": "multical21",
    "name": "Multical21",
    "id": "77634443",
    "total_m3": 0.094,
    "target_m3": 0.094,
    "max_flow_m3h": 0,
    "flow_temperature_c": 127,
    "external_temperature_c": 4,
    "current_status": "DRY",
    "time_dry": "22-31 days",
    "time_reversed": "",
    "time_leaking": "",
    "time_bursting": "",
    "timestamp": "2021-12-18T15:41:56Z",
    "device": "im871a[00102807]",
    "rssi_dbm": -76
  },
  "alarm": "no alarm",
  "last_alarm": "no alarm",
  "periode": "2021-12-18",
  "month": "2021-12",
  "year": "2021-12",
  "timestamp": "2021-12-18T15:41:56Z",
  "last_update": "2021-12-18 16:41:56.879069",
  "elapsed_time": "0:00:00",
  "icon": "mdi:billboard",
  "unit_of_measurement": "㎥",
  "last_reset": "1970-01-01T00:00:00+00:00",
  "state_class": "measurement",
  "device_class": "energy",
  "version": "1.0.3",
  "attribution": "Data provided by Peter Siebler",
  "data_provider": "raspberrypi"
}
```

zibous, 2021-12-18:
@cvjensen 

o.k. and do you see the mqtt messages fpr your defined topic ?
if yes this will hold all data from the json file.

cvjensen, 2021-12-18:
Yes I do , I have just changed the config.py MQTT_PUBLISH_TOPIC and MQTT_STATE_TOPIC so it should be the same as your smartmeter-water.yaml . Now I just have to figure out how I get it shown in HA 😄 thanks for the help so far !

zibous, 2021-12-18:
@cvjensen 
o.k fine.
But i see that some values are negativ:

```
"m3": {
    "current": 0,
    "hour": 0.094,
    "day": 0.094,
    "month": 0.094,
    "year": -117.636
  },
```
This happens if the previous data (data/77634443.json) holds no valid data. You can fix this if you
edit the data. The calculation is allways:

```
 
 result = previous data (77634443.json) - current data (mqtt payload).
 
```

zibous, 2021-12-18:
> it should be the same as your smartmeter-water.yaml
Or yo can change the sensor topics:

```yaml
## ########################################
## Verbrauchsmessung Wasser
## Aktueller Verbrauch
## Liter =  total_m3 * 1000.000
## ########################################

## ######################################
## all from wmbusmeters
## MQTT-TOPIC: tele/wasser/verbrauch
## ######################################

## {
##  "media":"water",
##  "meter":"izar",
##  "name":"watermeter",
##  "id":"43430778",
##  "total_m3":166.204,
##  "last_month_total_m3":162.614,
##  "last_month_measure_date":"2020-07-01",
##  "remaining_battery_life_y":13,
##  "current_alarms":"no_alarm",
##  "previous_alarms":"no_alarm",
##  "timestamp":"2020-07-14T14:18:45Z"
## }

- platform: mqtt
  name: "Wasserzähler aktuell"
  state_topic: "tele/wasser/verbrauch"
  unit_of_measurement: "m³"
  value_template: "{{ value_json['total_m3']|default(0.000)}}"
  icon: mdi:counter

- platform: mqtt
  name: "Wasserzähler letzer Monat"
  state_topic: "tele/wasser/verbrauch"
  unit_of_measurement: "m³"
  value_template: "{{ value_json['last_month_total_m3']|default(0.000)}}"
  icon: mdi:counter

- platform: mqtt
  name: "Wasserzähler Alarm"
  state_topic: "tele/wasser/verbrauch"
  value_template: "{{ value_json['current_alarms']|default('OFFLINE')}}"
  json_attributes_topic: "tele/wasser/verbrauch"
  icon: mdi:alarm-light-outline

- platform: mqtt
  name: "Wasserzähler voriger Alarm"
  state_topic: "tele/wasser/verbrauch"
  value_template: "{{ value_json['previous_alarms']|default('OFFLINE')}}"
  json_attributes_topic: "tele/wasser/verbrauch"
  icon: mdi:alarm-light-outline

- platform: mqtt
  name: "Wasserzähler Zeit"
  state_topic: "tele/wasser/verbrauch"
  value_template: "{{ value_json['timestamp']|default('OFFLINE')}}"
  json_attributes_topic: "tele/wasser/verbrauch"
  icon: mdi:timetable

- platform: mqtt
  name: "Wasserzähler vorige Messung"
  state_topic: "tele/wasser/verbrauch"
  value_template: "{{ value_json['timestamp']|default('OFFLINE')}}"
  json_attributes_topic: "tele/wasser/verbrauch"
  icon: mdi:timetable
[... 158 lines cut ...]
        'mechanical_fraud':'Mechanischer Felhler',
        'no_alarm':'Kein Alarm',
        'general_alarm':'Allgemeiner Alarm',
        'offline': 'Sensor OFFLINE'} 
    %}
    {% set state = value_json['last_alarm']|default('offline') %}
    {% set alarm = mapper[state] if state in mapper %}
    {{ alarm }}

- platform: mqtt
  name: "Zählerstand"
  state_topic: "tele/smartmeter/wasser"
  value_template: "{{ value_json['total_m3']|default(0.000)}}"
  unit_of_measurement: "m³"
  json_attributes_topic: "tele/smartmeter/wasser"
  icon: mdi:counter

- platform: mqtt
  name: "Hauswasserzähler Status"
  state_topic: "tele/smartmeter/wasser"
  value_template: "{{ value_json['last_update']|default('OFFLINE')}}"
  json_attributes_topic: "tele/smartmeter/wasser"
  icon: mdi:timetable

```

cvjensen, 2021-12-18:
Alright but how do you suggest I get smartmeter-water.yaml and input_numbers/water.yaml included in my configuration.yaml? 😕

zibous, 2021-12-18:
@cvjensen 
I use a config folder and include all yaml's.

see: https://www.home-assistant.io/docs/configuration/splitting_configuration/#example-include_dir_list

[image]

Homassistant configuration.yaml

```yaml

## #####################################
## Groups as names
## #####################################
# group: !include config/groups.yaml/
group: !include_dir_merge_named config/groups/

## #####################################
## Include automation as list
## #####################################
## scene: !include_dir_merge_list config/scenes/

## #####################################
## Include automation as list
## #####################################
## automation: !include automations.yaml
automation: !include_dir_merge_list config/automations/

## #####################################
## Include scripts as names
## #####################################
## script: !include scripts.yaml
script: !include_dir_merge_named config/scripts/

## #####################################
## Include camera as list
## #####################################
camera: !include_dir_merge_list config/cameras/

## #########################################
## Include number, select,text and boolean
## #########################################
input_number: !include_dir_merge_named config/input_numbers/
input_select: !include_dir_merge_named config/input_selects/

input_boolean: !include_dir_merge_named config/input_booleans/

input_datetime: !include_dir_merge_named config/input_datetime/
input_text: !include_dir_merge_named config/input_text/

## #####################################
## Include media players
## #####################################
## media_player: !include_dir_merge_list config/media_players/

## #####################################
## Include notifications as list
## #####################################
notify: !include_dir_merge_list config/notifications/

## #####################################
## Include sensors as list
## #####################################
sensor: !include_dir_merge_list config/sensors/

## #####################################
## Include binary sensors and device tracker
## #####################################
binary_sensor: !include_dir_merge_list config/binarysensors/

## #####################################
## Include switches as list
## #####################################
## switch: !include_dir_merge_list config/switches/

## #####################################
## Include climate as list
## #####################################
## climate: !include_dir_merge_list config/climate/

## #####################################
## Include locks as list
## #####################################
## lock: !include_dir_merge_list config/locks/

## #####################################
## Include lights as list
## #####################################
light: !include_dir_merge_list config/lights/

## #####################################
## Include covers as list
## #####################################
## cover: !include_dir_merge_list config/covers/

## #####################################
## Include fans as list
## #####################################
## fan: !include_dir_merge_list config/fans/

## #####################################
## Include timers as list
## #####################################
timer: !include_dir_merge_named config/timers/

## #####################################
## Include counter as names
## #####################################
counter: !include_dir_merge_named config/counters/

## #####################################
## Include utility meter as names
## #####################################
utility_meter: !include_dir_merge_named config/utility_meters/

## #####################################
## Include shell commands
## #####################################
shell_command: !include_dir_merge_named shell_commands
```
```

cvjensen, 2021-12-18:
Alright , I'm started to move some stuff out of my configuration.yaml. As a start I will just have a single file for each thing like automations.yaml, lights.yaml and so on.

I have now make the following in my configuration.yaml:

```
sensor: !include config_files/smartmeter-water.yaml
input_number: !include config_files/input_numbers.yaml
```

smartmeter-water.yaml is, yes, your file. And input_numbers.yaml is a copy of your examle input_number/water.yaml.

But it seems that only the input_numbers is loaded in HA ... I cant find any of the sensors in the smartmeter-water.yaml. 😑

cvjensen, 2021-12-18:
Can I only have one single "sensor:" inside configuration.yaml? ...

zibous, 2021-12-18:
@cvjensen 
!include  do not work.

you must use : 

sensor: !include_dir_merge_list config_files/sensors/
input_number: !include_dir_merge_named config_files/input_numbers/

+ config_files
  + input_numbers
      input_numbers.yaml
   + sensors
      smartmeter-water.yaml

cvjensen, 2021-12-18:
Awesome , thanks for all the help and tips about config structure. I now have both the input_numbers and the sensors in HA. I'm not sure all the sensor data is correct and I still have alot of stuff that has to be corrected. BUT. I have the data in HA and that is great. It shows me that I now can move on with the installation of the watermeter 👍 Thanks alot and Merry christmas to you @zibous ! 🎄 🎁 🎅

closed 2021-12-18 by cvjensen

zibous, 2021-12-18:
@cvjensen 
OK. I am happy that I could help. Thanks for the wishes and good luck with the implementation.