zibous/ha-watermeter #7: calculator.py doesnt count properly "current" "hour" "day" usage

nayspl, 2022-02-08:
Running on latest github version output.json:

`
    "name": "Woda Dom",
    "device": "watermeter",
    "deviceid": "4113f6d2",
    "date": "2022-02-08",
    "time": "08:04:10",
    "total_m3": 241.298,
    "ratio": 1.01,
    "m3": {
        **"current": 0.0,**
        **"hour": 241.298,**
        **"day": 241.298,**
        "month": 1.31,
        "year": 123.568
    },
    "liter": {
        **"current": 0.0,**
        **"hour": 241298.0,**
        **"day": 241298.0,**
        "month": 1310.0,
        "year": 123568.0
    },
    "last_total": {
        "hour": "2020-14-09",
        "hour_m3": 0.0,
        "day": "2020-07-14",
        "day_m3": 0.0,
        "month": "2022-01",
        "month_m3": 239.988,
        "year": "2019-12-31",
        "year_m3": 117.73
    },
    "wmbusmeter": {
        "media": "water",
        "meter": "izar",
        "name": "watermeter",
        "id": "4113f6d2",
        "prefix": "C18SB",
        "serial_number": "085586",
        "total_m3": 241.298,
        "last_month_total_m3": 239.988,
        "last_month_measure_date": "2022-01-31",
        "remaining_battery_life_y": 7,
        "current_alarms": "underflow",
        "previous_alarms": "no_alarm",
        "transmit_period_s": 8,
        "manufacture_year": "2018",
        "timestamp": "2022-02-08T08:04:10Z",
        "device": "rtlwmbus[00000001]",
        "rssi_dbm": 123
    },
    "alarm": "underflow",
    "last_alarm": "no_alarm",
    "periode": "2022-02-08",
    "month": "2022-02",
    "year": "2022-02",
    "timestamp": "2022-02-08T08:04:10Z",
    "last_update": "2022-02-08 09:04:10.115986",
    "elapsed_time": "0:00:00",
    "icon": "mdi:billboard",
    "unit_of_measurement": "\u33a5",
    "last_reset": "1970-01-01T00:00:00+00:00",
    "state_class": "measurement",
    "device_class": "energy",
    "version": "1.0.3",
    "attribution": "Data provided by Peter Siebler",
    "data_provider": "lejbek1"
}`

I'm not python expert but from my understanding code looks OK. Dont know where is an issue...

cut from config,py
`# date and time formats
DATEFORMAT_TIMESTAMP = '%Y-%m-%dT%H:%M:%SZ'
DATEFORMAT_CURRENT   = '%Y-%m-%d %H:%M:%S.%f'
DATEFORMAT_HOUR = '%H'
DATEFORMAT_DAY = '%Y-%m-%d'
DATEFORMAT_MONTH = '%Y-%m'
DATEFORMAT_YEAR = '%Y-%m'
TIME_FORMAT = '%H:%M:%S'

# application settings
SMARTMETER_ID = '4113f6d2'  # internal used smartmeter identification
SMARTMETER_NAME = 'Woda Dom'  # Name of the smartmeter
APPS_VERSION = '1.0.3'  # Application version

# Datafile, if defined every hour the data will be stored as csv to this location
DATAFILE = os.path.join(os.path.dirname(__file__), 'data/' + SMARTMETER_ID + ".json")

# Reportfile, if defined every end of month data will be saved to this location
DAYREPORTFILE = os.path.join(os.path.dirname(__file__), 'data/' + SMARTMETER_ID + "-day-report.csv")

# Reportfile, if defined every end of month data will be saved to this location
REPORTFILE = os.path.join(os.path.dirname(__file__), 'data/' + SMARTMETER_ID + "-report.csv")

# all for the mqtt brocker
MQTT_HOST = '192.168.1.192'  # mqtt brocker host ! must be a string
MQTT_PORT = 1883  # mqtt brocker port ! must be a number
MQTT_APPID = '*****'  # mqtt brocker user name
MQTT_PASSWORD = '********'  # mqtt brocker user password
MQTT_KEEPALIVE = 60  # mqtt brocker keep alive time

# all topic settings
# test: mosquitto_sub -h localhost -p 1883   -u 'pi' -P 'raspberry' -v -t 'tele/wasser/verbrauch'
MQTT_TOPIC = 'tele/wasser/verbrauch'  # listen to this wmbusmeters topic

# all application publish topics
MQTT_TOPIC_NAME = 'IZAR'
MQTT_PUBLISH_TOPIC = 'tele/' + MQTT_TOPIC_NAME + '/wasser'  # used for publishing new data
MQTT_STATE_TOPIC = 'tele/' + MQTT_TOPIC_NAME + '/wasser/status'  # mqtt state topic
MQTT_AVAILABILITY_TOPIC = 'tele/' + MQTT_TOPIC_NAME + '/LWT'  # mqtt availability topic Online | Offline
MQTT_ENABLE_LOGGING = False  # enable | disable debug logging mqtt

# switch logging
LOG_LEVEL = 10  # DEBUG: 10
# LOG_LEVEL = 20  # INFO: 20
# LOG_LEVEL = 30  # WARNING: 30
# LOG_LEVEL = 40  # ERROR: 40
# LOG_LEVEL = 50  # CRITICAL: 50
# LOG_LEVEL = 100 # DISABLED: 100
LOG_DIR = './logs/'

# write summary data to console
DATALOG_ENABLED = True  # enable True | False, writes datalog to console
DATALOG_FILENAME = './logs/data.csv'  # output folder for data log
`
data.csv file doesnt create  as welll i dont see any datalog on console too

zibous, 2022-02-08:
It looks like you're running it for the first time. 
As a result, the last value is 0 and that is why the `current Hour` and `Day` are so high. 
Should change on the next run.

You can check the whole thing with MQTT Explorer.
https://mqtt-explorer.com/

nayspl, 2022-02-09:
Hello,

Actually it is running since few weeks, so that is a reason i report the issue, but you are right i didn't edit initial values in my config.py. 
Will do it now. 

Yes i'm using MQTT explorer all the time. +tip. 
I will edit initial values right now and back to you soon. Thank you

zibous, 2022-02-09:
I can't understand that because the python script does the following:

1. When it is started for the first time, the default settings are taken from config.py. 
    `result = current value - value form config.py`
2. If data is collected, a `data/4113f6d2.json` is created. 
3. When running again, the saved data is taken from the `data/4113f6d2.json` and the calculations are carried out.
    `result = current value - previous value from 4113f6d2.json`

## Calculation

```python
          self.cd['m3']['hour']=round(meterM3 - float(self.pd['last_total']['hour_m3']), 3)
          self.cd['m3']['day']=round(meterM3 - float(self.pd['last_total']['day_m3']), 3)
```

```ìni
meterM3: current Meter value
self.pd: last value from config.py | 4113f6d2.json
```

if the value for hour / day is always so high (total_m3== m3. hour), then it could be that the calculation is always made via the config.py settings and not from data/4113f6d2.json.

Can you find the file data/4113f6d2.json and does it contain this data ?

nayspl, 2022-02-09:
1.  I removed all files from `data/  ` before first run then edited config.py with init values.
2. Data is collecting and stored as `data/4113f6d2.json` (but with high values )
3. Yes that is right based on calculator but i have no idea why the values are always so high ? some calculation is not working right for me. I cant figure out by my own what is wrong, like a said from code point of view everything looks good
Even, When I reedit manually `values from 4113f6d2.json` with some lower values it always generate high one.I started thinking maybe there is some problem with UTC or local time... (its my guessing).

my config.py
```
# data model smartmeter
WATERMETER_DATA = {
    "name": SMARTMETER_NAME,
    "device": "watermeter",
    "deviceid": SMARTMETER_ID,
    "date": "",
    "time": "",
    "total_m3": 241.43,
    "ratio": 0.00,
    "m3": {
        "current": 0.001,
        "hour": 0.232,
        "day": 1.242,
        "month":  1.442,
        "year": 123.7
    },
    "liter": {
        "current": 1.0,
        "hour": 232.0,
        "day": 1242.0,
        "month": 1442.0,
        "year": 123700.0
    },
    "last_total": {
        "hour": "2020-14-09",
        "hour_m3": 0.000,
        "day": "2020-07-14",
        "day_m3": 0.000,
        "month": "2020-07",
        "month_m3": 0.000,
        "year": "2019-12-31",
        "year_m3": 117.730
    },
    "wmbusmeter": "",
    "alarm": "No alarm",
    "last_alarm": "No alarm",
    "periode": now.strftime(DATEFORMAT_DAY),
    "month": now.strftime(DATEFORMAT_MONTH),
    "year": now.strftime(DATEFORMAT_YEAR),
    "timestamp": now.strftime(DATEFORMAT_TIMESTAMP),
    "last_update": now.strftime(DATEFORMAT_CURRENT),
    "elapsed_time": 0,
    "icon": "mdi:billboard",
    "unit_of_measurement": u"\u33A5",
    "last_reset": "1970-01-01T00:00:00+00:00",
    "state_class": "measurement",
    "device_class": "energy",
    "version": APPS_VERSION,
    "attribution": "Data provided by Peter Siebler"
}
```
and `data/4113f6d2.json`

```
{
  "name": "Woda Dom",
  "device": "watermeter",
  "deviceid": "4113f6d2",
  "date": "2022-02-09",
  "time": "12:42:24",
  "total_m3": 241.43,
  "ratio": 1.01,
  "m3": {
    "current": 0,
    "hour": 241.43,
    "day": 241.43,
    "month": 1.442,
    "year": 123.7
  },
  "liter": {
    "current": 0,
    "hour": 241430,
    "day": 241430,
    "month": 1442,
    "year": 123700
  },
  "last_total": {
    "hour": "2020-14-09",
    "hour_m3": 0,
    "day": "2020-07-14",
    "day_m3": 0,
    "month": "2022-01",
    "month_m3": 239.988,
    "year": "2019-12-31",
    "year_m3": 117.73
  },
  "wmbusmeter": {
    "media": "water",
    "meter": "izar",
    "name": "watermeter",
    "id": "4113f6d2",
    "prefix": "C18SB",
    "serial_number": "085586",
    "total_m3": 241.43,
    "last_month_total_m3": 239.988,
    "last_month_measure_date": "2022-01-31",
    "remaining_battery_life_y": 7,
    "current_alarms": "underflow",
    "previous_alarms": "no_alarm",
    "transmit_period_s": 8,
    "manufacture_year": "2018",
    "timestamp": "2022-02-09T12:42:24Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 123
  },
  "alarm": "underflow",
  "last_alarm": "no_alarm",
  "periode": "2022-02-09",
  "month": "2022-02",
  "year": "2022-02",
  "timestamp": "2022-02-09T12:42:24Z",
  "last_update": "2022-02-09 13:42:24.367142",
  "elapsed_time": "0:00:00",
  "icon": "mdi:billboard",
  "unit_of_measurement": "㎥",
  "last_reset": "1970-01-01T00:00:00+00:00",
  "state_class": "measurement",
  "device_class": "energy",
  "version": "1.0.3",
  "attribution": "Data provided by Peter Siebler",
  "data_provider": "lejbek1"
}
```

zibous, 2022-02-09:
What does the console log show?

The determination for the hour and day values:

```python 

 current_dts = datetime.strptime(self.__checkPayload__('timestamp', now.strftime(DATEFORMAT_TIMESTAMP)), DATEFORMAT_TIMESTAMP)
 previous_dts = datetime.strptime(self.__checkPayload__('timestamp', now.strftime(DATEFORMAT_TIMESTAMP)), DATEFORMAT_TIMESTAMP)

 # check update last total for hour
 if(current_dts.strftime(DATEFORMAT_HOUR) != previous_dts.strftime(DATEFORMAT_HOUR)):
     # new hour
     log.debug(f"✔︎ Update last total hour: {current_dts.strftime(DATEFORMAT_HOUR)} prev: {previous_dts.strftime(DATEFORMAT_HOUR)}")
     self.cd['last_total']['hour']=current_dts.strftime(DATEFORMAT_HOUR)
     self.cd['last_total']['hour_m3']=meterM3
     # new hour, store the previous data to
     self.__printLogData()
 else:
     self.cd['last_total']['hour']=self.pd['last_total']['hour']
     self.cd['last_total']['hour_m3']=self.pd['last_total']['hour_m3']

 # check update last total for day
 if(current_dts.strftime(DATEFORMAT_DAY) != previous_dts.strftime(DATEFORMAT_DAY)):
     # new day
     log.debug(f"✔︎ Update last total day: {current_dts.strftime(DATEFORMAT_DAY)} prev: {previous_dts.strftime(DATEFORMAT_DAY)}")
     self.cd['last_total']['day']=current_dts.strftime(DATEFORMAT_DAY)
     self.cd['last_total']['day_m3']=meterM3
     self.__savedayreport__()
 else:
     self.cd['last_total']['day']=self.pd['last_total']['day']
     self.cd['last_total']['day_m3']=self.pd['last_total']['day_m3']
```
If there is a problem with  UTC or local time you can see every time the message:
` log.debug(f"✔︎ Update last total day: {current_dts.strftime(DATEFORMAT_DAY)} prev:.... `

nayspl, 2022-02-09:
```
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,629 - INFO: ✔︎ MQTT Message received topic tele/wasser/verbrauch (__main__:65)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,630 - INFO: ✔︎ MQTT Publish state topic tele/IZAR/wasser/status (__main__:65)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,630 - DEBUG: Init calculator (lib.calculator:62)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,631 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,631 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:65)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,631 - DEBUG: ✔︎ previous data loaded /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:62)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,632 - INFO: ✔︎ Start calculation (lib.calculator:65)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,632 - INFO: ✔︎ new data calculation (lib.calculator:65)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,633 - DEBUG: update periodes and timestamps (lib.calculator:62)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,633 - DEBUG: update water leak status (lib.calculator:62)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,633 - INFO: ✔︎ Publish the new data tele/IZAR/wasser (lib.calculator:65)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,634 - DEBUG: ✔︎ saved current data to /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:62)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,636 - DEBUG: ✔︎ Message Id 677 published. (__main__:62)
lut 09 22:32:01 lejbek1 python3[905125]: 2022-02-09 22:32:01,636 - DEBUG: ✔︎ Message Id 678 published. (__main__:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,479 - INFO: ✔︎ MQTT Message received topic tele/wasser/verbrauch (__main__:65)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,480 - INFO: ✔︎ MQTT Publish state topic tele/IZAR/wasser/status (__main__:65)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,481 - DEBUG: Init calculator (lib.calculator:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,481 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,481 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:65)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,482 - DEBUG: ✔︎ previous data loaded /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,482 - INFO: ✔︎ Start calculation (lib.calculator:65)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,483 - INFO: ✔︎ new data calculation (lib.calculator:65)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,483 - DEBUG: update periodes and timestamps (lib.calculator:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,483 - DEBUG: update water leak status (lib.calculator:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,483 - INFO: ✔︎ Publish the new data tele/IZAR/wasser (lib.calculator:65)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,484 - DEBUG: ✔︎ saved current data to /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,485 - DEBUG: ✔︎ Message Id 679 published. (__main__:62)
lut 09 22:32:11 lejbek1 python3[905125]: 2022-02-09 22:32:11,485 - DEBUG: ✔︎ Message Id 680 published. (__main__:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,175 - INFO: ✔︎ MQTT Message received topic tele/wasser/verbrauch (__main__:65)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,175 - INFO: ✔︎ MQTT Publish state topic tele/IZAR/wasser/status (__main__:65)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,176 - DEBUG: Init calculator (lib.calculator:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,176 - DEBUG: ✔︎ Load previous data (lib.calculator:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,176 - INFO: ✔︎ get previous data from /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:65)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,177 - DEBUG: ✔︎ previous data loaded /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,177 - INFO: ✔︎ Start calculation (lib.calculator:65)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,178 - INFO: ✔︎ new data calculation (lib.calculator:65)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,178 - DEBUG: update periodes and timestamps (lib.calculator:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,178 - DEBUG: update water leak status (lib.calculator:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,179 - INFO: ✔︎ Publish the new data tele/IZAR/wasser (lib.calculator:65)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,180 - DEBUG: ✔︎ saved current data to /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,180 - DEBUG: ✔︎ Message Id 681 published. (__main__:62)
lut 09 22:32:20 lejbek1 python3[905125]: 2022-02-09 22:32:20,181 - DEBUG: ✔︎ Message Id 682 published. (__main__:62)
[... cut ...]
lut 09 22:32:54 lejbek1 python3[905125]: 2022-02-09 22:32:54,822 - INFO: ✔︎ Publish the new data tele/IZAR/wasser (lib.calculator:65)
lut 09 22:32:54 lejbek1 python3[905125]: 2022-02-09 22:32:54,823 - DEBUG: ✔︎ saved current data to /opt/ha-watermeter/data/4113f6d2.json (lib.calculator:62)
lut 09 22:32:54 lejbek1 python3[905125]: 2022-02-09 22:32:54,823 - DEBUG: ✔︎ Message Id 689 published. (__main__:62)
lut 09 22:32:54 lejbek1 python3[905125]: 2022-02-09 22:32:54,824 - DEBUG: ✔︎ Message Id 690 published. (__main__:62)

```

zibous, 2022-02-11:
@nayspl 

You can try the new version:

1. Update and check the new settings for config.py
2. Enable Calc Debugging: DEBUG_DATA = True, enables WARNING Messages for calculations.
3. Delete /data/4113f6d2.json
4. Start application: `python3 app.py`

```log
2022/02/11 10:46:01.730  - DEBUG: Init calculator (lib.calculator: Line 69)
2022/02/11 10:46:01.731  - DEBUG: ✔︎ Load previous data (lib.calculator: Line 69)
2022/02/11 10:46:01.731  - INFO:  ✔︎ get previous data from /Users/petsie1612/ips_apps/ha-watermeter/data/43430778.json (lib.calculator: Line 72)
2022/02/11 10:46:01.731  - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator: Line 69)
2022/02/11 10:46:01.731  - INFO:  ✔︎ Start calculation (lib.calculator: Line 72)
2022/02/11 10:46:01.734  - WARNING: ✔︎ Date current:2022-02-11 09:46:01, prevoius:2022-02-11 09:46:01,last month:2022-02-01 00:00:00, last year:2022  (lib.calculator: Line 75)
2022/02/11 10:46:01.734  - INFO:  ✔︎ new data calculation m3, liter (lib.calculator: Line 72)
2022/02/11 10:46:01.735  - INFO:  ✔︎ Init first run, try to adjust the prevoius values. (lib.calculator: Line 72)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: result = data[current,hour,day,month,year] - previous[current,hour,day,month,year] (lib.calculator: Line 75)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: current m3   : 0.0=343.955-343.955 (lib.calculator: Line 75)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: current hour : 0.0=343.955-343.955 (lib.calculator: Line 75)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: current day  : 0.0=343.955-343.955 (lib.calculator: Line 75)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: current month: 2.776=343.955-341.179 (lib.calculator: Line 75)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: current year : 343.955=343.955-0.0 (lib.calculator: Line 75)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: Check update last total hour: 09 prev: 09 (lib.calculator: Line 75)
2022/02/11 10:46:01.735  - WARNING: ⁍ DATA: Check last total day update: 2022-02-11 prev: 2022-02-11 (lib.calculator: Line 75)
2022/02/11 10:46:01.736  - WARNING: ⁍ DATA: Check last total month update: 2022-02 prev: 2022-02 (lib.calculator: Line 75)
2022/02/11 10:46:01.736  - WARNING: ⁍ DATA: Check last year update: 2022 prev: 2022 (lib.calculator: Line 75)
2022/02/11 10:46:01.736  - DEBUG: ✔︎ update periodes and timestamps (lib.calculator: Line 69)
2022/02/11 10:46:01.736  - DEBUG: ✔︎ update water leak status (lib.calculator: Line 69)
2022/02/11 10:46:01.736  - INFO:  ✔︎ Publish the new data tele/IZAR_TEST/wasser (lib.calculator: Line 72)
```

nayspl, 2022-02-12:
> @nayspl
> 
> You can try the new version:
> 
>     1. Update and check the new settings for config.py   - done
> 
>     2. Enable Calc Debugging: DEBUG_DATA = True, enables WARNING Messages for calculations.  - done
> 
>     3. Delete /data/4113f6d2.json  - done
> 
>     4. Start application: `python3 app.py` - done
> 
Here is my out after changes:
```
root@lejbek1:/opt/ha-watermeter# ./app.py
2022/02/12 00:58:19.486  - INFO:  ❖ Watermeter data application start  -------------------- (__main__: Line 168)
2022/02/12 00:58:19.486  - DEBUG: ❖ Watermeter MQTT Client 192.168.1.192 (__main__: Line 126)
2022/02/12 00:58:19.487  - INFO:  ❖ Watermeter connect to MQTT Client 192.168.1.192, tele/IZAR/LWT (__main__: Line 142)
2022/02/12 00:58:19.488  - DEBUG: ❖ 192.168.1.192 connected OK, subscribe to topic tele/wasser/verbrauch (__main__: Line 84)
2022/02/12 00:58:19.489  - DEBUG: ❖ Message Id 1 published. (__main__: Line 92)
2022/02/12 00:58:26.675  - INFO:  ❖ MQTT Message received topic tele/wasser/verbrauch (__main__: Line 96)
2022/02/12 00:58:26.676  - INFO:  ❖ MQTT Publish state topic tele/IZAR/wasser/status (__main__: Line 103)
2022/02/12 00:58:26.676  - DEBUG: ✔︎ Init calculator (lib.calculator: Line 36)
2022/02/12 00:58:26.676  - DEBUG: ✔︎ Load previous data (lib.calculator: Line 89)
2022/02/12 00:58:26.677  - INFO:  ✔︎ get previous data from /opt/ha-watermeter/data/4113f6d2.json (lib.calculator: Line 245)
2022/02/12 00:58:26.677  - DEBUG: ✔︎ init previous data, datafile not found! (lib.calculator: Line 256)
2022/02/12 00:58:26.677  - INFO:  ✔︎ Start calculation (lib.calculator: Line 94)
2022/02/12 00:58:26.681  - WARNING: ✔︎ Date current:2022-02-11 23:58:26, prevoius:2022-02-11 23:58:26,last month:2022-01-31 00:00:00, last year:2022  (lib.calculator: Line 103)
2022/02/12 00:58:26.682  - INFO:  ✔︎ new data calculation m3, liter (lib.calculator: Line 127)
2022/02/12 00:58:26.682  - INFO:  ✔︎ Init first run, try to adjust the prevoius values. (lib.calculator: Line 131)
2022/02/12 00:58:26.682  - WARNING: ⁍ DATA: result = data[current,hour,day,month,year] - previous[current,hour,day,month,year] (lib.calculator: Line 149)
2022/02/12 00:58:26.682  - WARNING: ⁍ DATA: current m3   : 0.0=241.87-241.87 (lib.calculator: Line 150)
2022/02/12 00:58:26.682  - WARNING: ⁍ DATA: current hour : 0.0=241.87-241.87 (lib.calculator: Line 151)
2022/02/12 00:58:26.683  - WARNING: ⁍ DATA: current day  : 0.0=241.87-241.87 (lib.calculator: Line 152)
2022/02/12 00:58:26.683  - WARNING: ⁍ DATA: current month: 1.882=241.87-239.988 (lib.calculator: Line 153)
2022/02/12 00:58:26.683  - WARNING: ⁍ DATA: current year : 124.14=241.87-117.73 (lib.calculator: Line 154)
2022/02/12 00:58:26.683  - WARNING: ⁍ DATA: Check update last total hour: 23 prev: 23 (lib.calculator: Line 158)
2022/02/12 00:58:26.684  - WARNING: ⁍ DATA: Check last total day update: 2022-02-11 prev: 2022-02-11 (lib.calculator: Line 173)
2022/02/12 00:58:26.684  - WARNING: ⁍ DATA: Check last total month update: 2022-02 prev: 2022-02 (lib.calculator: Line 187)
2022/02/12 00:58:26.684  - WARNING: ⁍ DATA: Check last year update: 2022 prev: 2022 (lib.calculator: Line 202)
2022/02/12 00:58:26.684  - DEBUG: ✔︎ update periodes and timestamps (lib.calculator: Line 216)
2022/02/12 00:58:26.685  - DEBUG: ✔︎ update water leak status (lib.calculator: Line 227)
2022/02/12 00:58:26.685  - INFO:  ✔︎ Publish the new data tele/IZAR/wasser (lib.calculator: Line 232)
2022/02/12 00:58:26.686  - DEBUG: ✔︎ saved current data to /opt/ha-watermeter/data/4113f6d2.json (lib.calculator: Line 274)
2022/02/12 00:58:26.686  - DEBUG: ❖ Message Id 3 published. (__main__: Line 92)
2022/02/12 00:58:26.686  - DEBUG: ❖ Message Id 4 published. (__main__: Line 92)
```
I saw some little change in `config.py` I think it was a source of mine issue:
```
- DATEFORMAT_YEAR = '%Y-%m'
+ DATEFORMAT_YEAR = '%Y'
```

closed 2022-02-12 by nayspl