zibous/ha-watermeter #8: calculator.py doesn't change hour and day

Silvester5, 2022-02-21:
Hi Zibous,
thanks for the great work on the watermeter.
My watermeter is counting up Wasser/h and Wasser/day.
Is there an issue in calculator.py?

           current_dts = datetime.strptime(self.__checkPayload__('timestamp', now.strftime(DATEFORMAT_TIMESTAMP)), DATEFORMAT_TIMESTAMP)
            previous_dts = datetime.strptime(self.__checkPayload__('timestamp', now.strftime(DATEFORMAT_TIMESTAMP)), DATEFORMAT_TIMESTAMP)
            lastmonth_dts = datetime.strptime(self.__checkPayload__('last_month_measure_date', now.strftime(DATEFORMAT_DAY)), DATEFORMAT_DAY)

            if (DEBUG_DATA == True):x
                log.logger.warning("✔︎ Date current:{}, prevoius:{},last month:{}, last year:{} ".format(current_dts,previous_dts,lastmonth_dts, previous_dts.strftime(DATEFORMAT_YEAR)))

            self.elapsed_time = current_dts - previous_dts`

zibous, 2022-02-21:
@Silvester5 
Yes you are right, there is an issue in my code.

Update to the new version or change the line:
`previous_dts = datetime.strptime(self.__checkPayload__('timestamp', now.strftime(DATEFORMAT_TIMESTAMP)), DATEFORMAT_TIMESTAMP)`
TO:
`previous_dts = datetime.strptime(self.pd['timestamp'] ,DATEFORMAT_TIMESTAMP)`

Thanks.

Silvester5, 2022-02-21:
Thank you.
But the second problem is, that the timestamp from wmbusmeter is in UTC time and is in my case one hour back :(

`{
	"name": "Wasserzähler Haus",
	"device": "watermeter",
	"deviceid": "20001312",
	"date": "2022-02-21",
	"time": "19:03:03",
	"total_m3": 469.718,
	"ratio": 1.1,
	"m3": {
		"current": 0.161,
		"hour": 0.161,
		"day": 1.6,
		"month": 42.556,
		"year": 288.8
	},
	"liter": {
		"current": 161.0,
		"hour": 161.0,
		"day": 1600.0,
		"month": 42556.0,
		"year": 288800.0
	},
	"last_total": {
		"hour": "19",
		"hour_m3": 469.718,
		"day": "2022-02-20",
		"day_m3": 468.118,
		"month": "2021-12",
		"month_m3": 427.162,
		"year": "2021",
		"year_m3": 180.918
	},
	"wmbusmeter": {
		"media": "water",
		"meter": "izar",
		"name": "watermeter",
		"id": "20001312",
		"prefix": "",
		"serial_number": "000000",
		"total_m3": 469.718,
		"last_month_total_m3": 427.162,
		"last_month_measure_date": "2021-12-01",
		"remaining_battery_life_y": 3,
		"current_alarms": "no_alarm",
		"previous_alarms": "no_alarm",
		"transmit_period_s": 8,
		"manufacture_year": "0",
		"timestamp": "2022-02-21T19:03:03Z",
		"device": "cul",
		"rssi_dbm": -74
	},
	"alarm": "no_alarm",
	"last_alarm": "no_alarm",
	"periode": "2022-02-21",
	"month": "2022-02",
	"year": "2022",
	"timestamp": "2022-02-21T19:03:03Z",
	"last_update": "2022-02-21 20:03:03.278222",
	"elapsed_time": "0:38:20",
	"icon": "mdi:billboard",
	"unit_of_measurement": "㎥",
	"last_reset": "1970-01-01T00:00:00+00:00",
	"state_class": "measurement",
	"device_class": "energy",
	"version": "1.0.3",
	"attribution": "Data provided by Peter Siebler",
	"first_run": false,
	"data_provider": "thomas-NUC8i5BEH"
}
`

Thank you

zibous, 2022-02-22:
@Silvester5 

Yes, but all others are in local (system) date:

```
            self.cd['periode']=now.strftime(DATEFORMAT_DAY)
            self.cd['month']=now.strftime(DATEFORMAT_MONTH)
            self.cd['year']=now.strftime(DATEFORMAT_YEAR)
            self.cd['elapsed_time']=str(self.elapsed_time)
            self.cd['last_update']=now.strftime(DATEFORMAT_CURRENT)
```

Silvester5, 2022-02-22:
@zibous 

You are right, thank you, now it works fine :)

closed 2022-02-22 by Silvester5