zibous/ha-watermeter #9: Hass.io Home Assistant issues

Krotu, 2023-01-19:
I am having a few different issues trying to use this project, I am new to using github, sorry if having the issues together is incorrect usage.

I am using Hass.io Home assistant on a Raspberry Pi 3. I have a bunch of devices added to home assistant and it seems to be functioning normally.

I bought the blue dvb-t dongle pictured. Plugged into the PI, the pi can see it is plugged in but lsusb does not identify it:
Bus 001 Device 004: ID 0bda:2838 (nothing displayed after this)

dmesg shows:
Product: RTL2838UHIDIR
Manufacturer: Realtek

I managed to install the development stuff and rtl-sdr and ran into 2 further issues:
1) rtl_test could not find a device to use
2) all the development stuff and rtl-sdr vanish after a reboot.

Is there any further instructions on how to use a RTL2838 instead of the RTL2832 when using Hass.io version of Home assistant?

Thank you.

Krotu, 2023-01-19:
apparently the project has to run as an addon for both the hardware (usb) access and for the compiled code to persist through reboots.

zibous, 2023-01-20:
I only use Homeassistant as a Docker application on my NUC, so I don't know how it works with a HAS AddOn application. 
But the wmbusmeter with dvb-t dongle requires a lot of resources and therefore I only use it on my own Raspberry.

see: https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-with-rtl-sdr.md

There is some information on this topic at https://github.com/weetmuts/wmbusmeters/issues

Krotu, 2023-01-26:
I tried getting this working on a clean Raspbian Buster install but I am still hitting too many issues. I think I have the depenencies installed (the pi can see the usb dvb-t dongle now), but I can't figure out how to test rtl-wmbus or wmbusmeters.

Running app.py gives me `ERROR: An unhandled exception occured: error() missing 1 required positional argument: 'msg'.                                                               Traceback: [<FrameSummary file /opt/ha-watermeter/./app.py, line 169 in <module>>] (__main__: Line 91)`

zibous, 2023-01-26:
Hi,

> but I can't figure out how to test rtl-wmbus or wmbusmeters.
 If you have the `mqtt-explorer`, you can check whether it works.
see:  https://mqtt-explorer.com/

> watermeter/./app.py, line 169 in <module>>] (__main__: Line 91)
This error message comes when there is an error with the MQTT Brocker.
Or there is an MQTT topic, but no content - then the configuration at wmbusmeters would not be correct.
`missing msg` is comming, when no MQTT payload is present.

Krotu, 2023-01-28:
It turns out that https://github.com/weetmuts/wmbusmeters has a home assistant addon. I have everything working now using that, thanks for your time though.

closed 2023-01-28 by Krotu

zibous, 2023-02-02:
@Krotu 

simple 10 Euro solution:
https://github.com/zibous/ha-watermeter/tree/master/esphome