zibous/ha-watermeter #10: Diehl 171 meter - success with wmbusmeters

babajun12, 2023-02-26:
Has anyone been successful in receiving data from a Diehl 171 meter whith ESPhome? Serial output doesn´t show any meter values.
I was able to read data out of the meter with wmbusmeters on a Raspberry:

Auto driver  : hydrus
Best driver  : topaseskr 17/19
Using driver : hydrus 00/00
000   : 3e length (62 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 50570371 dll-id (71035750)
008   : 70 dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : eb tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 3005 tpl-cfg 0530 (AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
015   : 2f2f already decrypted check bytes

"media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"71035750",
    "flow_temperature_c":9.2,
    "status":"OK",
    "total_m3":11.377,
    "total_at_date_m3":5.264,
    "at_datetime":"2023-01-31 23:59",
    "remaining_battery_life_y":14.039987,
    "timestamp":"2023-02-26T11:52:44Z"

So I changed config.yaml, type to "hydrus". Does it load the right module then and if not, is it possible to include it?

sensor:

  - platform: wmbus
    name: "Wasseruhr Anzeige"
    id: "waterdisplay"
    meter_id: !secret watermeterId
    type: hydrus
    unit_of_measurement: "㎥"

Thanks,
Gerhard

zibous, 2023-02-26:
Hi
see: https://github.com/SzczepanLeon/esphome-components/tree/main/components/wmbus

`type (Required, string): 
    Meter type. Currently 
          amiplus, 
          apator08, 
          apator162, 
          bmeters, 
          elf, 
         evo868, 
         hydrocalm3, 
         izar, 
         mkradio3, 
         mkradio4, 
         ultrimis, 
         unismart, 
        fhkvdataiii are supported`

babajun12, 2023-02-26:
Ok, found that before.
Is there a chance to integrate something like driver_hydrus.h ?

zibous, 2023-02-27:
if you get telegrams  you can use "remote" wmbus (Using wmbusmeters i a pipe) metering with mqtt or HA addon.

```
wmbus:
  ## connction to CC1101
  mosi_pin: GPIO13 #D7  Attached to Hardware SPI controller MOSI SPI Interface
  miso_pin: GPIO12 #D6  Attached to Hardware SPI controller MISO SPI Interface
  clk_pin: GPIO14  #D5  Attached to Hardware SPI controller CLK
  cs_pin: GPIO15   #D8  Controls Boot Mode; Attached to Hardware SPI controller CS SPI Interface 10k Pull-Down, boot fails if pulled high !!!
  gdo0_pin: GPIO04 #D2  High Impedance
  gdo2_pin: GPIO05 #D1  High Impedance

  ## Component to receive wMBus frame (via CC1101) and send it via TCP or UDP to wmbusmeters (as HEX or rtl-wmbus format).
  ## On client side you can use netcat to receive packets: nc -lku 7337
  # Using wmbusmeters i a pipe
  clients:
    - name: "wmbusmeters"
      ip_address: !secret wmbusmeter_host
      port: 7227
      format: rtlwMBus
      transport: UDP
```

see: https://github.com/SzczepanLeon/esphome-components/blob/main/docs/wmbusgw.md

closed 2023-04-14 by zibous