zibous/ha-watermeter #11: IZAR RC 868 i W R4 compatibility

a-d-r-i-a-n-d, 2023-03-04:
[image]

Hi,
Thanks for your hard work. I live in UK and I wanted to check if this would work with the meter from the above picture. Any help is much appreciated.
Cheers.

zibous, 2023-03-05:
Yes i think it ist working, because it ist a 868 device.

urechi, 2023-03-05:
Thanks @zibous, I will give it a go.

a-d-r-i-a-n-d, 2023-03-10:
Hi @zibous ,

I've managed to flash it tonight but I get some errors:
```
wMBus-lib: Error during decoding '3 out of 6'
```
Full logs file here:
[logs.txt](https://github.com/zibous/ha-watermeter/files/10937556/logs.txt)

Any help is much appreciated.

a-d-r-i-a-n-d, 2023-03-10:
Hi @zibous ,

I've managed to flash it tonight but I get some errors:
```
wMBus-lib: Error during decoding '3 out of 6'
```
Full logs file here:
[logs.txt](https://github.com/zibous/ha-watermeter/files/10937600/logs.txt)

Any help is much appreciated.

zibous, 2023-03-10:
@a-d-r-i-a-n-d 
Sorry but i am on Holiday and will be back next week.

a-d-r-i-a-n-d, 2023-03-10:
No worries, enjoy your holiday. I’ll speak to you next week.

a-d-r-i-a-n-d, 2023-03-10:
I don't see that error anymore since I worked out my hex meter id and used the correct one in `secrets.yaml`
That being said I am now trying to understand the readings.

[image]

zibous, 2023-03-11:
Yes also in my case only the hex Meter id ist working. You have negative values, reset the globals or edit the inital valuest in the yaml config. Alternative you can Set the valuest with a homeassistant service call.

closed 2023-04-14 by zibous