zibous/ha-watermeter #12: Help with setting up the Program on the D1mini

compumark, 2023-04-14:
Hi,
I am new to ESPhome and found your project. I tried to follow the steps on your repository, but what do you mean with:
[image]

Where should I flash this? My ESPhome server is up and running and I copied the files into the docker directory. Maybe I missed there one step...

[image]

[image]

I also created the secrets file
[image]
The ID - I guess - is the Serialnumber of the watermeter DIEHL device, which is readable on the sticker? 
What is as host required? the IP from the esphome board?

Thanks a lot for any help to get this running :-)
best regards,
Mark

compumark, 2023-04-15:
Thank you for you reply. 
That helped me and I was able to get the ID and finish the setup. 
I found out that the watermeeter seems to send every 15 minutes a signal according to the manufacturer
https://www.diehl.com/metering/en/products-solutions/products-services/system-software/system-software-components/izar-rc-i-r4-at/#izar-rc-i-r4 
at least I am not getting readings every 8 seconds... 
Do you get threre a more precise reading?  

sometimes the reading seems to fail - see below highlighted - as a result a "nan" is sent 

[15:45:41][D][text_sensor:067]: 'watermeterdata': Sending state '439.566|0.014|0.043|0.205|0.205|0.205|0.000|0.205|2023-04-15T15:45:34'
[15:45:48][D][text_sensor:067]: 'Online seit': Sending state '6m 54s'
[15:45:48][D][sensor:126]: 'Uptime': Sending state 413.85800 s with 0 decimals of accuracy
[15:45:50][D][text_sensor:067]: 'Online seit': Sending state '6m 54s'
[15:45:51][D][sensor:126]: 'Wasser Tag': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:45:52][D][sensor:126]: 'Wasser Aktuell': Sending state 0.01401 m³ with 3 decimals of accuracy
[15:46:00][D][sensor:126]: 'Wasser WiFi Signal': Sending state -63.00000 dBm with 0 decimals of accuracy
[15:46:26][D][sensor:126]: 'Wasser letzer Monat': Sending state 0.00000 m³ with 3 decimals of accuracy
[15:46:29][D][sensor:126]: 'Wasser Woche': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:46:29][D][sensor:126]: 'Wasser Monat': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:46:30][D][sensor:126]: 'Wasser WiFi Signal': Sending state -64.00000 dBm with 0 decimals of accuracy
[15:46:32][D][sensor:126]: 'Wasser Stunde': Sending state 0.04300 m³ with 3 decimals of accuracy
[15:46:33][D][sensor:126]: 'Boot counter': Sending state 9.00000  with 0 decimals of accuracy
[15:46:34][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -35 dBm LQI: 128 T: 1944A511780772821621A161100013E118A40B421D11DF7F6164 (26)
[15:46:34][D][sensor:126]: 'Wasseruhr Anzeige': Sending state **439.57001** m³ with 3 decimals of accuracy
[15:46:34][I][main:323]:  -------  SET NEW VALUES !!!!!
[15:46:34][D][sensor:126]: 'Wasser Aktuell': Sending state 0.00400 m³ with 3 decimals of accuracy
[15:46:34][D][sensor:126]: 'Wasser Stunde': Sending state 0.04700 m³ with 3 decimals of accuracy
[15:46:34][D][sensor:126]: 'Wasser Tag': Sending state 0.20901 m³ with 3 decimals of accuracy
**[15:46:34][E][json:039]: Could not allocate memory for JSON document! Requested 488 bytes, largest free heap block: 488 bytes
[15:46:34][D][sensor:126]: 'Wasser Woche': Sending state 0.20901 m³ with 3 decimals of accuracy
[15:46:34][E][json:039]: Could not allocate memory for JSON document! Requested 184 bytes, largest free heap block: 184 bytes
WARNING 192.168.1.39: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for 192.168.1.39
WARNING Disconnected from API
INFO Successfully connected to 192.168.1.39**
[15:47:08][D][sensor:126]: 'Boot counter': Sending state 10.00000  with 0 decimals of accuracy
[15:47:09][D][sensor:126]: 'Wasser Jahr': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:47:11][D][sensor:126]: 'Wasser WiFi Signal': Sending state -63.00000 dBm with 0 decimals of accuracy
[15:47:12][D][text_sensor:067]: 'Online seit': Sending state '0s'
[15:47:13][D][sensor:126]: 'Wasser letzer Monat': Sending state 0.00000 m³ with 3 decimals of accuracy
[15:47:14][D][sensor:126]: 'Wasser Stunde': Sending state 0.04300 m³ with 3 decimals of accuracy
[15:47:17][D][sensor:126]: 'Wasser Aktuell': Sending state 0.01401 m³ with 3 decimals of accuracy
[15:47:17][D][esp8266.preferences:238]: Saving preferences to flash...
[15:47:22][D][sensor:126]: 'Wasser Monat': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:47:23][D][text_sensor:067]: 'Timestamp': Sending state '2023-04-15T15:47:22'
[15:47:24][D][sensor:126]: 'Wasser Woche': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:47:24][D][text_sensor:067]: 'watermeterdata': Sending state **nan|**0.014|0.043|0.205|0.205|0.205|0.000|0.205|2023-04-15T15:47:22
[15:47:24][D][sensor:126]: 'Wasser Tag': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:47:28][D][text_sensor:067]: 'Online seit': Sending state '54s'
[15:47:28][D][sensor:126]: 'Uptime': Sending state 53.62100 s with 0 decimals of accuracy
[15:47:41][D][sensor:126]: 'Wasser WiFi Signal': Sending state -65.00000 dBm with 0 decimals of accuracy
[15:48:08][D][sensor:126]: 'Boot counter': Sending state 10.00000  with 0 decimals of accuracy
[15:48:09][D][sensor:126]: 'Wasser Jahr': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:48:11][D][sensor:126]: 'Wasser WiFi Signal': Sending state -64.00000 dBm with 0 decimals of accuracy
[15:48:12][D][text_sensor:067]: 'Online seit': Sending state '54s'
[15:48:13][D][sensor:126]: 'Wasser letzer Monat': Sending state 0.00000 m³ with 3 decimals of accuracy
[15:48:14][D][sensor:126]: 'Wasser Stunde': Sending state 0.04300 m³ with 3 decimals of accuracy
[15:48:17][D][sensor:126]: 'Wasser Aktuell': Sending state 0.01401 m³ with 3 decimals of accuracy
[15:48:22][D][sensor:126]: 'Wasser Monat': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:48:23][D][text_sensor:067]: 'Timestamp': Sending state '2023-04-15T15:48:22'
[15:48:24][D][sensor:126]: 'Wasser Woche': Sending state 0.20502 m³ with 3 decimals of accuracy
[15:48:24][D][text_sensor:067]: 'watermeterdata': Sending state 'nan|0.014|0.043|0.205|0.205|0.205|0.000|0.205|2023-04-15T15:48:22'

and homeassistant gets an "unbekannt" status
[image]

Do you know why this is happening? 

Furthermore I also corrected this two entries to make it compatible with the HA energy card configruation

`unit_of_measurement: "m³" 
state_class: "measurement" geändert auf  state_class: "total"
`
best regards,
Mark

zibous, 2023-04-15:
hi Mark,

> [15:46:34][E][json:039]: Could not allocate memory for JSON document! Requested 488 bytes, largest free heap block: 488 bytes
[15:46:34][D][sensor:126]: 'Wasser Woche': Sending state 0.20901 m³ with 3 decimals of accuracy
[15:46:34][E][json:039]: Could not allocate memory for JSON document! Requested 184 bytes, largest free heap block: 184 bytes
WARNING 192.168.1.39: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for 192.168.1.39
WARNING Disconnected from API
INFO Successfully connected to 192.168.1.39

I think the problem is this error. 

Which version of ESPHome  do you use ?

you can try it w/o the script ` - id: post_device_state`, delete or comment the call in line

```yaml
      line 547:  id(post_device_state).execute();
```

compumark, 2023-04-15:
I am using this version https://registry.hub.docker.com/r/esphome/esphome/ running on my Synology Docker Container. 
[v2023.3.2]
I dont have a line 547 and also no content of this script. 
I am using this pre-release https://github.com/zibous/ha-watermeter/releases/tag/1.0.1 
in the yaml it says appversion: "1.1.1"

yes, both are indeed correct, but when set to "measurement" HA gave an error that the "last_reset" argument is missing, therefore I tried it with the "total" and HA didnt gave an error.

zibous, 2023-04-15:
> [v2023.3.2]
Ok, same Version here but on a NUC.

see: I dont have a line 547 and also no content of this script.

O.k the version `https://github.com/zibous/ha-watermeter/releases/tag/1.0.1` do not have this, only the current.
But the error comes from ESPHOME and the version 1.0.1 does not use JSON. So I can't explain that.

>HA gave an error that the "last_reset" argument is missing, therefore I tried it with the "total" and HA didnt gave an error.

I use Home Assistant 2023.4.4, Frontend 20230411.1 - latest. `state_class: "measurement"`is working.

compumark, 2023-04-15:
I found the 1.1.5 on your github repository  https://github.com/zibous/ha-watermeter/blob/f4bf78711a9e9b96fcbd940e071a1ddc8204c15e/esphome/wmbus-minid1.yaml 

There you have the state_class: total_increasing defined instead of the "measurement" from the 1.0.1
I just seen that I use Home Assistant 2023.3.0 Frontend 20230301.0 - latest - maybe they changed there as well, but I wanted to finish this integration before installing the April release. 

Is this the most actual version? can I use that as well - or do I have there other pre-requirements?

zibous, 2023-04-15:
> [15:47:24][D][text_sensor:067]: 'watermeterdata': Sending state **nan|**0.014|0.043|0.205|0.205|0.205|0.000|0.205|2023-04-15T15:47:22

-->  means, that the id(waterdisplay).state has no value.

but you can try to  disable the template.

```
- platform: template
    id: watermeterdata
    update_interval: ${update_interval}
```

> Is this the most actual version? can I use that as well - or do I have there other pre-requirements?

Yes but has only more features.

compumark, 2023-04-15:
I tried the v1.1.5 an excluded the "postresultdata to service url path"
The validation goes through, but the compiling stops with an error

```
INFO Reading configuration /config/wmbus-water_v1.1.5.yaml...
INFO Generating C++ source...

INFO Backup config will take: 3355 bytes
INFO Compiling app...
Processing water-meter-izar (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)

--------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/SzczepanLeon/wMbus-lib @ 0.9.14
INFO Installing git+https://github.com/SzczepanLeon/wMbus-lib @ 0.9.14
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-npdtlh4s'...
Error: Package version 0.9.15+sha.25dfb37 doesn't satisfy requirements 0.9.14 based on PackageMetaData <type=library name=wMbus-lib version=0.9.15+sha.25dfb37 spec={'owner': None, 'id': None, 'name': 'wMbus-lib', 'requirements': '0.9.14', 'uri': 'git+https://github.com/SzczepanLeon/wMbus-lib'}
```

must be some dependencies. I downloaded the full directory https://github.com/zibous/ha-watermeter/tree/master/esphome 
and extracted it to the esphome 
I need to investigate there tomorrow

zibous, 2023-04-16:
@compumark 

Sorry..

The new version should work.
see new version: https://github.com/zibous/ha-watermeter/blob/master/esphome/wmbus-minid2.yaml
and  https://github.com/zibous/ha-watermeter/blob/master/esphome/README.md

compumark, 2023-04-16:
Hi, no problem :)
This one is working now. I added under the section # reset all global vars the "id(current_value) = 0.00;" as this was missing, what I noticed. Dont sure if needed, but yesterday when playing around it was helpful.

Do you really get every 8 seconds a signal from the water meter? 
Mine are coming in very strange Intervalls . Is the programm always listening to the frequency or is it polling actively and by luck it captures a transmission? 

[13:10:15][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -36 dBm LQI: 128 T: 1944A511780772821621A151100013E1EDD6D1BF7F65B5124A63 (26)
[13:10:15][D][sensor:126]: 'Wasseruhr Anzeige': Sending state 440.03000 m³ with 3 decimals of accuracy
[13:10:15][I][main:436]: Water Display value: 440.029999, last value: 0.000000
[13:10:15][D][sensor:126]: 'Wasser Aktuell': Sending state 0.00000 L with 2 decimals of accuracy
[13:10:15][I][main:462]: Reset current value to: 0.000000
[13:10:15][D][text_sensor:067]: 'Zählerstand aktualisiert': Sending state '2023-04-16T13:10:15'

The next reading was 9 minutes later from the neighbors meter
[13:19:11][D][wmbus:158]: Meter ID [0x82670778] RSSI: -78 dBm LQI: 129 not found in configuration T: 1944A511780767821621A111100013F22528AF483F65E310FC5D (26)

the next was again the neighbours
[13:25:24][D][wmbus:158]: Meter ID [0x82670778] RSSI: -79 dBm LQI: 145 not found in configuration T: 1944A511780767821621A151100013F2799589E4B82A847C9857 (26)
[13:25:24][D][sensor:126]: 'Wasser Gestern': Sending state 0.00000 L with 2 decimals of accuracy

18 minutes later I got my next reading.
[13:28:04][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -36 dBm LQI: 128 T: 1944A511780772821621A131100013E1B9356445BB8D61C81C6C (26)
[13:28:04][D][sensor:126]: 'Wasseruhr Anzeige': Sending state 440.05600 m³ with 3 decimals of accuracy
[13:28:04][I][main:436]: Water Display value: 440.056000, last value: 440.029999
[13:28:04][D][sensor:126]: 'Wasser Aktuell': Sending state 26.00098 L with 2 decimals of accuracy
[13:28:04][D][sensor:126]: 'Wasser Stunde': Sending state 26.00098 L with 2 decimals of accuracy
[13:28:04][D][sensor:126]: 'Wasser Tag': Sending state 26.00098 L with 2 decimals of accuracy
[13:28:04][D][sensor:126]: 'Wasser Woche': Sending state 26.00098 L with 2 decimals of accuracy
[13:28:04][D][sensor:126]: 'Wasser Monat': Sending state 26.00098 L with 2 decimals of accuracy
[13:28:04][D][sensor:126]: 'Wasser Jahr': Sending state 26.00098 L with 2 decimals of accuracy
[13:28:04][I][main:457]: Set current value to 0.000000 litre and publish the data
[13:28:04][D][text_sensor:067]: 'Zählerstand aktualisiert': Sending state '2023-04-16T13:28:04'

Than again 2 minutes later
[13:30:02][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -36 dBm LQI: 128 T: 1944A511780772821621A111100013E1956BF713F82AD27E2E69 (26)
[13:30:02][D][sensor:126]: 'Wasseruhr Anzeige': Sending state 440.05801 m³ with 3 decimals of accuracy
[13:30:02][I][main:436]: Water Display value: 440.058014, last value: 440.056000
[13:30:02][D][sensor:126]: 'Wasser Aktuell': Sending state 2.01416 L with 2 decimals of accuracy
[13:30:02][D][sensor:126]: 'Wasser Stunde': Sending state 28.01514 L with 2 decimals of accuracy
[13:30:02][D][sensor:126]: 'Wasser Tag': Sending state 28.01514 L with 2 decimals of accuracy
[13:30:02][D][sensor:126]: 'Wasser Woche': Sending state 28.01514 L with 2 decimals of accuracy
[13:30:02][D][sensor:126]: 'Wasser Monat': Sending state 28.01514 L with 2 decimals of accuracy
[13:30:02][D][sensor:126]: 'Wasser Jahr': Sending state 28.01514 L with 2 decimals of accuracy
[13:30:02][I][main:457]: Set current value to 0.000000 litre and publish the data

[13:32:27][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -38 dBm LQI: 128 T: 1944A511780772821621A121100013E1AC1A2DEE9A5EB813056E (26)
[13:32:27][D][sensor:126]: 'Wasseruhr Anzeige': Sending state 440.05801 m³ with 3 decimals of accuracy
[13:32:27][I][main:436]: Water Display value: 440.058014, last value: 440.058014
[13:32:27][D][sensor:126]: 'Wasser Aktuell': Sending state 0.00000 L with 2 decimals of accuracy
[13:32:27][I][main:462]: Reset current value to: 0.000000

and then again since 10 minutes, nothing... 

Can this be an antenna specific issue?

compumark, 2023-04-16:
das nächste war dann am
[13:49:15][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -47 dBm LQI: 128 T: 1944A511780772821621A111100013E1956BF713F82AD27E2E69 (26)

und 
[13:51:13][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -46 dBm LQI: 128 T: 1944A511780772821621A171100013E1E78842E93CC206A47866 (26)
[13:51:13][D][sensor:126]: 'Wasseruhr Anzeige': Sending state 440.05801 m³ with 3 decimals of accuracy
[13:51:13][I][main:436]: Water Display value: 440.058014, last value: 440.058014

und dann wieder nichts - also sehr komisch
oh sorry, for switching to german - just noticed ;-)

zibous, 2023-04-16:
> [13:51:13][I][wmbus:071]: Using driver 'izar' for ID [0x82720778] RSSI: -46 dBm LQI: 128 T:

It could also be because the Wemos D1 Mini and CC1101 is too far from the water meter. 
The signal on the CC1101 is not exactly high and fluctuates very strongly. 
I would try to bring the D1MINI closer to the water meter.

- RSSI: -36 dBm LQI: 128
- RSSI: -38 dBm LQI: 128 
- RSSI: -46 dBm LQI: 128 
- RSSI: -47 dBm LQI: 128

Here is the log from my test device:
```log
17:51:38 | [D] | [sensor:109] | 'Wlan Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
17:51:38 | [D] | [sensor:109] | 'Wlan Qualität': Sending state 82.00000 Signal % with 0 decimals of accuracy
17:51:38 | [D] | [sensor:109] | 'Wasser Gestern': Sending state 108.00171 L with 2 decimals of accuracy
17:51:42 | [D] | [sensor:109] | 'Wasser Woche': Sending state 309.02100 L with 2 decimals of accuracy
17:51:46 | [D] | [text_sensor:064] | 'Timestamp': Sending state '2023-04-16T17:51:45'
17:51:46 | [D] | [sntp:077] | Synchronized time: 2023-04-16 17:51:46
17:51:46 | [D] | [main:373] | Synchronized sntp clock
17:51:46 | [D] | [sensor:109] | 'Wasser Monat': Sending state 309.02100 L with 2 decimals of accuracy
17:51:46 | [I] | [wmbus:071] | Using driver 'izar' for ID [0x43430778] RSSI: -61 dBm LQI: 128 T: 1944A511780743434418A251150013CEBDB57AE7EACE9035EFCC (26)
17:51:46 | [D] | [sensor:109] | 'Wasseruhr Anzeige': Sending state 466.92001 m³ with 3 decimals of accuracy
17:51:46 | [I] | [main:434] | Water Display value: 466.920013, last value: 466.920013
17:51:46 | [D] | [sensor:109] | 'Wasser Aktuell': Sending state 0.00000 L with 2 decimals of accuracy
17:51:46 | [I] | [main:460] | Reset current value to: 0.000000
17:51:46 | [D] | [text_sensor:064] | 'Zählerstand aktualisiert': Sending state '2023-04-16T17:51:46'
17:51:50 | [D] | [sensor:109] | 'Boot counter': Sending state 0.00000  with 0 decimals of accuracy
17:51:50 | [D] | [wmbus:158] | Meter ID [0x43410778] RSSI: -88 dBm LQI: 131 not found in configuration T: 1944A511780741434418A251150013CCF3214B8353FFAF56CF9A (26)
17:51:59 | [D] | [wmbus:158] | Meter ID [0x43410778] RSSI: -87 dBm LQI: 136 not found in configuration T: 1944A511780741434418A261150013CCCA50917E318BC53BE49D (26)
17:52:04 | [I] | [wmbus:071] | Using driver 'izar' for ID [0x43430778] RSSI: -61 dBm LQI: 128 T: 1944A511780743434418A271150013CE93EBE9B1A9692383DDC9 (26)
17:52:04 | [D] | [sensor:109] | 'Wasseruhr Anzeige': Sending state 466.92001 m³ with 3 decimals of accuracy
17:52:04 | [I] | [main:434] | Water Display value: 466.920013, last value: 466.920013
17:52:04 | [D] | [sensor:109] | 'Wasser Aktuell': Sending state 0.00000 L with 2 decimals of accuracy
17:52:04 | [I] | [main:460] | Reset current value to: 0.000000
```

compumark, 2023-04-16:
Interesting.  My device is roughly 3m away from the meter. But seeing your dB counts shows that yours are worse than mine with the -61db in comparison to mine with -40 range. I will move it nearby to the meter with a longer power cable and will monitor this.
But maybe it was really configured from the water department of our city?

zibous, 2023-04-16:
> Interesting. My device is roughly 3m away from the meter

Then you have to have a source of interference ...
My Watermeter is 10m away and in another room.

> I will move it nearby to the meter with a longer power cable...

Ohh it could also be the power supply, I had such an effect on another device.

compumark, 2023-04-16:
Tried now 3 different power supplies and 3 different usb cables, all the same effect. No readings since 20 minutes. 
What I also noticed during building, that the card was not fine cut and the 3 holes where to wire the antenna in the middle are only existing as half. So I did solder another cable in-between the card and the antenna to fit the antenna. I already contacted Amazon and I will get tomorrow a new one. Maybe this longer antenna has also negative impact  on the setup.

zibous, 2023-04-16:
> So I did solder another cable in-between the card and the antenna to fit the antenna. I already contacted Amazon and I will get tomorrow a new one. Maybe this longer antenna has also negative impact on the setup.

But that would explain a few things. If you set the log level to VERBOSE, you would have to see more.

ok. i use 2 to theese on: https://amzn.eu/d/i5YwBkR
No Problems.

compumark, 2023-04-17:
Update from my side. I received the new antenna board, same crappy cut as the old one.

[image]

I wired it in the evening, but far away from the 8seconds reading. 

here the whole day:
[image]

and since the board change:
[image]

I found the same boards on ebay: https://www.ebay.at/itm/233689592394?var=533375579177
maybe I risk to order 2 of them, maybe there are another batch and do not have the wrong cut

zibous, 2023-04-18:
How close is the CC1101 to the D1MINI ? 

It is strange that it works almost all the time and only sometimes fails. 

If you use the neighbor's water meter (he has a much worse signal) it is the same ?

compumark, 2023-04-18:
The wires between the 2 boards are around 8cm length.
I can try to shorten them as well in the evening. 
I need to check the log files for the neighbors meter intervals, will do that as well in the evening.

zibous, 2023-04-19:
> I found the same boards on ebay: https://www.ebay.at/itm/233689592394?var=533375579177

Better here: https://schlauhaus.biz/produkt/cc1101-funkmodul/
The contact is in Germany

zibous, 2023-04-28:
@compumark 

I can now see the error with me too, but only with the second D1MINI. 

- Both are wired indent and have the same hardware and 
from the same supplier and same order.

- One works fine and the other always boots with a JSON error. 

As it looks it could also be due to the D1Mini.
In the variant with an ESP32, the version works perfectly.

compumark, 2023-04-28:
Hi @zibous ,
thanks for the follow up - I have not received the new antenna boards yet, but I will try in the next days the setup in a different house and see if i get there more intervals. 

The error you mentioned now is the JSON error which appears in the debug log sometimes or the issue that you are not getting the readings every 8s?

Whats the ESP32 variant? is this another board?

zibous, 2023-05-06:
Hi,

I found and bought an adapter board for the CC1101. Cost $2.85 (no shipping). 
This certainly makes it easier to connect the CC1101.

[image]

I'm curious how long the delivery will take.
see: <https://oshpark.com/shared_projects/dVZXCMXP>

SzczepanLeon, 2023-05-06:
Soldering that CC1101 module is pain. That's why I have my own board for D1 mini.
[image]

zibous, 2023-05-06:
@SzczepanLeon 

Confirm, I already destroyed one, you can only solder once...

? That's why I have my own board for D1 mini.
Upload has not work, can't see your board.

Do you sell those too?

SzczepanLeon, 2023-05-06:
Yes, but I'm waiting for PCBs. Should be around 3 days. Blue from picture is my v1, now will be v2 (black and with small diferences), next v3 will be with pads for SMA antenna connector (and of course in another color ;) )

zibous, 2023-05-06:
@SzczepanLeon 

Perfect looks really good. 

How can i order 2 of them and what would that cost me?

SzczepanLeon, 2023-05-06:
I will know details around Tuesday.

compumark, 2023-05-14:
@zibous  I received my 2 new C1101 cards, but they are even worse cut :-( 
But I will try anyway to solder them.

When trying to compile the working script again I now receive errors: 
Anyone an idea why? 
The yaml is the same I compiled weeks ago without an error, but also this is not working to compile anymore.

Seems that something was changed on the esphome components
https://github.com/SzczepanLeon/esphome-components

But I have current no clue what and how to use an older library of the esphome components, that worked... 
any help is appreciated. :-) 

INFO Reading configuration /config/water-meter.yaml...
Failed config

sensor.wmbus: [source /config/water-meter.yaml:423]
  platform: wmbus
  
  [name] is an invalid option for [sensor.wmbus]. Please check the indentation.
  name: Wasseruhr Anzeige
  
  [id] is an invalid option for [sensor.wmbus]. Please check the indentation.
  id: waterdisplay
  meter_id: 0
  type: izar
  
  [unit_of_measurement] is an invalid option for [sensor.wmbus]. Please check the indentation.
  unit_of_measurement: m³
  
  [state_class] is an invalid option for [sensor.wmbus]. Please check the indentation.
  state_class: total_increasing
  
  [device_class] is an invalid option for [sensor.wmbus]. Please check the indentation.
  device_class: water
  
  [accuracy_decimals] is an invalid option for [sensor.wmbus]. Please check the indentation.
  accuracy_decimals: 3
  
  [on_value] is an invalid option for [sensor.wmbus]. Please check the indentation.

SzczepanLeon, 2023-05-14:
https://github.com/SzczepanLeon/esphome-components/issues/23#issuecomment-1531269396

zibous, 2023-05-14:
yes has changed:

compumark, 2023-05-14:
Thanks @SzczepanLeon and @zibous - that worked for me. 

I am now testing the 2.07 version - nice improvements and very well documented - congrats to that.
When I am flashing my productive watermeter with a connection to HA, I assume all my values are gone. 
I assume changing the global variables section to my last known values like "yesterday_value, week, month and year" should be the way to go to continue with the measurement. Am I correct?

SzczepanLeon, 2023-05-14:
> For me, however, the D1Mini are critical because they have little memory. ESP32 works better.

There is D1 mini ESP32 also:
[image]

zibous, 2023-05-14:
> I assume changing the global variables section to my last known values like "yesterday_value, week, month and year" should be the way to go to continue with the measurement. Am I correct?

- No, the Homeassistant history data for a sensor is only based on the `name` by the esphome configuration:
- global variables are only used internal by ESPHOME

Example:  `name: "Wasseruhr Anzeige"`---> Homeassistant `sensor.wasseruhr_anzeige`

```
total_water_m3:
      id: "waterdisplay"
      name: "Wasseruhr Anzeige"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
```

zibous, 2023-05-14:
@SzczepanLeon 

> There is also D1 mini ESP32:

Yes this one has  `ROM: 448 KB - SRAM: 520 KB`,  D1 mini ESP8266 80MHz only 80KB RAM.

compumark, 2023-05-14:
> * No, the Homeassistant history data for a sensor is only based on the `name` by the esphome configuration:
> * global variables are only used internal by ESPHOME
>
> Example: `name: "Wasseruhr [...]

But is then there an option to flash and continue with the same data?
eg: Thats my productive D1Mini
[image]

and this now my testing device with the 2.07 version
[image]

zibous, 2023-05-14:
Yes you can try with homeassistant service call : `esphome.water_meter_set_water_val`:

```
service: esphome.water_meter_set_water_val
data:
 water_val_hour: 83.98
 water_val_day: 414.00
 water_val_yesterday: 181.00
 water_val_week: 4479.98
 water_val_year: 18701.99
 water_val_lastmonth: 448.14
```

This will update the global vars.

LouiS22, 2023-06-30:
> Yes, but I'm waiting for PCBs. Should be around 3 days. Blue from picture is my v1, now will be v2 (black and with small diferences), next v3 will be with pads for SMA antenna connector (and of course in another color ;) )
> 
> [image]

Hello, is there webshop, link or any way to purchase this neat board? :)

SzczepanLeon, 2023-06-30:
> @SzczepanLeon Hello, is there webshop, link or any way to purchase this neat board? :)

I have one blue version without CC1101, and 5 black:
https://www.olx.pl/d/oferta/pcb-do-cc1101-adapter-wmbus-CID99-IDUYrwA.html

You can also order PCBs from:
https://www.pcbway.com/project/shareproject/CC1101_to_ESP_D1_mini_277f34e1.html

LouiS22, 2023-07-04:
> I have one blue version without CC1101, and 5 black: https://www.olx.pl/d/oferta/pcb-do-cc1101-adapter-wmbus-CID99-IDUYrwA.html
>
> You can also order PCBs from: [...]

@SzczepanLeon thanks for the reply. Quick question: do you happen to have a spare CC1101? :) Of course I will pay for that, too. Could you please drop me a PM to discuss further? Don't want to hijack this topic.

closed 2023-07-04 by zibous