zibous/ha-watermeter #13: Any chance to add support for Integra Topas ES KR?

sanchosk, 2023-04-21:
Hi.

This looks amazing, seem to support quite a lot of products already.
Any chance there is a possibility to add one more?
I have Integra Topas ES KR, which seems to support OMS standard (https://ch.integra-metering.com/product/topas-eskr/).
I would like to help to develop the support, just not sure what devices/HW do I need and how can I support the developers.

zibous, 2023-04-21:
@sanchosk 

Should work with radio 868 MHz according to OMS standard and there is a driver at wmbusmeters.
see: https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_topaseskr.cc

You can ask Szczepan Leon if it is possible to create the driver.
see: https://github.com/SzczepanLeon/esphome-components

sanchosk, 2023-04-28:
Hi again. The radio arrived, I've put together quickly the device, but I seem to have a problem.
The meterID gets detected, but afterwards I get message that it "can't get value from telegram ID [0x<myWaterMeterID>] 'izar'
The problem, I guess, is the 'izar' part, as my meter uses the wmbus protocol type 'topaseskr'.
I tried to change the type in the sensor -> type, but it does not seem to work.
During boot, the esp reports available drivers, but topaseskr is not listed:
`Apr 28 21:14:37 water-meter-esp wmbus ﻿[C][wmbus:342]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, elf, evo868, fhkvdataiii, hydrocalm3, itron, izar, mkradio3, mkradio4, qheat, qwater, ultrimis, unismart, vario451`

Any idea what am I doing wrong, plz?

zibous, 2023-04-29:
hi,

> Any idea what am I doing wrong, plz?

- You can ask Szczepan Leon if it is possible to create the driver or ask him for a workaround.

   see: https://github.com/SzczepanLeon/esphome-components

 - To test your's use the simple version (`clean build files` must be executed before using):
 
   https://github.com/zibous/ha-watermeter/blob/master/esphome/wmbus-minid2.yaml
   
   This one use the wmbus components from Szczepan, not the local one.

zibous, 2023-04-29:
hi,

Szczepan Leon makes a good job, in Version 2.0 this is allready present.

see: https://github.com/SzczepanLeon/esphome-components/blob/main/components/wmbus/driver_topaseskr.h

QAD - try this
```
## ------------------------------------------------------------------
## WMBUS METER D1 MINI + CC1101 for Diehl IZAR RC 868 I R4 PL
##
## usage:  1. Using Homassistant API (discovery + devices)
##         2. External custom component from SzczepanLeon
##
## ------------------------------------------------------------------
substitutions:
  device_name_short: "water-meter-izar"
  device_description: "Wasserzähler Wemos D1 Min, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL) simple SzczepanLeon"

  update_interval: 60s
  log_level: "VERBOSE" # use VERBOSE mode to find the meterId 0x43430778
  appversion: "1.1.7"

  domain: !secret domain

esphome:
  name: ${device_name_short}
  # HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
  platform: ESP8266
  board: d1_mini
  esp8266_restore_from_flash: true
  arduino_version: recommended
  comment: ${device_description}
  project:
    name: "OE9psj.wmbus-minid2"
    version: ${appversion}
  build_path: ./build/${device_name_short}

  ## ----------------------------------------------------------------
  ## External components for wmbusmeter
  ## ----------------------------------------------------------------
  on_boot:
    # At this priority, pretty much everything should already be initialized.
    priority: -100.0
    then:
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"

  on_shutdown:
    priority: 600
    then:
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} is down!"

## ----------------------------------------------------------------
## External components for wmbusmeter
## ----------------------------------------------------------------
external_components:
  # use exterban component 2.0 from
  # https://github.com/SzczepanLeon/esphome-components
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]
[... 629 lines cut ...]
    internal: true
    icon: mdi:clock-start
    entity_category: "diagnostic"

  # waterdisplay last update
  - platform: template
    name: Zählerstand aktualisiert
    id: watermeter_lastupdate
    icon: mdi:clock-start
    entity_category: "diagnostic"

  # device timestamp
  # disabled for api (memory ??)
  - platform: template
    name: Timestamp
    id: systime
    internal: true
    entity_category: "diagnostic"
    lambda: char str[20];
      time_t currTime = id(sntp_time).now().timestamp;
      strftime(str, sizeof(str), "%Y-%m-%dT%H:%M:%S", localtime(&currTime));
      return (std::string) str;
# eof water-meter-izar.yaml

```

sanchosk, 2023-04-29:
I feel like absolute failure here.
Trying to make your code work with ESP32...
I did try to combine it with the code you provided in your repo, but I am failing miserably.
Below is the code, this is the error:
`INFO Reading configuration /config/esphome/watermeter.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block mapping
  in "/config/esphome/watermeter.yaml", line 481, column 5:
      - platform: wmbus
        ^
expected <block end>, but found '<block mapping start>'
  in "/config/esphome/watermeter.yaml", line 491, column 6:
         rssi:
         ^
`
I copy-pasted the whole section around wmbus again and again, nothing seems to be helping :(

`## ------------------------------------------------------------------
## WMBUS METER D1 MINI + CC1101 for Diehl IZAR RC 868 I R4 PL
##
## usage:  1. Using Homassistant API (discovery + devices)
##         2. External custom component from SzczepanLeon
##
## ------------------------------------------------------------------
substitutions:
  device_name_short: "water-meter-izar"
  device_description: "Wasserzähler Wemos D1 Min, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL) simple SzczepanLeon"

  update_interval: 60s
  log_level: "VERBOSE" # use VERBOSE mode to find the meterId 0x43430778
  appversion: "1.1.7"

  domain: !secret domain

esphome:
  name: ${device_name_short}
  comment: ${device_description}
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}

  ## ----------------------------------------------------------------
  ## External components for wmbusmeter
  ## ----------------------------------------------------------------
  on_boot:
    # At this priority, pretty much everything should already be initialized.
    priority: -100.0
    then:
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"

  on_shutdown:
    priority: 600
    then:
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} is down!"

## ----------------------------------------------------------------
## External components for wmbusmeter
## ----------------------------------------------------------------
external_components:
[... 655 lines cut ...]
    name: Online seit
    id: uptime_human
    internal: true
    icon: mdi:clock-start
    entity_category: "diagnostic"

  # waterdisplay last update
  - platform: template
    name: Zählerstand aktualisiert
    id: watermeter_lastupdate
    icon: mdi:clock-start
    entity_category: "diagnostic"

  # device timestamp
  # disabled for api (memory ??)
  - platform: template
    name: Timestamp
    id: systime
    internal: true
    entity_category: "diagnostic"
    lambda: char str[20];
      time_t currTime = id(sntp_time).now().timestamp;
      strftime(str, sizeof(str), "%Y-%m-%dT%H:%M:%S", localtime(&currTime));
      return (std::string) str;
# eof water-meter-izar.yaml`

zibous, 2023-04-30:
@sanchosk 

Sorry, my mistake. You can try this version:

```
## ------------------------------------------------------------------
## WMBUS METER D1 MINI + CC1101 for Diehl IZAR RC 868 I R4 PL
##
## this version shows all watermeters. Use this for the first test
##
## Remark:
##   Some Wemos D1 Mini has memory problems, for me one is working
##   another one (same order) reboots with JSON Errors.
##   ESP8266-12F Mini NodeMcu mit ESP8266-12F
## ------------------------------------------------------------------
substitutions:
  device_name_short: "water-meter-test"
  device_description: "Wasserzähler Wemos D1 Min, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) - all watermeters"
  appversion: "2.0.3"
  domain: !secret domain
  update_interval: 60s 
  # use VERBOSE mode to find the meterId and shows all informations for the wbmus
  log_level: "VERBOSE" 

# ----------------------------------------------------------------
# HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
# ----------------------------------------------------------------  
esp8266:
  board: d1_mini
  restore_from_flash: true

esphome:
  name: ${device_name_short}
  comment: ${device_description}
  project:
    name: "OE9psj.wmbus-minid1"
    version: ${appversion}
  build_path: ./build/${device_name_short}
  on_boot:
    # At this priority, pretty much everything should already be initialized.
    priority: -100.0
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
  on_shutdown:
    priority: 700
    then:
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} is down!"

## ----------------------------------------------------------------
## External components
## ----------------------------------------------------------------
external_components:
  # uses the latest version from SzczepanLeon
  # https://github.com/SzczepanLeon/esphome-components
  # You can make ESPHome check the repository every time by setting this option to 0s
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

## ----------------------------------------------------------------
## Global variables
[... 172 lines cut ...]
    id: wifi_signal_db
    update_interval: 60s
    internal: true
    entity_category: "diagnostic"

  # Wifi quality RSSI in percentage
  # # Received Signal Strength (RSSI) is a measure of incoherent (raw) RF power in a channel.
  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "Wlan ESP32 Qualität"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"

  # CC1101 quality RSSI in percentage
  # Received Signal Strength (RSSI) is a measure of incoherent (raw) RF power in a channel.
  - platform: copy
    source_id: wmbus_cc1101_rssi
    name: "Wlan CC1101 RSSI"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
```

zibous, 2023-04-30:
solved, see: 
https://github.com/SzczepanLeon/esphome-components/issues/18

closed 2023-04-30 by zibous

sanchosk, 2023-04-30:
@zibous Do you have some "thank you coffee" option?

zibous, 2023-04-30:
@sanchosk 

> Do you have some "thank you coffee" option?

Thanks yes, i like and need coffe...
https://www.buymeacoffee.com/zibous