zibous/ha-watermeter #14: What could be wrong ?

Freeman656, 2023-04-29:
Hi,
I've successfully installed and wired the Wemos D1 Mini + CC1101 on esphome and the wifi communication gives me some logs.
No error visible.
But I don't get any communication from my water sensor (izar rc 868 i r4, not the PL one, does it make a difference ?).
The water sensor is underground about 50cm, and the D1mini about 20m from it in the house (the closest it can be with wifi connected).
This communication is supposed to be 500m and/or 1500m range, so I guess it's fine even underground ?

Moreover, in the documentation of the wiring (https://github.com/zibous/ha-watermeter/blob/master/esphome/watermeter_d1mini.md), there is an inconsistency for the GD02 -> D1 and GD00 -> D2 which is not the same in the picture wiring (GD02 -> D2 and GD00 -> D1), so which one is right ? I even tried both without success.

I there any way of checking if the CC1101 is working fine ?

Thanks for your help.
Cédric

zibous, 2023-04-29:
> Moreover, in the documentation of the wiring (https://github.com/zibous/ha-watermeter/blob/master/esphome/watermeter_d1mini.md), there is an inconsistency for the GD02 -> D1 and GD00 -> D2 which is [...]

ohhh thanks, the wiring picture is wrong.

```
 gdo0_pin: GPIO04 #D2  High Impedance
 gdo2_pin: GPIO05 #D1  High Impedance
```

> Is there any way of checking if the CC1101 is working fine ?

- use watermeterID: 0
- set loglevel to DEBUG or VERBOSE

> Note: I have 2 D1MINI and they are wired indent and have the same hardware and from the same supplier and same order. One works fine and the other always boots with a JSON error. The ESP32 variant works w/o theese errors.

zibous, 2023-04-30:
Hi,

I did some tests and found that only the WEMOS D1 works for me. 

If you don't get any data, but the CC1101 is recognized correctly, it may be due to the gdo0_pin/gdo2_pin wiring. 

If they are swapped, then no values come. 

Since the Wemos D1 Mini  is critical, I would recommend the ESP32 variant.

see: <https://github.com/zibous/ha-watermeter#wemos-d1mini--vs-esp32-v4>

Freeman656, 2023-05-01:
I don't have any JSON error with the D1 mini, but I'm using the D1 mini v4.0.0 (new style/component, and more memory). So I guess it should be OK.
I also tested on NodeMCUv3 which give me the same results, no data coming from the water meter.
Maybe I'm too far away from the water meter, I don't know, I'll try to next to it and let you know.

zibous, 2023-05-01:
Hi
> I don't have any JSON error with the D1 mini, but I'm using the D1 mini v4.0.0 (new style/component, and more memory).

this one : https://opencircuit.shop/product/wemos-d1-mini-v4.0-wifi-module

w/o Data you can'nt see the JSON ERROR.
Do you use the latest D1Mini Test YAML ?
https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-d1mini-test.yaml

Can you post your Logmessages with loglevel "DEBUG" or "VERBOSE".

Freeman656, 2023-05-01:
Yes, the D1 mini is this one, but a chinese copy from ali I think...
I'm using the yaml from a few days ago, not this one.
Please find the logs attached.

[logs_wmbus-minid1_logs.txt](https://github.com/zibous/ha-watermeter/files/11364007/logs_wmbus-minid1_logs.txt)

zibous, 2023-05-01:
OK.

1. Please use the latest wm-d1mini-test.yaml:
   Because your installation is old: `'project_version = 1.1.7`
   https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-d1mini-test.yaml
   this one uses the latest wbmus component from github

3. Before install make a  ' Clean Build files' to clear the old build.

     [image]

    Because your installation uses the old wbusmeter component.
    ```log
        [08:53:43][C][wmbus:312]: wM-Bus v1.4.8:`
        [08:53:43][C][wmbus:342]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, elf, evo868, fhkvdataiii, hydrocalm3, itron, izar, mkradio3, mkradio4, qheat, qwater, ultrimis, unismart, vario451
      NEW:  --> [09:06:41][C][wmbus:314]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, elf, evo868, fhkvdataiii, hydrocalm3, itron, izar, mkradio3, mkradio4, qheat, qwater, sharky774, topaseskr, ultrimis, unismart, vario451
    ```

4. Check your CC1101 connection, be shure that:

    ```
    GDO0 Pin: GPIO5 --> connected to gdo0_pin (CC1101)
    GDO2 Pin: GPIO4  --> connected to gdo2_pin (CC1101)
    ```
    If they are wrong (swapped), then no values come.

Freeman656, 2023-05-01:
All done. Keep in mind, I'm still using the ModeMCUv3 right now, that's why the board is on esp01_1m.
Still the same result with the following wiring :

mosi_pin: GPIO13
miso_pin: GPIO12
clk_pin: GPIO14
cs_pin: GPIO15
gdo0_pin: GPIO05
gdo2_pin: GPIO04

[image]

new logs :
[logs_water-meter-izar-nodemcu_logs.txt](https://github.com/zibous/ha-watermeter/files/11364273/logs_water-meter-izar-nodemcu_logs.txt)

zibous, 2023-05-01:
HI,

I installed a trial version on a NodeMCU V3  without C1101:
> **board: nodemcuv2** instead of board board = esp01_1m**

```
Processing water-meter-nodemcu (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@3.2.0)

11:19:26][C][wmbus:284]: wM-Bus v2.0.5:
[11:19:26][C][wmbus:301]:   CC1101 SPI bus:
[11:19:26][C][wmbus:302]:     MOSI Pin: GPIO13
[11:19:26][C][wmbus:303]:     MISO Pin: GPIO12
[11:19:26][C][wmbus:304]:     CLK Pin:  GPIO14
[11:19:26][C][wmbus:305]:     CS Pin:   GPIO15
[11:19:26][C][wmbus:306]:     GDO0 Pin: GPIO5
[11:19:26][C][wmbus:307]:     GDO2 Pin: GPIO4
[11:19:26][E][wmbus:320]:   Check connection to CC1101!  <-- ERROR !
```

[image]

If the C1101 had no voltage or the other connections were not available, then the error message would appear:
`Check connection to CC1101! `. You can test it by simply removing wires

>The water sensor is underground about 50cm, and the D1mini about 20m from it in the house 

Or everything works perfectly, only the distance to the water meter is too great. For me it's 6m in the basement and thick concrete walls in between. I know that there are several water meters at my position, but I only get 2 more that are in the immediate vicinity (8 - 10m). I can't get the others that are further away.

Freeman656, 2023-05-01:
OK, so I tried removing the supply wires and I get the 'Check connection' error in the logs, so it seems that the CC1101 is recognized when everything is plugged into correctly.
I also double checked the wiring which is fine, the changed the board name.
Still nothing I move the CC1101 a bit closer , it's now around 20m in open air, with the sensor meter still underground, but I won't be able to do much for this problem :)
Next step is to try getting even closer, maybe just above the sensor but I'm not sure I'll get wifi connection that far away.
I'll let you know.

zibous, 2023-05-01:
Hi

> Next step is to try getting even closer, maybe just above the sensor but I'm not sure I'll get wifi connection that far away.
> I'll let you know.

O.k. 20m are already a challenge for a WLAN connection.  In addition, the WLAN power interferes with the 868Mhz traffic. 
I removed the CC1101 with longer cables from the device.

I ordered an antenna. Will test how the range increases...
https://amzn.eu/d/itdbwwz

zibous, 2023-05-01:
**Testcase NodeMCU V3**

[image]

[image]

> NOTE: When i connect the device to the Homeassistant i get the JSON Errors.

```yaml
---
## ------------------------------------------------------------------
## WMBUS NODEMCU V3 + CC1101 for Diehl IZAR RC 868 I R4 PL
##
## usage:  1. Using Homassistant API (discovery + devices)
##         2. Using HTTP Post to service server
##         3. Using wmbusmeters i a pipe
##
## This version shows data from the definded watermeter.
## Uses the watermeterID from the secrets.yaml file.
##
## WEBUI: http://water-meter.local
##
## Remark:
##   When device is connected to HA:   reboots with JSON Errors.
## ------------------------------------------------------------------
substitutions:
  device_name_short: "water-meter"
  device_description: "Wasserzähler Wemos D1 Mini, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) - my Watermeter"
  update_interval: 60s

  log_level: "DEBUG"
  appversion: "2.0.5"

  domain: !secret domain
  service_url: !secret service_host
  service_enabled: "false"
  service: "izar.watermeter"

  # additional sensors
  disabled_addOns: "true"

# ----------------------------------------------------------------
# HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
# ----------------------------------------------------------------
esp8266:
  board: nodemcuv2
  restore_from_flash: true

## ----------------------------------------------------------------
## ESPHOME APPLICATION
## ----------------------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  project:
    name: "OE9psj.wmbus-minid1-v1"
    version: ${appversion}
  build_path: ./build/${device_name_short}
  on_boot:
    # At this priority, pretty much everything should already be initialized.
    priority: -100.0
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
  on_shutdown:
[... 792 lines cut ...]
  # curl -i http://water-meter-izar.local/text_sensor/watermeterdata
  - platform: template
    id: watermeterdata
    update_interval: ${update_interval}
    internal: true
    lambda: |-
      char buf[512];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(month_value),
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;

```

```

commoswiss, 2023-05-05:
Hi,
I tested on my neighbour water meter, which is closer than mine, and it's working fine !
So, it's only a range issue, I need to be ~1m away from the device in order to get a signal...
I'll try to work on a repeater to extends the range in open air.
Thanks for your work and help.

Freeman656, 2023-05-05:
Hi,
I tested on my neighbour water meter, which is closer than mine, and it's working fine !
So, it's only a range issue, I need to be ~1m away from the device in order to get a signal...
I'll try to work on a repeater to extends the range in open air.
Thanks for your work and help.

zibous, 2023-05-05:
Hi,
Good news...

I ordered an antenna and get it,but my C1101 is brocken,  the pin are terrible and very delicate. 
Two no longer work ( broken),  have to wait until I get a new module. 

Maybe the antenna is a variant for you.
https://amzn.eu/d/itdbwwz

closed 2023-05-05 by zibous

zibous, 2023-05-13:
@Freeman656 

Or a BT Relay like this, but doesn't necessarily have to be a MiThermometer.
see: https://github.com/maciekn/esp32-meters-relay

Or with Ethernet 100Mb interface :
ESP32-POE
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

pyzimmer, 2024-01-06:
@zibous 
> I ordered an antenna and get it,but my C1101 is brocken, the pin are terrible and very delicate. Two no longer work ( broken), have to wait until I get a new module.
> 
> Maybe the antenna is a variant for you. https://amzn.eu/d/itdbwwz

Hello,
Have you tried the new antenna? Does it improve the range? Would an alternative to the C1101 module have a better range? I need a range of 50 meters. Thanks

SzczepanLeon, 2024-01-06:
> Would an alternative to the C1101 module have a better range? I need a range of 50 meters.

Maybe you can use DIY yagi antenna?

pyzimmer, 2024-01-06:
Good idea, what do you think about this one ? 
[https://fr.aliexpress.com/item/32556400936.html?spm=a2g0o.productlist.main.11.56702a8bPAQahp&algo_pvid=bde066ed-9bae-46b1-a123-ab341c13a1d5&algo_exp_id=bde066ed-9bae-46b1-a123-ab341c13a1d5-5&pdp_npi=4%40dis%21EUR%2116.10%2116.1%21%21%2117.22%21%21%40210387a117045626327907767e87b8%2158741394362%21sea%21FR%21760614743%21&curPageLogUid=uwhdxO3DQsu8&utparam-url=scene%3Asearch%7Cquery_from%3A](url)

SzczepanLeon, 2024-01-13:
For that price I will test it.

SzczepanLeon, 2024-01-19:
Yagi rulezzzzzzzzzzz.