zibous/ha-watermeter #15: Multiple water meters with a single ESP

tomav, 2023-05-31:
Hi @zibous, thanks for this project, works great! 👏

I was wondering if there was a way to collect data from 2 water meters (with different IDs) with a single ESP32?
I suppose it is but can't find an example.

Other question: have you noticed different data between what's on the physical water meter and the data retreived via RF?

Thanks.

zibous, 2023-05-31:
Hi,

Add for each watermeter a `platform: wmbus` section with the `meter_id`
see: <https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-d1mini-simple.yaml>

```
 - platform: wmbus
    # watermeter 1: nearly to mine
    meter_id: 0x43410778
    type: izar
    add_prefix: True
    # get the total watermter m3 from the wmbus telegram
    total_water_m3:
      name: "Watermeter"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter

  - platform: wmbus
    # watermeter 3: nearly to mine
    meter_id: 0x43890785
    type: izar
    add_prefix: True
    # get the total watermter m3 from the wmbus telegram
    total_water_m3:
      name: "Watermeter"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
```

> Other question: have you noticed different data between what's on the physical water meter and the data retreived via RF?

For my IZAR no, same as with wmbusmeters.

tomav, 2023-06-04:
Thanks, that was straightforward!
I can notice an offset of 165L between my water meter and what I get from the ESP, but the offset seems consistent overtime so not an issue.

closed 2023-06-04 by tomav

zibous, 2023-06-05:
> I can notice an offset of 165L between my water meter

interesting, because IZAR always sends data in 8sec intervals, unless a heartbeat filter is used. 

```
      # Send the value periodically with the specified time interval.
      # If the sensor value changes during the interval the interval will not reset.
      # The last value of the sensor will be sent. 60s means, that every minute the
      # last state will be published.
      # The IZAR Watermeter will publish data all 8s (see transmit_period_s),
      # but we need this only evers minute.
      # filters:
      #   - heartbeat: 60s
```

With my water meter, the values are identical to the analog display:

[image]