zibous/ha-watermeter #16: hi how i can read  DME_07 ?

novalan18, 2023-07-24:
hello 
i have wmbus on pi but i want to make it with esp

thanks for your work  all work i can show telegram but i need dme_07 driver  for read value ?

Auto driver  : dme_07
Best driver  : c5isf 06/06
Using driver : dme_07 00/00
000   : 1e length (30 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 31136083 dll-id (83601331)
008   : 7b dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 28 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 13 vif (Volume l)
019 C!: 43A70600 ("total_m3":436.035)
023   : 02 dif (16 Bit Integer/Binary Instantaneous value)
024   : FD vif (Second extension FD of VIF-codes)
025   : 17 vife (Error flags (binary))
026 C!: 0000 ("status":"OK")
028   : 2F skip
029   : 2F skip
030   : 2F skip

{
    "media":"water",
    "meter":"dme_07",
    "name":"",
    "id":"83601331",
    "status":"OK",
    "total_m3":436.035,
    "timestamp":"2023-07-24T08:17:22Z"
}

thanks

zibous, 2023-07-24:
Hi,
I use theesphome-components from  Szczepan Leon, but currently the dme_07 is not present. 
You could ask Szczepan Leon if he would add the dme_07 component.

https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_dme_07.cc
https://github.com/SzczepanLeon/esphome-components

novalan18, 2023-07-24:
thanks for your reply , i go make it now .

SzczepanLeon, 2023-07-24:
> https://github.com/SzczepanLeon/esphome-components

dme07 is available from 2.2.23

zibous, 2023-07-24:
@SzczepanLeon 
Perfect Service, thanks.

closed 2023-07-24 by zibous