zibous/ha-watermeter #17: Universal configuration

MariuszWoszczynski, 2023-08-30:
I can see by the degree of code compilation that you are familiar with esphome. Is there any chance to make a universal configuration to select driver type, ID, Key from Home Assistant?

I tried to figure it out functionally from the ESPHome side and something worked, but I don't know how to substitute it for the sensors section (type: id(my_global_string).c_str() - don`t work)

Example for select driver and write ID:

```select:
  - platform: template
    name: "Meter type"
    id: meter_type
    optimistic: true
    options:
      - "apator162"
      - "apator08"
      - "izar"
      - "hydrocalm3"
      - "mkradio3" 
      - "mkradio4" 
      - "ultrimis"
      - "unismart"
      - "bmeters"
      - "amiplus"
      - "elf"
      - "evo868"
      - "fhkvdataiii"
    initial_option: "izar"
    set_action:
      - globals.set:
          id: my_global_string
          value: !lambda 'return x.c_str();'
      - logger.log:
          format: "Wybrano licznik: %s"
          args: ["x.c_str()"]

globals:
   - id: my_global_string
     type: std::string
     restore_value: no 
     initial_value: '"izar"'

number:
  - platform: template
    name: "Meter ID (DEC)"
    id: Meter_ID
    mode: box
    optimistic: true
    min_value: 0
    max_value: 99999999
    step: 1

SzczepanLeon, 2023-08-30:
I've told You that with my code (current version) it is not possible.

C++ code is generated from YAML, so binary is only for that YAML configuration.

Big refactorisation is needed to achieve "template-wise" binary.

MariuszWoszczynski, 2023-08-31:
Yes, you pointed out to me where something needs to be changed, but I'm not a programmer :) You wrote that on average you have time for this, so I am looking for other solutions :)

zibous, 2023-09-02:
see: 
> I've told You that with my code (current version) it is not possible.

closed 2023-09-02 by zibous