zibous/ha-watermeter #19: Finding Watermeter ID with the test install

blundell, 2023-09-04:
I've done this step:

> Install [wm-esp32_test.yaml](https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-esp32_test.yaml) To find the watermeterId  you can set the watermeterId to 0 and the log_level: "VERBOSE" to find your meterId.
see log messages to find your watermeterID

but I cannot see `watermeterID` in my logs:

```
NFO ESPHome 2023.8.2
INFO Reading configuration /config/esphome/esphome-web-988b80.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Starting log output from water-meter-esp.local using esphome API
INFO Successfully connected to water-meter-esp.local
[19:52:15][I][app:102]: ESPHome version 2023.8.2 compiled on Sep  4 2023, 19:39:45
[19:52:15][I][app:104]: Project Diehl_IZAR_RC_868.Watermeter version 2.0.7
[19:52:15][C][wifi:543]: WiFi:
[19:52:15][C][wifi:379]:   Local MAC: redacted
[19:52:15][C][wifi:380]:   SSID: [redacted]
[19:52:15][C][wifi:381]:   IP Address: redacted
[19:52:15][C][wifi:383]:   BSSID: [redacted]
[19:52:15][C][wifi:384]:   Hostname: 'water-meter-esp'
[19:52:15][C][wifi:386]:   Signal strength: -56 dB ▂▄▆█
[19:52:15][V][wifi:388]:   Priority: 1.0
[19:52:15][C][wifi:390]:   Channel: redacted
[19:52:15][C][wifi:391]:   Subnet: redacted
[19:52:15][C][wifi:392]:   Gateway: redacted
[19:52:15][C][wifi:393]:   DNS1: redacted
[19:52:15][C][wifi:394]:   DNS2: 0.0.0.0
[19:52:15][C][logger:301]: Logger:
[19:52:15][C][logger:302]:   Level: VERBOSE
[19:52:15][C][logger:303]:   Log Baud Rate: 0
[19:52:15][C][logger:305]:   Hardware UART: UART0
[19:52:15][C][logger:309]:   Level for 'wmbus': DEBUG
[19:52:15][C][logger:309]:   Level for 'wMBus-lib': DEBUG
[19:52:15][C][template.sensor:022]: Template Sensor 'Device Boot counter'
[19:52:15][C][template.sensor:022]:   State Class: 'measurement'
[19:52:15][C][template.sensor:022]:   Unit of Measurement: ''
[19:52:15][C][template.sensor:022]:   Accuracy Decimals: 0
[19:52:15][C][template.sensor:022]:   Icon: 'mdi:counter'
[19:52:15][C][template.sensor:023]:   Update Interval: 60.0s
[19:52:15][C][template.switch:068]: Template Switch 'Device Boot Counter reset'
[19:52:15][C][template.switch:091]:   Restore Mode: always OFF
[19:52:15][C][template.switch:057]:   Optimistic: NO
[19:52:16][C][safe_mode_switch:068]: Safe Mode Switch 'Device Restart (Safe Mode)'
[19:52:16][C][safe_mode_switch:070]:   Icon: 'mdi:restart-alert'
[19:52:16][C][safe_mode_switch:091]:   Restore Mode: always OFF
[19:52:16][C][restart:068]: Restart Switch 'Restart'
[19:52:16][C][restart:070]:   Icon: 'mdi:restart'
[19:52:16][C][restart:091]:   Restore Mode: always OFF
[19:52:16][C][factory_reset.switch:068]: Factory Reset Switch 'Device Restart Factory'
[19:52:16][C][factory_reset.switch:070]:   Icon: 'mdi:restart-alert'
[19:52:16][C][factory_reset.switch:091]:   Restore Mode: always OFF
[19:52:16][C][copy.sensor:015]: Copy Sensor 'Device WLAN Signal'
[19:52:16][C][copy.sensor:015]:   Device Class: 'signal_strength'
[19:52:16][C][copy.sensor:015]:   State Class: 'measurement'
[19:52:16][C][copy.sensor:015]:   Unit of Measurement: '%'
[19:52:16][C][copy.sensor:015]:   Accuracy Decimals: 0
[19:52:16][C][captive_portal:088]: Captive Portal:
[19:52:16][C][backup:033]: Backup:
[19:52:16][C][backup:034]:   URL path is /config.yaml
[19:52:16][C][backup:036]:   Basic authentication enabled
[19:52:16][C][web_server:161]: Web Server:
[19:52:16][C][web_server:162]:   Address: water-meter-esp.local:80
[19:52:16][C][sntp:053]: SNTP Time:
[19:52:16][C][sntp:054]:   Server 1: '0.at.pool.ntp.org'
[19:52:16][C][sntp:055]:   Server 2: '0.pool.ntp.org'
[19:52:16][C][sntp:056]:   Server 3: '1.pool.ntp.org'
[19:52:16][C][sntp:057]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[19:52:16][C][mdns:112]: mDNS:
[19:52:16][C][mdns:113]:   Hostname: water-meter-esp
[... 96 lines cut ...]
[19:53:42][D][sensor:094]: 'wifi_signal_db': Sending state -59.00000 dBm with 0 decimals of accuracy
[19:53:42][V][sensor:043]: 'Device WLAN Signal': Received new state -59.000000
[19:53:42][D][sensor:094]: 'Device WLAN Signal': Sending state 82.00000 % with 0 decimals of accuracy
[19:53:42][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[19:53:42][V][json:056]: Size after shrink 80 bytes
[19:54:11][V][sensor:043]: 'Device Boot counter': Received new state 2.000000
[19:54:11][D][sensor:094]: 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
[19:54:11][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[19:54:11][V][json:056]: Size after shrink 80 bytes
[19:54:42][V][sensor:043]: 'wifi_signal_db': Received new state -57.000000
[19:54:42][D][sensor:094]: 'wifi_signal_db': Sending state -57.00000 dBm with 0 decimals of accuracy
[19:54:42][V][sensor:043]: 'Device WLAN Signal': Received new state -57.000000
[19:54:42][D][sensor:094]: 'Device WLAN Signal': Sending state 86.00000 % with 0 decimals of accuracy
[19:54:42][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[19:54:42][V][json:056]: Size after shrink 80 bytes
[19:55:11][V][sensor:043]: 'Device Boot counter': Received new state 2.000000
[19:55:11][D][sensor:094]: 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
[19:55:11][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[19:55:11][V][json:056]: Size after shrink 80 bytes
```

Any advice of what else I can investigate?

I was wondering if my soldering was correct, can that be verified from the logs?
Perhaps  I need to get closer to the water meter?

zibous, 2023-09-05:
Hi,
Log actually looks pretty good. 

Possible causes for you could be: 

 - CC1101 connected correctly but broken 
 - Device too far from the water meter.
 
However, you can test whether you have connected the CC1101 correctly by deliberately doing the wiring incorrectly, then an error message should appear in the log.

Normally IZAR RC 868 I R4 PL sends a telegram every 8s.

This is my Log:
```
INFO ESPHome 2023.8.2
INFO Reading configuration /config/wm-esp32-test.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Generating C++ source...
INFO Backup config will take: 2065 bytes
INFO Compiling app...
Processing water-meter-esp (board: az-delivery-devkit-v4; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.4.0
INFO Installing platformio/espressif32 @ 5.4.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.4.0 has been installed!
INFO espressif32@5.4.0 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing git+https://github.com/SzczepanLeon/wMbus-lib#1.2.22
INFO Installing git+https://github.com/SzczepanLeon/wMbus-lib#1.2.22
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-9g2wijhm'...
Note: switching to '343d0f58339f2fc4062c8306e94a34efedff2e55'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Library Manager: wMbus-lib@1.2.22+sha.343d0f5 has been installed!
INFO wMbus-lib@1.2.22+sha.343d0f5 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
[... 396 lines cut ...]
[08:07:34][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[08:07:34][V][json:056]: Size after shrink 92 bytes
[08:07:34][V][sensor:043]: '0x00 Watermeter CC1101 RSSI': Received new state -81.000000
[08:07:34][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -81.00000 % with 0 decimals of accuracy
[08:07:34][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[08:07:34][V][json:056]: Size after shrink 92 bytes
[08:07:34][D][wmbus:184]: Meter ID [0x43410778] RSSI: -81 dBm LQI: 133 Mode: T1 not found in configuration T: 1944A511780741434418A251140013DB1EA79256156569725095 (26)
[08:07:34][W][component:204]: Component wmbus took a long time for an operation (0.08 s).
[08:07:34][W][component:205]: Components should block for at most 20-30ms.
[08:07:42][V][sensor:043]: '0x00 Watermeter CC1101 LQI': Received new state 130.000000
[08:07:42][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 130.00000 lqi with 0 decimals of accuracy
[08:07:42][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[08:07:42][V][json:056]: Size after shrink 92 bytes
[08:07:42][V][sensor:043]: '0x00 Watermeter CC1101 RSSI': Received new state -78.000000
[08:07:42][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -78.00000 % with 0 decimals of accuracy
[08:07:42][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[08:07:42][V][json:056]: Size after shrink 92 bytes
[08:07:42][D][wmbus:184]: Meter ID [0x43410778] RSSI: -78 dBm LQI: 130 Mode: T1 not found in configuration T: 1944A511780741434418A261140013DB27D648AB7711031F7B92 (26)
[08:07:42][W][component:204]: Component wmbus took a long time for an operation (0.09 s).
[08:07:42][W][component:205]: Components should block for at most 20-30ms.
[08:07:53][V][sensor:043]: 'wifi_signal_db': Received new state -52.000000
[08:07:53][D][sensor:094]: 'wifi_signal_db': Sending state -52.00000 dBm with 0 decimals of accuracy
[08:07:53][V][sensor:043]: 'Device WLAN Signal': Received new state -52.000000
[08:07:53][D][sensor:094]: 'Device WLAN Signal': Sending state 96.00000 % with 0 decimals of accuracy
```

grandyoz, 2023-09-18:
Bonjour,
J'ai le même problème que toi... je n'arrive pas à voir passer de trames...
Pourtant, j'ai déjà un wmbus qui fonctionne sur mon compteur avec une clé USB sur une raspberry...

Voilà mes trames : 

`
16:49:53 | [D] | [sensor:094] | 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
-- | -- | -- | --
16:49:53 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:49:53 | [V] | [json:056] | Size after shrink 80 bytes
16:49:53 | [V] | [json:056] | Size after shrink 80 bytes
16:50:47 | [V] | [sensor:043] | 'wifi_signal_db': Received new state -61.000000
16:50:47 | [D] | [sensor:094] | 'wifi_signal_db': Sending state -61.00000 dBm with 0 decimals of accuracy
16:50:47 | [V] | [sensor:043] | 'Device WLAN Signal': Received new state -61.000000
16:50:47 | [D] | [sensor:094] | 'Device WLAN Signal': Sending state 78.00000 % with 0 decimals of accuracy
16:50:47 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:50:47 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:50:47 | [V] | [json:056] | Size after shrink 80 bytes
16:50:47 | [V] | [json:056] | Size after shrink 80 bytes
16:50:53 | [V] | [sensor:043] | 'Device Boot counter': Received new state 2.000000
16:50:53 | [D] | [sensor:094] | 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
16:50:53 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:50:53 | [V] | [json:056] | Size after shrink 80 bytes
16:51:47 | [V] | [sensor:043] | 'wifi_signal_db': Received new state -63.000000
16:51:47 | [D] | [sensor:094] | 'wifi_signal_db': Sending state -63.00000 dBm with 0 decimals of accuracy
16:51:47 | [V] | [sensor:043] | 'Device WLAN Signal': Received new state -63.000000
16:51:47 | [D] | [sensor:094] | 'Device WLAN Signal': Sending state 74.00000 % with 0 decimals of accuracy
16:51:47 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:51:47 | [V] | [json:056] | Size after shrink 80 bytes
16:51:53 | [V] | [sensor:043] | 'Device Boot counter': Received new state 2.000000
16:51:53 | [D] | [sensor:094] | 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
16:51:53 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:51:53 | [V] | [json:056] | Size after shrink 80 bytes
16:52:47 | [V] | [sensor:043] | 'wifi_signal_db': Received new state -65.000000
16:52:47 | [D] | [sensor:094] | 'wifi_signal_db': Sending state -65.00000 dBm with 0 decimals of accuracy
16:52:47 | [V] | [sensor:043] | 'Device WLAN Signal': Received new state -65.000000
16:52:47 | [D] | [sensor:094] | 'Device WLAN Signal': Sending state 70.00000 % with 0 decimals of accuracy
16:52:47 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:52:47 | [V] | [json:056] | Size after shrink 80 bytes
16:52:53 | [V] | [sensor:043] | 'Device Boot counter': Received new state 2.000000
16:52:53 | [D] | [sensor:094] | 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
16:52:53 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:52:53 | [V] | [json:056] | Size after shrink 80 bytes
16:53:07 | [D] | [sntp:078] | Synchronized time: 2023-09-18 16:53:07
16:53:07 | [I] | [system:386] | Synchronized sntp clock
16:53:47 | [V] | [sensor:043] | 'wifi_signal_db': Received new state -59.000000
16:53:47 | [D] | [sensor:094] | 'wifi_signal_db': Sending state -59.00000 dBm with 0 decimals of accuracy
16:53:47 | [V] | [sensor:043] | 'Device WLAN Signal': Received new state -59.000000
16:53:47 | [D] | [sensor:094] | 'Device WLAN Signal': Sending state 82.00000 % with 0 decimals of accuracy
16:53:47 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:53:47 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:53:47 | [V] | [json:056] | Size after shrink 80 bytes
16:53:47 | [V] | [json:056] | Size after shrink 80 bytes
16:53:53 | [V] | [sensor:043] | 'Device Boot counter': Received new state 2.000000
16:53:53 | [D] | [sensor:094] | 'Device Boot counter': Sending state 2.00000  with 0 decimals of accuracy
16:53:53 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:53:53 | [V] | [json:056] | Size after shrink 80 bytes
`

J'ai bien vérifié mon montage électrique et il a l'air juste.

Si vous avez un conseil à nous donner !
Merci !

zibous, 2023-09-18:
@grandyoz 

Which device d1mini, esp32 ??
Is CC1101 SPI bus  valid ?

```
[wmbus:394]: wM-Bus v2.2.29:
[wmbus:411]:   CC1101 SPI bus:
[wmbus:412]:     MOSI Pin: GPIO23
[wmbus:413]:     MISO Pin: GPIO19
[wmbus:414]:     CLK Pin:  GPIO18
[wmbus:415]:     CS Pin:   GPIO5
[wmbus:416]:     GDO0 Pin: GPIO16
[wmbus:417]:     GDO2 Pin: GPIO17
[wmbus:424]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, dme07, elf, evo868, fhkvdataiii, hydrocalm3, hydrus, iperl, itron, izar, mkradio3, mkradio4, qheat, qwater, sharky774, topaseskr, ultrimis, unismart, vario451
[wmbus:442]:   Meter:
[wmbus:443]:     ID: 0 [0x00000000]
[wmbus:444]:     Type: izar
[wmbus:445]:     Mode: T1
[wmbus:446]:     Key: ''
```

grandyoz, 2023-09-18:
Hi Zibous, thanks for your hek-lp !
I use a ESP32 and I read log from the device web page. I never seen that log.
I don't know what to do...

zibous, 2023-09-18:
Hi,

You can use the ESPHOME Dashboard and select the device "View Log" browser.

[image]

or use the https://web.esphome.io/ with chrome 

New Testcase using https://web.esphome.io/ with chrome browser.

    1. Connect to the device
    2. Prepare for first use
    3. When finish set the local WIFI and "Visit Device"
    4. OTA Update water-meter-esp32-izar-test.bin

see: <https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/water-meter-esp32-izar-test.bin>

grandyoz, 2023-09-18:
Thanks, I'll test that and I will tell you !

grandyoz, 2023-09-18:
That's very strange because I did what you said and in https://web.esphome.io/ with chrome, I have just few logs lines : 
[image]
and on device web page, I see nothing. When I click on restart, I see some lines but nothing more...
[image]

I never get as logs as you have... Perhaps I have a "ESPHome" problem...

grandyoz, 2023-09-18:
I have this device : https://fr.aliexpress.com/item/1005002074380868.html?spm=a2g0o.order_list.order_list_main.73.10b15e5bEqxmXu&gatewayAdapt=glo2fra
But despite it is written 868Mhz in the title, it's written 433MHz on the board, isn't it a problem ?
Perhaps I should see more logs, but I probably have another problem after !
What do you think ?

zibous, 2023-09-18:
Try to "Log a Console" and restart the device .
Put the device near to your watermeter. Wait vor 5 -30 min, sometime it needs time..

grandyoz, 2023-09-18:
My wmbusmeter on my raspberry is working fine every 8 secs... why do you think it can take 30 mins ?

zibous, 2023-09-18:
> My wmbusmeter on my raspberry is working fine every 8 secs... why do you think it can take 30 mins ?

Yes that is correct, the IZAR sends all 8s a telegram but wmbusmetere needs time to intialize

> Technologie sans fil CC1101 avec antenne SMA, Transcsec Tech pour Ardu37315, 433, 868, 915 Z successifs

Are you shure the this CC1101 is for 868 Mhz enabled , because on the Board says 444Mhz.
My IZAR Watermeter uses 868 Mhz.

i use this https://amzn.eu/d/5Rs9MKz or this one https://amzn.eu/d/cAPtUio

grandyoz, 2023-09-18:
I think mine is not for 868 and that is a problem...
I will buy a different one.
I will tell you if I succeed with the new module !
Thanks a lot for your library and for your help ! Really !

zibous, 2023-09-18:
ok.

The CC1101 is cheaper on Amazon.de see: https://amzn.eu/d/h9joMXH

grandyoz, 2023-10-13:
Indeed, it was my device... !
I don't understand what should I set in "service_url: !secret service_host" variable. And so, I can't build the esp32 project.
Could you help me ?
Thanks !

zibous, 2023-10-13:
@grandyoz 

>  don't understand what should I set in "service_url: !secret service_host" variable.

This is a special service the sends all data to a webserver.

```
service_host: http://<your webserver>:<PORT>
```
see: > https://github.com/zibous/ha-watermeter/blob/master/tools/simpleserver.py

**First use to find the watermeter ID:** 
> https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-esp32-test.yaml

**For production use:**
> https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-esp32-simple.yaml

Webserver:
```
## ---------------------------------------------------
## COMPONENT WEBSERVER
## ---------------------------------------------------
web_server:
  port: 80
  version: 2
  # js_url: !secret webserver_jsurl
```
Info see:
> https://github.com/zibous/ha-watermeter/tree/master/esphome/webserver

**Alternatively, you could also use the pre compiled test versions to test the hardware**
> see: https://github.com/zibous/ha-watermeter/tree/master/esphome/testcases

grandyoz, 2023-10-13:
Ok, simple version works.
But because I had read
`2nd Step: Edit the secrets.yaml and change the watermeterID

Install [wm-esp32.yaml](https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-esp32.yaml)
Check log messages`

I tried with wm-esp32.yaml...
But, ok, I will stay on simple version !

Thanks !

grandyoz, 2023-10-13:
Is il possible to change the update frequency ? It seems to be 5 minutes.

zibous, 2023-10-13:
@grandyoz 

> Is it possible to change the update frequency ? It seems to be 5 minutes.

My watermeter reports all 8s ...

[image]

I don't know any setting, because wbmusmeters listens to the telegrams that the water meter sends (usually every 8s) and evaluates them.

grandyoz, 2023-10-13:
Yes, you're right.
It's was a display problem in Home Assistant.
All is good.
Thanks again for your help !

zibous, 2023-10-13:
@grandyoz 

You are welcome

closed 2023-10-13 by zibous

blundell, 2023-10-13:
Mines still not working but 😂

zibous, 2023-10-14:
@blundell 

1.  Check the wireings (change gdo0_pin,  gdo2_pin)
2. Try to get closer to the watermeter
3. Try another CC1101

blundell, 2023-10-18:
Thanks!

I've reported the meter to the utility company as I think its not giving off a signal! 

Time will tell if they fix it or even reply 🤞