zibous/ha-watermeter #20: unable to compile

Woodpeckercz, 2023-09-12:
Hi, 

thanks for putting this together.
I'm trying to use the d1mini version.
I'm having some issues with compiling this.
I use the ESPHome addon in homeassistant and when I try to compile this, I'm getting the following error.

`In file included from /data/esp_wmbus/.piolibdeps/esp_wmbus/wMbus-lib/rf_mbus.cpp:6:
/data/esp_wmbus/.piolibdeps/esp_wmbus/wMbus-lib/tmode_rf_settings.hpp:8:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory

**********************************************************************************
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"
* Web  > https://registry.platformio.org/search?q=header:ELECHOUSE_CC1101_SRC_DRV.h
*
**********************************************************************************

    8 | #include <ELECHOUSE_CC1101_SRC_DRV.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/esp_wmbus/.pioenvs/esp_wmbus/lib521/wMbus-lib/rf_mbus.cpp.o] Error 1
========================= [FAILED] Took 52.25 seconds =========================`

Can you please help me?

zibous, 2023-09-13:
> * Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!

ELECHOUSE_CC1101 is loaded by the wmbus lib from Szczepan Leon. 
I would first try using the ESP Dashboard with a “clean build” (deletes all loaded libs from the project).

Woodpeckercz, 2023-09-13:
Tried doing that again. 
I even removed all sensors and any stuff that might cause errors from the code.

Here's my current configuration:

`esphome:
  name: esp-wmbus

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: "redacted"
  password: "redacted"
  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Wmbus Fallback Hotspot"
    password: "redacted"

captive_portal:
web_server:
  port: 80

time:
  - platform: sntp
    id: time_sntp

external_components:
      

  - source: github://SzczepanLeon/esphome-components@main
    components: [wmbus]

wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO5
  clk_pin:  GPIO2
  cs_pin:   GPIO14
  gdo0_pin: GPIO15
  gdo2_pin: GPIO16

  led_pin: GPIO0
  led_blink_time: "1s"

`

Here's the log from install:

`INFO Reading configuration /config/esphome/11.yaml...

INFO Detected timezone 'Europe/Dublin'

INFO Generating C++ source...

INFO Compiling app...

Processing esp-wmbus (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
[... 307 lines cut ...]
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!

*

* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"

* Web  > https://registry.platformio.org/search?q=header:ELECHOUSE_CC1101_SRC_DRV.h

*

**********************************************************************************

    8 | #include <ELECHOUSE_CC1101_SRC_DRV.h>

      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

*** [/data/esp-wmbus/.pioenvs/esp-wmbus/lib523/wMbus-lib/rf_mbus.cpp.o] Error 1

========================= [FAILED] Took 56.51 seconds =========================

`

Hopefully this will be of some use.

zibous, 2023-09-13:
@Woodpeckercz 

Which esphome (docker, system) installation and which version do you use ?

I use:
> ESPHome by Nabu Casa with docker | [Fund development](https://esphome.io/guides/supporters.html) | [2023.8.3](https://www.esphome.io/)

Normally all required libs are managed and loaded by the wmbus component. 
Except there is not enough storage space or permission problems or the internet connection is bad. 

One attempt (not pretty, but worth testing) would be to load the SmartRC-CC1101 Driver Lib via the ESPHome configuration.

```
esphome:
    ....
    ....
    libraries:
        - SPI
        - "SmartRC-CC1101-Driver-Lib"
```

Woodpeckercz, 2023-09-13:
Thanks, 
It still didn't work. There has to be something wrong with my esphome(addon) in the homeassistant (esphome version 2022.3.1).
I installed the cli esphome on windows and the file compiled and flashed fine.

Woodpeckercz, 2023-09-14:
I'm getting some error in the log now, error decoding ?

zibous, 2023-09-14:
> It still didn't work. There has to be something wrong with my esphome(addon) in the homeassistant (esphome version 2022.3.1).

Yes, that can be the problem and that is the reason why I run ESPHOME in a Docker container on my Linux machine.

```bash
#!/bin/bash

# ---------------------------------------------
# sudo bash esphome.sh
# see: https://hub.docker.com/r/esphome/esphome
# ----------------------------------------------

IDu=$(id -u) # UID
IDg=$(id -g) # GID

CURRENTURL=http://$(hostname).$(dnsdomainname)
DOCKERIMAGE=esphome/esphome:latest
CONTAINERLABEL=esphome
APPSDATA=$PWD${DOCKER_APPSDIR}${CONTAINERLABEL}
DOCKER_TIMEZONE=Europe/Berlin

mkdir -p ${APPSDATA}

echo "Try to remove previuos installation ${CONTAINERLABEL}..."
docker stop ${DOCKERIMAGE} && docker rm ${DOCKERIMAGE}

echo "Install Docker container ${CONTAINERLABEL}."
docker run --detach \
  --name ${CONTAINERLABEL} \
  --privileged=true \
  --env TZ=${DOCKER_TIMEZONE} \
  --restart unless-stopped \
  --volume /run/udev:/run/udev \
  --volume /etc/localtime:/etc/localtime:ro \
  --volume ${APPSDATA}/config:/config \
  --network host \
  ${DOCKERIMAGE}

echo "Docker container ${CONTAINERLABEL} ready."
echo "Run WEBGUI: ${CURRENTURL}:6052"
```

zibous, 2023-09-14:
> I'm getting some error in the log now, error decoding ?

Frist try the simple version:
```yaml
## ------------------------------------------------------------------
## WMBUS METER D1 MINI + CC1101 for Diehl IZAR RC 868 I R4 PL
## HARDWARE: D1 MINI ESP8266 80MHz, 80KB RAM, 4MB Flash
##
## CC1101 TESTCASE FOR ALL WATERMETERS
## DO NOT USE THIS CONFIGURATION FOR PRODUCTION
##
##   1. uses esphome-components latest version from github
##   2. uses syslog component  (optional)
##      tail -f /var/log/syslog | grep "water-meter"
##
## This version shows data from the all watermeters.
## Use this for frist test to find your watermeterID.
##
## ------------------------------------------------------------------
substitutions:

  ## device settings
  device_name_short: "water-meter-dmini"
  device_description: "TEST: Wasserzähler Wemos D1 Min, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) - find watermeters"
  projectname: "Diehl IZAR RC 868.Watermeter-Test"
  appversion: "2.0.8"

  ## all watermeters wmid: "0"
  ## your watermeter: wmid: !secret watermeterId
  wmid: "0"
  
  ## logger settings, use DEBUG or VERBOSE to see the telegram data
  log_level: "VERBOSE"    # In production mode use "WARN"
  log_wmbus: "VERBOSE"    # Loglevel for wmbus meters component
  log_baudrate: "0"     # Disable uart logger messages log baudrate=0

## ----------------------------------------------------------------
## HARDWARE: WEMOS D1 MINI ESP8266 80MHz, 80KB RAM, 4MB Flash
## ----------------------------------------------------------------
# wemmos D1 Mini  = d1_mini
# D1 mit externer Antenne = d1_mini_pro
# esp01 mit 1MB Speicher = esp01_1m
esp8266:
  board: d1_mini
  restore_from_flash: true

preferences:
  flash_write_interval: 1min

## ---------------------------------------------------
## ESPHOME Configuration
## ---------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}
  on_boot:
      then:
        - globals.set:
            id: boot_counter
            value: !lambda "return id(boot_counter)+=1;"
        - logger.log:
            level: WARN
            tag: "system"
[... 289 lines cut ...]
  - platform: factory_reset
    icon: mdi:lock-reset
    name: Restart factory default

## ---------------------------------------------------
## TEXT SENSORS
## ---------------------------------------------------
text_sensor:
  - platform: debug
    # Reports the device information:
    device:
      name: "Device Info"
      entity_category: "diagnostic"
    # Reports the last reboot reason in a human-readable form.
    reset_reason:
      name: "Reset Reason"
      entity_category: "diagnostic"

  # Last boot timestamp
  - platform: template
    name: "Last Boot timestamp"
    disabled_by_default: true
    id: device_lastBoot_time
    icon: mdi:clock-start
```

zibous, 2023-09-14:
@Woodpeckercz 

Here you can test firmware. I would flash with "ESPHOME Flasher".
> Firmware : https://github.com/zibous/ha-watermeter/tree/master/esphome/testcases

> ESPHOME Flasher: https://github.com/esphome/esphome-flasher/releases

Woodpeckercz, 2023-09-14:
Ok, I flashed the code from earlier as I 'm not currently at home so I did it remotely. 
How do i access the logs? 
Thx

zibous, 2023-09-14:
> How do i access the logs?

On ESPHome select the device and "LOGS".

[image]

Woodpeckercz, 2023-09-14:
only realised it after I posted the question.

16:10:33 | [V] | [sensor:043] | 'Heap Free': Received new state 25288.000000
16:10:33 | [D] | [sensor:093] | 'Heap Free': Sending state 25288.00000 B with 0 decimals of accuracy
16:10:33 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:33 | [V] | [json:056] | Size after shrink 76 bytes
16:10:33 | [V] | [sensor:043] | 'Heap Max Block': Received new state 22296.000000
16:10:33 | [D] | [sensor:093] | 'Heap Max Block': Sending state 22296.00000 B with 0 decimals of accuracy
16:10:33 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:33 | [V] | [json:056] | Size after shrink 80 bytes
16:10:34 | [V] | [sensor:043] | 'Heap Fragmentation': Received new state 2.000000
16:10:34 | [D] | [sensor:093] | 'Heap Fragmentation': Sending state 2.00000 % with 1 decimals of accuracy
16:10:34 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:34 | [V] | [json:056] | Size after shrink 80 bytes
16:10:34 | [V] | [sensor:043] | 'Application Loop Time': Received new state 117.000000
16:10:34 | [D] | [sensor:093] | 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
16:10:34 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:34 | [V] | [json:056] | Size after shrink 84 bytes
16:10:34 | [W] | [component:204] | Component debug took a long time for an operation (0.10 s).
16:10:34 | [W] | [component:205] | Components should block for at most 20-30ms.
16:10:38 | [V] | [sensor:043] | 'Heap Free': Received new state 25288.000000
16:10:40 | [D] | [sensor:093] | 'Heap Free': Sending state 25288.00000 B with 0 decimals of accuracy
16:10:40 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:40 | [V] | [json:056] | Size after shrink 76 bytes
16:10:40 | [V] | [sensor:043] | 'Heap Max Block': Received new state 22192.000000
16:10:40 | [D] | [sensor:093] | 'Heap Max Block': Sending state 22192.00000 B with 0 decimals of accuracy
16:10:41 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:41 | [V] | [json:056] | Size after shrink 80 bytes
16:10:41 | [V] | [sensor:043] | 'Heap Fragmentation': Received new state 3.000000
16:10:41 | [D] | [sensor:093] | 'Heap Fragmentation': Sending state 3.00000 % with 1 decimals of accuracy
16:10:41 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:41 | [V] | [json:056] | Size after shrink 80 bytes
16:10:41 | [V] | [sensor:043] | 'Application Loop Time': Received new state 116.000000
16:10:41 | [D] | [sensor:093] | 'Application Loop Time': Sending state 116.00000 ms with 0 decimals of accuracy
16:10:41 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
16:10:41 | [V] | [json:056] | Size after shrink 84 bytes
16:10:41 | [W] | [component:204] | Component debug took a long time for an operation (0.10 s).
16:10:41 | [W] | [component:205] | Components should block for at most 20-30ms.

zibous, 2023-09-14:
But this is not the test version because the logging is not filtered. The log lines are important when booting, they show whether the CC1101 is connected correctly and is working .

Log Filter:
```
## ---------------------------------------------------
## LOGGER COMPONENT
## ---------------------------------------------------
logger:
  id: appslogger
  level: "VERBOSE"
  baud_rate: "0"
  logs:
    wmbus: "VERBOSE"
    wMBus-lib: "VERBOSE"
    sensor: WARN
    sensor.filter: WARN
    text_sensor: WARN
    api.service: ERROR
    json: ERROR
    mqtt: WARN
    scheduler: ERROR
    wifi: WARN
    component: ERROR
    api: WARN
```

Log Testversion:

```
INFO ESPHome 2023.8.3
INFO Reading configuration /config/wm-d1mini-izar-test.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Starting log output from water-meter-izar-test.local using esphome API
INFO Successfully connected to water-meter-izar-test.local
[17:16:18][I][app:102]: ESPHome version 2023.8.3 compiled on Sep 14 2023, 14:22:35
[17:16:18][I][app:104]: Project Diehl_IZAR_RC_868.Watermeter version 2.0.9
[17:16:18][C][logger:301]: Logger:
[17:16:18][C][logger:302]:   Level: VERBOSE
[17:16:18][C][logger:303]:   Log Baud Rate: 0
[17:16:18][C][logger:305]:   Hardware UART: UART0
[17:16:18][C][logger:309]:   Level for 'wmbus': VERBOSE
[17:16:18][C][logger:309]:   Level for 'wMBus-lib': VERBOSE
[17:16:18][C][logger:309]:   Level for 'sensor': WARN
[17:16:18][C][logger:309]:   Level for 'sensor.filter': WARN
[17:16:18][C][logger:309]:   Level for 'text_sensor': WARN
[17:16:18][C][logger:309]:   Level for 'api.service': ERROR
[17:16:18][C][logger:309]:   Level for 'json': ERROR
[17:16:18][C][logger:309]:   Level for 'mqtt': WARN
[17:16:18][C][logger:309]:   Level for 'scheduler': ERROR
[17:16:18][C][logger:309]:   Level for 'wifi': WARN
[17:16:18][C][logger:309]:   Level for 'component': ERROR
[17:16:18][C][logger:309]:   Level for 'api': WARN
[17:16:18][C][restart:068]: Restart Switch 'Restart Device'
[17:16:18][C][restart:070]:   Icon: 'mdi:restart'
[17:16:18][C][restart:090]:   Restore Mode: always OFF
[17:16:18][C][safe_mode_switch:068]: Safe Mode Switch 'Restart (Safe Mode)'
[17:16:18][C][safe_mode_switch:070]:   Icon: 'mdi:restart-alert'
[17:16:18][C][safe_mode_switch:090]:   Restore Mode: always OFF
[17:16:18][C][factory_reset.switch:068]: Factory Reset Switch 'Restart factory default'
[17:16:18][C][factory_reset.switch:070]:   Icon: 'mdi:lock-reset'
[17:16:18][C][factory_reset.switch:090]:   Restore Mode: always OFF
[17:16:18][C][captive_portal:088]: Captive Portal:
[17:16:18][C][web_server:161]: Web Server:
[17:16:18][C][web_server:162]:   Address: water-meter-izar-test.local:80
[17:16:18][C][sntp:053]: SNTP Time:
[17:16:18][C][sntp:054]:   Server 1: '0.at.pool.ntp.org'
[17:16:18][C][sntp:055]:   Server 2: '0.pool.ntp.org'
[17:16:18][C][sntp:056]:   Server 3: ''
[17:16:18][C][sntp:057]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[17:16:18][C][mdns:112]: mDNS:
[17:16:18][C][mdns:113]:   Hostname: water-meter-izar-test
[17:16:18][V][mdns:114]:   Services:
[17:16:18][V][mdns:116]:   - _esphomelib, _tcp, 6053
[17:16:18][V][mdns:118]:     TXT: version = 2023.8.3
[17:16:18][V][mdns:118]:     TXT: mac = 483fda510e55
[17:16:18][V][mdns:118]:     TXT: platform = ESP8266
[17:16:18][V][mdns:118]:     TXT: board = d1_mini
[17:16:18][V][mdns:118]:     TXT: network = wifi
[17:16:18][V][mdns:118]:     TXT: project_name = Diehl_IZAR_RC_868.Watermeter
[17:16:18][V][mdns:118]:     TXT: project_version = 2.0.9
[17:16:18][V][mdns:116]:   - _http, _tcp, 80
[17:16:18][C][ota:093]: Over-The-Air Updates:
[17:16:18][C][ota:094]:   Address: water-meter-izar-test.local:8266
[17:16:18][C][ota:097]:   Using Password.
[17:16:18][C][wifi_signal.sensor:009]: WiFi Signal 'Device WLAN RSSI Quality'
[17:16:18][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[17:16:18][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[17:16:18][C][wifi_signal.sensor:009]:   Unit of Measurement: '%'
[17:16:18][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[17:16:18][V][wifi_signal.sensor:009]:   Unique ID: '483fda510e55-wifisignal'
[17:16:18][C][wmbus:394]: wM-Bus v2.2.29:
[17:16:18][C][wmbus:411]:   CC1101 SPI bus:
[17:16:18][C][wmbus:412]:     MOSI Pin: GPIO13
[17:16:18][C][wmbus:413]:     MISO Pin: GPIO12
[17:16:18][C][wmbus:414]:     CLK Pin:  GPIO14
[17:16:18][C][wmbus:415]:     CS Pin:   GPIO15
[17:16:18][C][wmbus:416]:     GDO0 Pin: GPIO4
[17:16:18][C][wmbus:417]:     GDO2 Pin: GPIO5
[17:16:18][E][wmbus:430]:   Check connection to CC1101!
```

Woodpeckercz, 2023-09-14:
here's the log
```
[16:44:06]SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[16:44:06][I][logger:262]: Log initialized
[16:44:06][I][app:029]: Running through setup()...
[16:44:06][V][app:030]: Sorting components by setup priority...
[16:44:06][D][template.switch:046]:   Restored state OFF
[16:44:06][D][switch:016]: 'Boot Counter reset' Turning OFF.
[16:44:06][W][system:066]: BOOTMESSAGE:water-meter-dmini API is connected, Device ready!
[16:44:06][V][wifi_esp8266:058]: Enabling STA.
[16:44:06]mode : sta(84:cc:a8:a1:06:7f)
[16:44:06]add if0
[16:44:06][V][wifi_esp8266:534]: Event: Changed Mode old=OFF new=STA
[16:44:06]wifi evt: 8
[16:44:06]sleep disable
[16:44:12]scandone
[16:44:12][V][wifi_esp8266:492]: Event: Changed AuthMode old=OPEN new=WPA2 PSK
[16:44:12]wifi evt: 2
[16:44:12]scandone
[16:44:13]state: 0 -> 2 (b0)
[16:44:13]state: 2 -> 3 (0)
[16:44:13]state: 3 -> 5 (10)
[16:44:13]add 0
[16:44:13]aid 36
[16:44:13]cnt
[16:44:13]
[16:44:13]connected with SDNet, channel 11
[16:44:13]dhcp client start...
[16:44:13][V][wifi_esp8266:468]: Event: Connected ssid='SDNet' bssid=66:22:32:F0:6D:32 channel=11
[16:44:13]wifi evt: 0
[16:44:15]ip:192.168.10.133,mask:255.255.255.0,gw:192.168.10.1
[16:44:15][V][wifi_esp8266:507]: Event: Got IP static_ip=192.168.10.133 gateway=192.168.10.1 netmask=255.255.255.0
[16:44:15]wifi evt: 3
[16:44:15][C][web_server:116]: Setting up web server...
[16:44:15][C][sntp:028]: Setting up SNTP...
[16:44:15][C][ota:093]: Over-The-Air Updates:
[16:44:15][C][ota:094]:   Address: water-meter-dmini.local:8266
[16:44:15][C][ota:097]:   Using Password.
[16:44:15]
[16:44:15]wMBus-lib: CC1101 version '20'
[16:44:15]wMBus-lib: CC1101 initialized
[16:44:15][I][app:062]: setup() finished successfully!
[16:44:15][I][app:102]: ESPHome version 2023.8.3 compiled on Sep 14 2023, 16:43:33
[16:44:15][I][app:104]: Project Diehl_IZAR_RC_868.Watermeter-Test version 2.0.8
[16:44:15][C][logger:301]: Logger:
[16:44:15][C][logger:302]:   Level: VERBOSE
[16:44:15][C][logger:303]:   Log Baud Rate: 115200
[16:44:15][C][logger:305]:   Hardware UART: UART0
[16:44:15][C][logger:309]:   Level for 'wmbus': VERBOSE
[16:44:15][C][logger:309]:   Level for 'wMBus-lib': VERBOSE
[16:44:15][C][logger:309]:   Level for 'sensor': WARN
[16:44:15][C][logger:309]:   Level for 'sensor.filter': WARN
[16:44:15][C][logger:309]:   Level for 'text_sensor': WARN
[16:44:15][C][logger:309]:   Level for 'api.service': ERROR
[16:44:15][C][logger:309]:   Level for 'json': ERROR
[16:44:15][C][logger:309]:   Level for 'mqtt': WARN
[16:44:15][C][logger:309]:   Level for 'scheduler': ERROR
[16:44:15][C][logger:309]:   Level for 'wifi': WARN
[16:44:15][C][logger:309]:   Level for 'component': ERROR
[16:44:15][C][logger:309]:   Level for 'api': WARN
[16:44:15][D][sntp:077]: Synchronized time: 2023-09-14 17:44:15
[16:44:15][W][system:209]: Synchronized sntp clock
[16:44:15][C][template.sensor:022]: Template Sensor 'Boot counter'
[16:44:15][C][template.sensor:022]:   State Class: 'measurement'
[16:44:15][C][template.sensor:022]:   Unit of Measurement: ''
[16:44:15][C][template.sensor:022]:   Accuracy Decimals: 0
[16:44:15][C][template.sensor:023]:   Update Interval: 60.0s
[16:44:16][C][template.text_sensor:020]: Template Sensor 'Last Boot timestamp'
[16:44:16][C][template.text_sensor:020]:   Icon: 'mdi:clock-start'
[16:44:16][C][template.switch:068]: Template Switch 'Boot Counter reset'
[... 76 lines cut ...]
[16:44:16][C][wmbus:448]:       Device Class: 'signal_strength'
[16:44:16][C][wmbus:448]:       State Class: 'measurement'
[16:44:16][C][wmbus:448]:       Unit of Measurement: 'lqi'
[16:44:16][C][wmbus:448]:       Accuracy Decimals: 0
[16:44:16][C][wmbus:448]:     Sensor '0x00 CC1101 RSSI'
[16:44:16][C][wmbus:448]:       Device Class: 'signal_strength'
[16:44:16][C][wmbus:448]:       State Class: 'measurement'
[16:44:16][C][wmbus:448]:       Unit of Measurement: 'dBm'
[16:44:16][C][wmbus:448]:       Accuracy Decimals: 0
[16:44:16][C][wmbus:448]:       Icon: 'mdi:rss'
[16:44:22]wMBus-lib: Processing T1 A frame
[16:44:22]wMBus-lib:  Error during decoding '3 out of 6'
[16:44:23]fpm close 3
[16:44:23]pm open,type:0 0
[16:44:37]wMBus-lib: Processing T1 A frame
[16:44:37]wMBus-lib:  Error during decoding '3 out of 6'
[16:44:41]wMBus-lib: Processing T1 A frame
[16:44:41]wMBus-lib:  Error during decoding '3 out of 6'
[16:44:43][D][esp8266.preferences:238]: Saving preferences to flash...
[16:44:57]wMBus-lib: Processing T1 A frame
[16:44:57]wMBus-lib:  Error during decoding '3 out of 6'
[16:45:03]wMBus-lib: Processing T1 A frame
[16:45:03]wMBus-lib:  Error during decoding '3 out of 6'

```

zibous, 2023-09-14:
ok. Log and init CC1101 looks good.
Actually, after a while you should get a telegram from your water meter or from another...

```
[16:44:37]wMBus-lib: Processing T1 A frame
[16:44:37]wMBus-lib:  Error during decoding '3 out of 6'
```

This message comes from the wmbus lib and does not necessarily have to be an error. 
But in this case only Szczepan Leon/esphome can help you.

> https://github.com/SzczepanLeon/esphome-components

Woodpeckercz, 2023-09-14:
Thanks for taking the time to help me with this.

Woodpeckercz, 2023-09-14:
ok it took about 20 minutes buy I finally got something.
```
[17:27:55][D][wmbus:179]: Meter ID [0xA0D7BAD3] RSSI: -90 dBm LQI: 130 Mode: T1 not found in configuration T: 2944304CD3BAD7A00604A231C500131A0633E9E98ED4A267007D510AEFF9EC5A05B298D85BD557ED13BC (42)
```

closed 2023-10-13 by zibous