zibous/ha-watermeter #21: Corrupted backtrace

Flow76320, 2023-09-14:
Hi @zibous!

Thanks for this work/tuto, this works like a charm!

Actually, this worked like a charm for months... But since I wanted to add a case for everything, the ESP seems to be corrupted. My soldering was quite poor for the external antena and it just broke. I made a new one and then... 
```
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (23870) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (23870) task_wdt:  - loopTask (CPU 1)
E (23870) task_wdt: Tasks currently running:
E (23870) task_wdt: CPU 0: IDLE
E (23870) task_wdt: CPU 1: loopTask
E (23870) task_wdt: Aborting.

abort() was called at PC 0x40103a48 on core 0

Backtrace:0x40083915:0x3ffbe9cc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
``` 
It reboots all the time.

I've checked my soldering, made new ones, changes the CC1101. 
Interesting point is that when I remove 3v3 for the CC1101, boot loops stop.

Any idea on how to fix this?

Thanks 🙏

zibous, 2023-09-14:
> Actually, this worked like a charm for months... But since I wanted to add a case for everything, the ESP seems to be corrupted. My soldering was quite poor for the external antena and it just broke. I made a new one and then...

Hi,

How do you flash the firmware ?

Using another microUSB cable helped.

Flow76320, 2023-09-14:
Hi,

Thanks for your fast answer.

I flashed with EspHome from HA, and reset with esptool.py from Espressif
website.

I changed the USB cable too

zibous, 2023-09-14:
> I flashed with EspHome from HA, and reset with esptool.py from Espressif website.

Ohh, This explains to some, HA ESPHome is currently unstable and there are some reports that flashing devices and creating firmware is not working !

Here you can test firmware. I would flash with "ESPHOME Flasher".
> Firmware : https://github.com/zibous/ha-watermeter/tree/master/esphome/testcases

> ESPHOME Flasher: https://github.com/esphome/esphome-flasher/releases

Flow76320, 2023-09-14:
This is good to know.

I tried with https://web.esphome.io/, click "prepare for first use" which
was fine.
Next I used "Install" with your firmware water-meter-esp32-izar-test.bin
<https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/water-meter-esp32-izar-test.bin>

Got:
```
[ 7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz,
APB: 80000000 Hz [ 40][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino
Event: 0 - WIFI_READY [ 136][V][WiFiGeneric.cpp:338] _arduino_event_cb():
STA Started [ 137][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event:
2 - STA_START [ 1033][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP
Started [ 3285][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10
- AP_START [ 3289][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped [
3292][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP [
3299][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started [
3304][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 -
AP_START [ 8566][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done:
ID: 128, Status: 0, Results: 7 [ 8566][D][WiFiGeneric.cpp:929]
_eventCallback(): Arduino Event: 1 - SCAN_DONE [
10919][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID:
Flo_WIFI, BSSID: c0:f4:e6:66:cf:d3, Channel: 5, Auth: WPA2_PSK [
10921][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 -
STA_CONNECTED [ 11923][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got
New IP:192.168.1.10 [ 11924][D][WiFiGeneric.cpp:929] _eventCallback():
Arduino Event: 7 - STA_GOT_IP [ 11927][D][WiFiGeneric.cpp:991]
_eventCallback(): STA IP: 192.168.1.10, MASK: 255.255.255.0, GW:
192.168.1.254 [ 11944][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP
Stopped [ 11945][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event:
11 - AP_STOP E (34240) task_wdt: Task watchdog got triggered. The following
tasks did not reset the watchdog in time: E (34240) task_wdt: - loopTask
(CPU 1) E (34240) task_wdt: Tasks currently running: E (34240) task_wdt:
CPU 0: IDLE E (34240) task_wdt: CPU 1: loopTask E (34240) task_wdt:
Aborting. abort() was called at PC 0x400fe918 on core 0
Backtrace:0x400838d9:0x3ffbeadc
|<-CORRUPTED
```

If I install directly without "prepare for first use", I got:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT) invalid header:
0x00000000 invalid header: 0x00000000 invalid header: 0x00000000 invalid
header: 0x00000000 invalid header: 0x00000000 invalid header:
0x00000000 invalid
header: 0x00000000 invalid header: 0x00000000 ets Jul 29 2019 12:21:46
```
Meanwhile, I also changed the AZ delivery board...

After the card change, I had another try for Esp Home / HA v2023.8.3. It
says the code is successfully uploaded, but no sensor nore the manual IP
address I configured is used. Very strange behaviour.

Thanks again for your help

zibous, 2023-09-15:
> After the card change, I had another try for Esp Home / HA v2023.8.3

Esp Home / HA v2023.8.3 do not work, try the test firmware from:

>Firmware : https://github.com/zibous/ha-watermeter/tree/master/esphome/testcases

I build this with ESPHOME Docker.

Flow76320, 2023-09-15:
Hi,

Yes I tried your firmware. The logs are those from my previous comment. I
installed it with web esp home

zibous, 2023-09-15:
> Yes I tried your firmware. The logs are those from my previous comment. I installed it with web esp home
Negativ, do not use "web esp home" use the ESPHOME Flasher.

zibous, 2023-09-15:
New Testcase using https://web.esphome.io/

- Connect to the device
- Prepare for first use
- When finish set the local WIFI and "Visit Device"
- OTA Update water-meter-esp32-izar-test.bin

Works for me.

[image]

Result:

[image]

Flow76320, 2023-09-17:
Hi,

The flash / update OTA is successful. The device is stable when I keep the
CC1101 unplugged.

Just after the antenna plug in and device reboot, the device is looping
again because of corruption
```
[ 10784][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 -
STA_CONNECTED [ 11791][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got
New IP:192.168.1.10 [ 11791][D][WiFiGeneric.cpp:929] _eventCallback():
Arduino Event: 7 - STA_GOT_IP [ 11794][D][WiFiGeneric.cpp:991]
_eventCallback(): STA IP: 192.168.1.10, MASK: 255.255.255.0, GW:
192.168.1.254 [ 11809][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP
Stopped [ 11810][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event:
11 - AP_STOP E (33946) task_wdt: Task watchdog got triggered. The following
tasks did not reset the watchdog in time: E (33946) task_wdt: - loopTask
(CPU 1) E (33946) task_wdt: Tasks currently running: E (33946) task_wdt:
CPU 0: IDLE E (33946) task_wdt: CPU 1: loopTask E (33946) task_wdt:
Aborting. abort() was called at PC 0x400fe918 on core 0
Backtrace:0x400838d9:0x3ffbeadc
|<-CORRUPTED ELF file SHA256: 0000000000000000 Rebooting... ets Jul 29 2019
12:21:46
```

After the first use preparation and antenna plugged in, no such logs.

zibous, 2023-09-17:
> The flash / update OTA is successful. The device is stable when I keep the CC1101 unplugged.

In any case, it looks like a bad connection to CC1101, misplaced pins, or a failed CC1101.
see: https://github.com/SzczepanLeon/esphome-components/issues/17

closed 2023-10-13 by zibous

Flow76320, 2023-10-20:
Hi @zibous,

Tell me if you want me to open a new issue.
I've received and tested a new E07-900MBL-01 rev v.1.1 to replace the CC1101 cards, tested 2 ESP32, and I still got the corruptions.

After some investigations, they only appear when calling 
```
wmbus:
  mosi_pin: GPIO23  ## SI:   braun   3: MOSI Attached to Hardware SPI controller MOSI SPI Interface
  miso_pin: GPIO19  ## SO:   grün    5: MISO Attached to Hardware SPI controller MISO SPI Interface
  clk_pin: GPIO18   ## SCLK: violett 4: SCK  Attached to Hardware SPI controller CLK
  cs_pin: GPIO05    ## CSN:  orange  8: CSN  Attached to Hardware SPI controller
  gdo0_pin: GPIO16  ## GD00: gelb    7: RX Clock output. High Impedance !
  gdo2_pin: GPIO17  ## GD02: weiss   6: TX FIFO status signals. High Impedance !
  log_unknown: False
```

I've built the testing code with Web ESP Home and from Docker container, then it can't be related to the compiler.
The pins are good, are I've tested providing power to the network card by the integrated USB or from 3.3V of ESP32.
Cables are all brand new too.

Here is the wired setup
[image]

By any chance, may you have another idea?