zibous/ha-watermeter #22: Help with R3 Diehl meters?!

alecsandes, 2023-10-19:
Hi @zibous,

I addressing to you as I haven't find similar issues I face ... and you have made the most explicit tutorials on the matter.

I have this meter: 
https://www.diehl.com/metering/en/products-solutions/products-services/water-metering/aries-is-at/
it is stated that it works with R3 radio mode: Frequency |   | MHz | 868.95 (R3 mode) and 434.47 (R3 mode)**
** 434 MHz available for DN15/110 Q3=2.5. Other variants on demand.

Now I followed your tutorial here: https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-with-rtl-sdr.md
Until the part with [Configuration rtlwmbus && wmbusmeters](https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-with-rtl-sdr.md#configuration-rtlwmbus--wmbusmeters)
As even at this point, the output was different from what you presented.

So by running the command:
sudo wmbusmeters --debug --t1 rtlwmbus

I get a lot of:
`(main) regular reset of rtlwmbus  will happen every 82800 seconds (wmbus) no alarm (expected activity) for rtlwmbus (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds No meters configured. Printing id:s of all telegrams heard! (serial) waiting for stop (serial) received ascii "T1;1;1;2023-09-13 22:24:58.000;105;123;215091CD;0x1944304ccd9150211404a233130013493c3e0e0290df3aaf95bc<0A>T1;1;1;2023-09-13 22:24:58.000;141;115;215091CD;0x1944304ccd9 ![204155722-9746eb5f-ce6b-455d-bda6-2f8f57cf0a21]x 150211404a233130013493c3e0e0290df3aaf95bc<0A>" (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-09-13 22:24:58.000;105;123;215091CD;0x1944304ccd9150211404a233130013493c3e0e0290df3aaf95bc<0A>T1;1;1;2023-09-13 22:24:58.000;141;115;215091CD;0x1944304ccd9150211404a233130013493c3e0e0290df3aaf95bc<0A>" (rtlwmbus) received full frame (meter) no meter handled  checking 0 templates. (diehl) preprocess necessary SAP_PRIOS (diehl) Pre-processing: setting device type to water meter for SAP PRIOS (wmbus) parseDLL @0 26 (wmbus) parseELL @10 16 (wmbus) parseNWL @10 16 (wmbus) parseAFL @10 16 (wmbus) parseTPL @10 16 (diehl) preprocess necessary SAP_PRIOS (diehl) Pre-processing: setting device type to water meter for SAP PRIOS (wmbus) parseDLL @0 26 (telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=215091cd VER=00 TYPE=07 (Water meter) (driver izar) DEV=rtlwmbus[00000001] RSSI=105 (wmbus) parseELL @10 16 (wmbus) parseNWL @10 16 (wmbus) parseAFL @10 16 (wmbus) parseTPL @10 16 (telegram) TPL CI=a2 Received telegram from: 215091cd manufacturer: (SAP) Sappel (0x4c30) type: Water meter (0x07) ver: 0x00 device: rtlwmbus[00000001] rssi: 105 dBm driver: izar (wmbus) 000   : 19 length (25 bytes) (wmbus) 001   : 44 dll-c (from meter SND_NR) (wmbus) 002   : 304c dll-mfct (SAP) (wmbus) 004   : cd915021 dll-id (215091cd) (wmbus) 008   : 00 dll-version (wmbus) 009   : 07 dll-type (Water meter) (wmbus) 010   : a2 tpl-ci-field (Mfct specific) (wmbus) 011 C?: 33130013493C3E0E0290DF3AAF95BC mfct specific (wmbus) telegram from 215091cd ignored by all configured meters! (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-09-13 22:24:58.000;141;115;215091CD;0x1944304ccd9150211404a233130013493c3e0e0290df3aaf95bc<0A>" (rtlwmbus) received full frame (wmbus) skipping already handled telegram leng=26. `

I took one telegram 1944304ccd9150211404a233130013493c3e0e0290df3aaf95bc and ran in through the analyzer
[image]
See, no information inside  the telegram, it is not decoded.

Than I gave a try with ESP 8266 NodeMCU with CC1101 board. The issue here is that the board is strange, it has on it a number, 433 and I assume it is set for 433 MHz.
However, using the Szczepan's code I have built the reader. And the output is like this:

```
[00:24:38][D][wmbus:179]: Meter ID [0x2150A9F8] RSSI: -86 dBm LQI: 128 Mode: T1 not found in configuration T: 1944304CF8A950211404A203330013017241955F03ABC73A7840 (26)
[00:24:41][D][wmbus:179]: Meter ID [0x216049FD] RSSI: -96 dBm LQI: 132 Mode: T1 not found in configuration T: 1944304CFD4960211404A2731500130D3E27E125F1D92091DC0C (26)
[00:24:41][D][wmbus:179]: Meter ID [0x214FD95F] RSSI: -94 dBm LQI: 129 Mode: T1 not found in configuration T: 1944304C5FD94F211404A25313001378544D9642A77F66128E67 (26)
[00:24:42][D][wmbus:179]: Meter ID [0x215091DC] RSSI: -81 dBm LQI: 128 Mode: T1 not found in configuration T: 1944304CDC9150211404A2231300135EBE4BB1354343F689C8A6 (26)
[00:24:43][D][wmbus:179]: Meter ID [0x214FF95F] RSSI: -90 dBm LQI: 128 Mode: T1 not found in configuration T: 1944304C5FF94F211404A2431300137AEDBDC4D3BBB8C70D0485 (26)
[00:24:44][D][wmbus:179]: Meter ID [0x214FF95E] RSSI: -90 dBm LQI: 134 Mode: T1 not found in configuration T: 1944304C5EF94F211404A2431300136DC8B21D06C59D01299B87 (26)
[00:24:47][D][wmbus:179]: Meter ID [0x214FF7D6] RSSI: -64 dBm LQI: 128 Mode: T1 not found in configuration T: 1944304CD6F74F211404A233130013882B480D2D2B10E9030123 (26)
[00:24:55][D][wmbus:179]: Meter ID [0x21509410] RSSI: -93 dBm LQI: 129 Mode: T1 not found in configuration T: 1944304C109450211404A24313001370E246A22F6BA01C0C02CA (26)
```
The same output I had using the DVB-T tuner ... and still, the telegrams are not decoded ...
https://wmbusmeters.org/analyze/1944304C109450211404A24313001370E246A22F6BA01C0C02CA

Do you have an idea, where is the issue, in hardware, in R3 mode, in encrypted telegram?

TIA!

zibous, 2023-10-21:
@alecsandes 
> However, using the Szczepan's code I have built the reader. And the output is like this...

It actually looks pretty good, you get a lot of telegrams. 

https://github.com/wmbusmeters/wmbusmeters/blob/f99dbadf88b9d1127c079511ae1480fcc5851871/src/meter_izar.cc

Have you ever tried this?

```
data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=rtlwmbus
  donotprobe=/dev/ttyAMA0
  listento=c1,t1,s1
  meterfiles=/config/wmbusmeters/logs/meter_readings/
  meterfilesaction=overwrite
  logtelegrams=true
  format=json
  logfile=/config/wmbusmeters/wmbusmeters.log
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=MainWater1
    driver=auto
    id=00238332
    key=00000000000000000000000000000000
mqtt:
  host: 192.168.x.xx
  port: 1883
  user: xxx
  password: xxx 
```

alecsandes, 2023-10-23:
Thank you very much for your reply!

I have multiple telegrams as I live in a block of apartments, 32 x 4 meters x 6 blocks (i don't know to which extent I receive the telegrams)

No, I haven't tried to define since I get issues with decoding the key.
If I understand right, if I put the "00000" key in here: https://wmbusmeters.org/analyze/1944304cfaa950211404a2231400134a2c75e61a25bb0d891f7d:auto:00000000000000000000000000000000 
it should decode my telegram, but it doesn't ...

What I am thinking also is to use https://github.com/ZeWaren/izar-prios-smart-meter-collector/blob/master/PC/prios_key_cracker.c
but in the case of Zewaren, he has 28 sets of chars, I have 26 ...

At this moment I am fighting with this error though
[image]

I need somehow to make it in MacOS environment

zibous, 2023-10-24:
> but in the case of Zewaren, he has 28 sets of chars, I have 26 ...

For wmbismeters you need a key to read the meter. 
But I don't have any information about how you can find it.

Maybe you can use this:
https://github.com/ZeWaren/izar-prios-smart-meter-collector/blob/master/PC/prios_key_cracker.c

closed 2024-04-02 by zibous

kmlZ, 2025-01-01:
> Thank you very much for your reply!
>
> I have multiple telegrams as I live in a block of apartments, 32 x 4 meters x 6 blocks (i don't know to which extent I receive the telegrams)
>
> No, I haven't [...]

Have you managed to read data? Did you crack the key?