zibous/ha-watermeter #24: Build wmbusmeters Error "fatal error: libxml/tree.h: No such file or directory"

BenRoe, 2023-10-23:
Hello,
i use a Raspberry Pi with Raspbian OS Lite 32bit and followed the steps [here](https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md#build-wmbusmeters) to build the wmbusmeters.

But i get the error 
```bash
src/xmq.c:39:9: fatal error: libxml/tree.h: No such file or directory
   39 | #include<libxml/tree.h>
      |         ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:151: build/xmq.o] Error 1
```

Any idea how to fix this build error?

<details>
  <summary>Full log</summary>
  
```bash
pi@raspberrypi:~/wmbusmeters $ make && make test
New version number generates new build/version.h
Building 1.14.0-37-g6f0773a
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aes.cc -c -E > build/aes.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aes.cc -MMD -c -o build/aes.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aescmac.cc -c -E > build/aescmac.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aescmac.cc -MMD -c -o build/aescmac.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/bus.cc -c -E > build/bus.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/bus.cc -MMD -c -o build/bus.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/cmdline.cc -c -E > build/cmdline.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/cmdline.cc -MMD -c -o build/cmdline.o
In file included from /usr/include/c++/12/vector:70,
                 from src/units.h:22,
                 from src/config.h:21,
                 from src/cmdline.h:21,
                 from src/cmdline.cc:18:
/usr/include/c++/12/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(iterator, _Args&& ...) [with _Args = {const MeterInfo&}; _Tp = MeterInfo; _Alloc = std::allocator<MeterInfo>]’:
/usr/include/c++/12/bits/vector.tcc:439:7: note: parameter passing for argument of type ‘std::vector<MeterInfo>::iterator’ changed in GCC 7.1
  439 |       vector<_Tp, _Alloc>::
      |       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/12/vector:64:
In member function ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = MeterInfo; _Alloc = std::allocator<MeterInfo>]’,
    inlined from ‘std::shared_ptr<Configuration> parseNormalCommandLine(Configuration*, int, char**)’ at src/cmdline.cc:670:28:
/usr/include/c++/12/bits/stl_vector.h:1287:28: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<MeterInfo*, std::vector<MeterInfo> >’ changed in GCC 7.1
 1287 |           _M_realloc_insert(end(), __x);
      |           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/config.cc -c -E > build/config.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/config.cc -MMD -c -o build/config.o
In file included from /usr/include/c++/12/vector:70,
                 from src/units.h:22,
                 from src/config.h:21,
                 from src/config.cc:18:
/usr/include/c++/12/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(iterator, _Args&& ...) [with _Args = {const SpecifiedDevice&}; _Tp = SpecifiedDevice; _Alloc = std::allocator<SpecifiedDevice>]’:
/usr/include/c++/12/bits/vector.tcc:439:7: note: parameter passing for argument of type ‘std::vector<SpecifiedDevice>::iterator’ changed in GCC 7.1
  439 |       vector<_Tp, _Alloc>::
      |       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/12/vector:64:
In member function ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = SpecifiedDevice; _Alloc = std::allocator<SpecifiedDevice>]’,
    inlined from ‘bool handleDeviceOrHex(Configuration*, std::string)’ at src/config.cc:386:42:
/usr/include/c++/12/bits/stl_vector.h:1287:28: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<SpecifiedDevice*, std::vector<SpecifiedDevice> >’ changed in GCC 7.1
 1287 |           _M_realloc_insert(end(), __x);
      |           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
/usr/include/c++/12/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(iterator, _Args&& ...) [with _Args = {const MeterInfo&}; _Tp = MeterInfo; _Alloc = std::allocator<MeterInfo>]’:
/usr/include/c++/12/bits/vector.tcc:439:7: note: parameter passing for argument of type ‘std::vector<MeterInfo>::iterator’ changed in GCC 7.1
  439 |       vector<_Tp, _Alloc>::
      |       ^~~~~~~~~~~~~~~~~~~
In member function ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = MeterInfo; _Alloc = std::allocator<MeterInfo>]’,
    inlined from ‘void parseMeterConfig(Configuration*, std::vector<char>&, std::string)’ at src/config.cc:204:28:
/usr/include/c++/12/bits/stl_vector.h:1287:28: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<MeterInfo*, std::vector<MeterInfo> >’ changed in GCC 7.1
 1287 |           _M_realloc_insert(end(), __x);
      |           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/dvparser.cc -c -E > build/dvparser.o.src
[... 127 lines cut ...]
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_im871a.cc -c -E > build/wmbus_im871a.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_im871a.cc -MMD -c -o build/wmbus_im871a.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_cul.cc -c -E > build/wmbus_cul.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_cul.cc -MMD -c -o build/wmbus_cul.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rtlwmbus.cc -c -E > build/wmbus_rtlwmbus.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rtlwmbus.cc -MMD -c -o build/wmbus_rtlwmbus.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rtl433.cc -c -E > build/wmbus_rtl433.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rtl433.cc -MMD -c -o build/wmbus_rtl433.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_simulator.cc -c -E > build/wmbus_simulator.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_simulator.cc -MMD -c -o build/wmbus_simulator.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rawtty.cc -c -E > build/wmbus_rawtty.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rawtty.cc -MMD -c -o build/wmbus_rawtty.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rc1180.cc -c -E > build/wmbus_rc1180.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_rc1180.cc -MMD -c -o build/wmbus_rc1180.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_utils.cc -c -E > build/wmbus_utils.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/wmbus_utils.cc -MMD -c -o build/wmbus_utils.o
g++ -I/usr/include/libxml2 -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/xmq.c -c -E > build/xmq.o.src
src/xmq.c:39:9: fatal error: libxml/tree.h: No such file or directory
   39 | #include<libxml/tree.h>
      |         ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:151: build/xmq.o] Error 1
```

</details>

zibous, 2023-10-23:
Do you update your system ?
see: [Update system](https://github.com/zibous/ha-watermeter/blob/master/docs/wmbusmeters-nano_cul.md#update-system)

```
The following is required before installation:

$ sudo apt-get update
$ sudo apt-get install build-essential libncurses-dev cmake automake make 
$ sudo apt-get install -y mosquitto mosquitto-clients librtlsdr-dev
```

BenRoe, 2023-10-23:
Yes did. Probably found the problem.
In the instructions the `libxml2-dev` is missing.

BenRoe, 2023-10-23:
Created a PR #25

closed 2023-10-23 by BenRoe