zibous/ha-watermeter #26: Cannot Prepare wm-esp32.yaml firmware

BMWfan, 2023-11-09:
Hello,

I cloned the repository and added this YAML file to my ESPHome Docker container in the /config directory:
https://raw.githubusercontent.com/zibous/ha-watermeter/master/esphome/wm-esp32.yaml

Whenever I try to install it and choose the "Plug into this computer" option, during the download preparation, I receive the following output:
```
INFO ESPHome 2023.10.6
INFO Reading configuration /config/wm-esp32.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
WARNING GPIO5 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

wifi: [source /config/wm-esp32.yaml:237]
  networks: 
    - ssid: !secret ssid2_pswd
      password: !secret ssid2_pswd
      priority: 1
    - ssid: !secret ssid2_pswd
      password: !secret ssid2_pswd
      priority: 2
    - ssid: !secret ssid2_pswd
      password: !secret ssid2_pswd
      priority: 3
  domain: !secret domain
  ap: 
    
    Must be string, got <class 'esphome.helpers.EInt'>. did you forget putting quotes around the value?.
    password: 1234
script: [source /config/wm-esp32.yaml:532]
  - id: set_status_message
    then: 
      - lambda: |-
          int msgcode = int(id(cc1101_state));
          std::string message = id(cc1101_state_message)[msgcode];
          ESP_LOGD("wmbus", "cc1101 state message: %s, error code: %d", message.c_str(), msgcode);
          id(watermeter_status_message).publish_state(message);
  - id: post_device_state
    then: 
      - if: 
          condition: 
            lambda: return "false =='true'";
          then: 
            - logger.log: 
                tag: system
                level: INFO
                format: Post new data to the <redacted>water-meter-esp, service enabled: false
            - http_request.post: 
                
                URL must have a URL scheme and host.
                url: <redacted>/water-meter-esp
                verify_ssl: False
                headers: 
                  Content-Type: application/json
                json: |-
                  root["device"] = "water-meter-esp";
                  root["waterdisplay"] = id(waterdisplay).state;
                  root["current"] = id(id(watercurrent)).state;
                  root["hour"] = id(waterhour).state;
                  root["day"] = id(waterday).state;
                  root["yesterday"] = id(wateryesterday).state;
                  root["week"] = id(waterweek).state;
                  root["month"] = id(watermonth).state;
                  root["lastmonth"] = id(waterhour).state;
                  root["year"] = id(wateryear).state;
                  root["bootcounter"] = id(bootcounter).state;
                  root["version"] = "water-meter-esp 2.0.9";
                  root["time"] = id(time_sntp).now().strftime("%Y-%m-%d %H:%M:%S");
                on_response: 
                  then: 
                    - lambda: |-
                        if (status_code == 200) {
                           ESP_LOGI ("system", "HTTP POST o.k");
                        } else {
                            ESP_LOGI ("system", "HTTP POST failed");
                        }
```

How can i solve that?

zibous, 2023-11-09:
hi 

> wifi: [source /config/wm-esp32.yaml:237]
  networks: 
    - ssid: !secret ssid2_pswd
      password: !secret ssid2_pswd
      priority: 1
    - ssid: !secret ssid2_pswd
      password: !secret ssid2_pswd
      priority: 2
    - ssid: !secret ssid2_pswd
      password: !secret ssid2_pswd
      priority: 3
  domain: !secret domain
  ap: 
    
    Must be string, got <class 'esphome.helpers.EInt'>. did you forget putting quotes around the value?.
    password: 1234

I suspect it's because of the secrets settings. Entries must be accompanied by quotes.

```
# Your Wi-Fi's SSID and password
wifi_ssid:  "*****"
wifi_password: "*****"

# Passwords for appl, ota and homeassistant
app_pswd: "****"
ota_pswd: "****"
```

BMWfan, 2023-11-10:
@zibous thank you for your reply.

I used the wm-esp32_test.yaml to get the right watermeterid and added that in the secret.yaml and installed wm-esp32-simple.yaml. 

How long does it take till i would see some informations?
[image]

I installed it 1 hour ago..

zibous, 2023-11-11:
> I installed it 1 hour ago..

Usually immediately or after 1 minute. But what is important is the log on startup, which shows whether the CC1101 is connected correctly:

```log
[15:14:57][C][wmbus:394]: wM-Bus v2.2.29:
[15:14:57][C][wmbus:411]:   CC1101 SPI bus:
[15:14:57][C][wmbus:412]:     MOSI Pin: GPIO23
[15:14:57][C][wmbus:413]:     MISO Pin: GPIO19
[15:14:57][C][wmbus:414]:     CLK Pin:  GPIO18
[15:14:57][C][wmbus:415]:     CS Pin:   GPIO5
[15:14:57][C][wmbus:416]:     GDO0 Pin: GPIO21
[15:14:57][C][wmbus:417]:     GDO2 Pin: GPIO22
```

You can also use the test image for the test:
see: 
https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/README.md
https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/water-meter-esp32-izar-test.bin

Another source of error could be that the ESP is too far away from the water meter or the telegram signal is disturbed.

BMWfan, 2023-11-12:
@zibous ok, i used water-meter-esp32-izar-test and got the following output:
```
Time	level	Tag	Message
10:44:12	[V]	[mdns:119]	
  - _http, _tcp, 80
10:44:12	[C]	[ota:097]	
Over-The-Air Updates:
10:44:12	[C]	[ota:098]	
  Address: water-meter-esp32-izar-test.local:3232
10:44:12	[C]	[ota:101]	
  Using Password.
10:44:12	[C]	[wifi_signal.sensor:009]	
WiFi Signal 'Device WLAN RSSI Quality'
10:44:12	[C]	[wifi_signal.sensor:009]	
  Device Class: 'signal_strength'
10:44:12	[C]	[wifi_signal.sensor:009]	
  State Class: 'measurement'
10:44:12	[C]	[wifi_signal.sensor:009]	
  Unit of Measurement: '%'
10:44:12	[C]	[wifi_signal.sensor:009]	
  Accuracy Decimals: 0
10:44:12	[V]	[wifi_signal.sensor:009]	
  Unique ID: 'a842e3480748-wifisignal'
10:44:12	[C]	[wmbus:394]	
wM-Bus v2.2.31:
10:44:12	[C]	[wmbus:411]	
  CC1101 SPI bus:
10:44:12	[C]	[wmbus:412]	
    MOSI Pin: GPIO23
10:44:12	[C]	[wmbus:413]	
    MISO Pin: GPIO19
10:44:12	[C]	[wmbus:414]	
    CLK Pin:  GPIO18
10:44:12	[C]	[wmbus:415]	
    CS Pin:   GPIO5
10:44:12	[C]	[wmbus:416]	
    GDO0 Pin: GPIO16
10:44:12	[C]	[wmbus:417]	
    GDO2 Pin: GPIO17
10:44:12	[C]	[wmbus:424]	
  Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, dme07, elf, evo868, fhkvdataiii, hydrocalm3, hydrus, iperl, itron, izar, mkradio3, mkradio4, qheat, qwater, sharky774, topaseskr, ultrimis, unismart, vario451
10:44:12	[C]	[wmbus:442]	
  Meter:
10:44:12	[C]	[wmbus:443]	
    ID: 0 [0x00000000]
10:44:12	[C]	[wmbus:444]	
    Type: izar
10:44:12	[C]	[wmbus:445]	
    Mode: T1
10:44:12	[C]	[wmbus:446]	
    Key: ''
10:45:01	[D]	[wmbus:184]	
Meter ID [0x94518450] RSSI: -58 dBm LQI: 128 Mode: C1 not found in configuration T: 4B4468505084519495377221327700A85CFE0730003005E3601F0C1AB03805E2E6B53917077E3C295E0E04D56DF0CB1C17CA7FE5282ECACF84EA191F11D97006B164F65CD5BCC50F853AD352 (76)
10:46:05	[D]	[wmbus:184]	
Meter ID [0x67504656] RSSI: -84 dBm LQI: 139 Mode: C1 not found in configuration T: 734468505646506728048C0090900F002C253C020DA0D0908634956EC80D7AFF0050071090193E1415BB6B370F8C43F002E6B103331F1E40ADE6A7FFEB9E3856F16F1BE5B508943A1C3EA6962FD317FAD25695213D4231D809A00F646E1F60288CFFB70D4F78FD3B94BBE3F8F781ED41A11F23E6 (116)
10:46:14	[D]	[wmbus:184]	
Meter ID [0x94518449] RSSI: -61 dBm LQI: 128 Mode: C1 not found in configuration T: 4B44685049845194953772366997942423FE06570030051AAA94041E0CF9CB66E9D7B4E243C8EDF4598F482021573E9BB92652E97F0F402D5660E2EF0F65A2E98E03BA7FA5EEA50FC8F51E4B (76)
10:46:47	[D]	[wmbus:184]	
Meter ID [0x00804949] RSSI: -67 dBm LQI: 132 Mode: C1 not found in configuration T: 4B44685049498000953772624610682423FE070E003005AC0C55A54A39CDE607A81B1C6CBDE04CE7447A22E41384C2FE45F4AAA346ABDE9B6CDEA0479645603F18D1376CD933E10F27D03ED8 (76)
10:48:02	[D]	[wmbus:184]	
Meter ID [0x95115710] RSSI: -74 dBm LQI: 131 Mode: C1 not found in configuration T: 4B44685010571195953772597197942423FE06B600300590805DC86417274383E28BF7B4F1438E46E177D1C0C7D59CEF9B107CD5992BF993467C5E59EFB65BAC85E5E31AA06EE10F1BA437E8 (76)
10:49:55	[D]	[wmbus:184]	
Meter ID [0x67504656] RSSI: -81 dBm LQI: 149 Mode: C1 not found in configuration T: 434468505646506728048C2091900F002D651D021D009727B3092C4944067AFF00200710E8E2FE1ADE09F741D1413505E21B589F6452FE47CEB7EBF59DD16EE66AE0E778 (68)
10:50:11	[D]	[wmbus:184]	
Meter ID [0x94518449] RSSI: -64 dBm LQI: 129 Mode: C1 not found in configuration T: 4B44685049845194953772366997942423FE06580030056CC5E9CFC5BFEB9782116B48BF73A832967899E900548DC0066AE638517383485E2A6F968A6963706A6FECCCB3A314E80F23432BB5 (76)
10:52:12	[D]	[wmbus:184]	
Meter ID [0x2B23EE06] RSSI: -76 dBm LQI: 255 Mode: C1 not found in configuration T: 5A4448C006EE232B2A6FE42A64EE0150B9FC0E8800600A5841FBAB8B14412F0FB6C1C5EF6EB7A0B8E4DEC8BDC785FF5F1CC20495DDD7B4F378A4BD60BD9CC0312BD1AFC48153B01FC22C63402B946D06713A47E3E83EB942AAD859 (91)
10:52:56	[D]	[wmbus:184]	
Meter ID [0x94518428] RSSI: -72 dBm LQI: 139 Mode: C1 not found in configuration T: 4B4468502884519495377220327700A85CFE07BE00300588E2C764D8B67B3C522896209D13427C1CA579C343ADDCCAAADC13AF5AEF4F41EB7231C08B718FB8A8517F4518180BC80FFEC49615 (76)
10:53:05	[D]	[wmbus:184]	
[... 20 lines cut ...]
Synchronized sntp clock
10:59:04	[D]	[wmbus:184]	
Meter ID [0x00804949] RSSI: -67 dBm LQI: 131 Mode: C1 not found in configuration T: 4B44685049498000953772624610682423FE071100300588C4AE3FD8C55ACDFD3844259605DC0C3B1FD8D8E2FFBFCD6FC430DA57D3592B14D9B12ACBBE2B4C6E952E30AC22D5580F21432014 (76)
10:59:12	[D]	[wmbus:184]	
Meter ID [0x95110300] RSSI: -61 dBm LQI: 128 Mode: C1 not found in configuration T: 4B44685000031195953772326997942423FE06F90030056F317CD7FB265B2E7E504EFC4860E1737704A3FAA75B597782E316BD90669A7CAE65410A6950C7B9297692E6A25086490FC5285FCE (76)
10:59:26	[D]	[wmbus:184]	
Meter ID [0x03400001] RSSI: -81 dBm LQI: 215 Mode: C1 not found in configuration T: 4B446870010040032A6FE4CAC824D0D08FF81DEC00C01513A3FA430745197D6903E0BEEE9D0318375B4094E73F4A99E38D2F4B395366F76180F47C3F59822A4F120081D1E9F0FD2781BAE4D1 (76)
10:59:59	[D]	[wmbus:184]	
Meter ID [0x95115710] RSSI: -77 dBm LQI: 132 Mode: C1 not found in configuration T: 4B44685010571195953772597197942423FE06B900300521BB6CC93D3B1C8FD7B6A127773646002F9E43BEFC027B461DFA931328EC8DC5720BF9BFBCDFC05DC530A27F24B577A30F387BBB76 (76)
11:00:11	[D]	[wmbus:184]	
Meter ID [0x95117783] RSSI: -72 dBm LQI: 130 Mode: C1 not found in configuration T: 4B4468508377119595377215327700A85CFE0746003005EED7007A0D5D7DC40AB6587FBB4E0882C68C42B3A97E6044C4FC3E82EB170C772C7FCB1318ED23F2E95D5101016187780FE7A1E558 (76)
[... cut ...]
[https://wmbusmeters.org/analyze/4B44685000031195953772326997942423FE06A2003005817634153299B85B1D564485706FAA009669EDC8CC3E366F970056483B75358E9BA36A60F3DE459CA4776F2D63EF8BC80FC229F796](https://wmbusmeters.org/analyze/4B44685000031195953772326997942423FE06A2003005817634153299B85B1D564485706FAA009669EDC8CC3E366F970056483B75358E9BA36A60F3DE459CA4776F2D63EF8BC80FC229F796)

I'm also not to far away im my opinion as the water-meter-esp32-izar-test identifies my watermeter. But i still have the same output as described in my previous post.

zibous, 2023-11-12:
Looks good, the CC1101 works perfectly and you get lots of water meter telegrams.

`watermeterId: 0x95110300`  is o.k if this is your watermeter

But it looks like you need a decryption key for the mkradio4a device so that the data can be evaluated.
`023 CE: 817634153299B85B1D564485706FAA009669EDC8CC3E366F970056483B75358E9BA36A60F3DE459CA4776F2D63EF8BC8 encrypted?

see: https://github.com/wmbusmeters/wmbusmeters/issues/947

As long as you don't get any values from wmbusmeters.org/analyze/, you won't be able to see any data from the ESP32 either

evilmumi, 2023-11-23:
> > I installed it 1 hour ago..
> 
> Usually immediately or after 1 minute. But what is important is the log on startup, which shows whether the CC1101 is connected correctly:
> 
> ```
> [15:14:57][C][wmbus:394]: wM-Bus v2.2.29:
> [15:14:57][C][wmbus:411]:   CC1101 SPI bus:
> [15:14:57][C][wmbus:412]:     MOSI Pin: GPIO23
> [15:14:57][C][wmbus:413]:     MISO Pin: GPIO19
> [15:14:57][C][wmbus:414]:     CLK Pin:  GPIO18
> [15:14:57][C][wmbus:415]:     CS Pin:   GPIO5
> [15:14:57][C][wmbus:416]:     GDO0 Pin: GPIO21
> [15:14:57][C][wmbus:417]:     GDO2 Pin: GPIO22
> ```
> 
> You can also use the test image for the test: see: https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/README.md https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/water-meter-esp32-izar-test.bin
> 
> Another source of error could be that the ESP is too far away from the water meter or the telegram signal is disturbed.
i tried this and get `  8196][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.0.183, MASK: 255.255.255.0, GW: 192.168.0.1
E (26803) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (26803) task_wdt:  - loopTask (CPU 1)
E (26803) task_wdt: Tasks currently running:
E (26803) task_wdt: CPU 0: IDLE
E (26803) task_wdt: CPU 1: loopTask
E (26803) task_wdt: Aborting.

abort() was called at PC 0x400fefa0 on core 0

Backtrace:0x400838d9:0x3ffbeadc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4`

BMWfan, 2023-11-28:
> Looks good, the CC1101 works perfectly and you get lots of water meter telegrams.
> 
> `watermeterId: 0x95110300` is o.k if this is your watermeter
> 
> But it looks like you need a decryption key for the mkradio4a device so that the data can be evaluated. `023 CE: 817634153299B85B1D564485706FAA009669EDC8CC3E366F970056483B75358E9BA36A60F3DE459CA4776F2D63EF8BC8 encrypted?`
> 
> see: [wmbusmeters/wmbusmeters#947](https://github.com/wmbusmeters/wmbusmeters/issues/947)
> 
> As long as you don't get any values from wmbusmeters.org/analyze/, you won't be able to see any data from the ESP32 either. But esphome-components has no driver for the `mkradio4a` (if this is your watermeter device).
> 
> `Available drivers: amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, dme07, elf, evo868, fhkvdataiii, hydrocalm3, hydrus, iperl, itron, izar, mkradio3, mkradio4, qheat, qwater, sharky774, topaseskr, ultrimis, unismart, vario451`
> 
> More Info you can get from Szczepan Leon https://github.com/SzczepanLeon/esphome-components

I wrote to Techem to get the keys. Hopefully i see after that then the expected values.

closed 2024-04-02 by zibous

WulleWulleWapWap, 2024-10-28:
> Hello,
> 
> I cloned the repository and added this YAML file to my ESPHome Docker container in the /config directory: https://raw.githubusercontent.com/zibous/ha-watermeter/master/esphome/wm-esp32.yaml
> 
> Whenever I try to install it and choose the "Plug into this computer" option, during the download preparation, I receive the following output:
> 
> ```
> INFO ESPHome 2023.10.6
> INFO Reading configuration /config/wm-esp32.yaml...
> INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
> WARNING GPIO5 is a Strapping PIN and should be avoided.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
> Failed config
> 
> script: [source /config/wm-esp32.yaml:532]
>   - id: set_status_message
>     then: 
>       - lambda: |-
>           int msgcode = int(id(cc1101_state));
>           std::string message = id(cc1101_state_message)[msgcode];
>           ESP_LOGD("wmbus", "cc1101 state message: %s, error code: %d", message.c_str(), msgcode);
>           id(watermeter_status_message).publish_state(message);
>   - id: post_device_state
>     then: 
>       - if: 
>           condition: 
>             lambda: return "false =='true'";
>           then: 
>             - logger.log: 
>                 tag: system
>                 level: INFO
>                 format: Post new data to the <redacted>water-meter-esp, service enabled: false
>             - http_request.post: 
>                 
>                 URL must have a URL scheme and host.
>                 url: <redacted>/water-meter-esp
>                 verify_ssl: False
>                 headers: 
>                   Content-Type: application/json
>                 json: |-
>                   root["device"] = "water-meter-esp";
>                   root["waterdisplay"] = id(waterdisplay).state;
>                   root["current"] = id(id(watercurrent)).state;
>                   root["hour"] = id(waterhour).state;
>                   root["day"] = id(waterday).state;
>                   root["yesterday"] = id(wateryesterday).state;
>                   root["week"] = id(waterweek).state;
>                   root["month"] = id(watermonth).state;
>                   root["lastmonth"] = id(waterhour).state;
>                   root["year"] = id(wateryear).state;
>                   root["bootcounter"] = id(bootcounter).state;
>                   root["version"] = "water-meter-esp 2.0.9";
>                   root["time"] = id(time_sntp).now().strftime("%Y-%m-%d %H:%M:%S");
>                 on_response: 
>                   then: 
>                     - lambda: |-
>                         if (status_code == 200) {
>                            ESP_LOGI ("system", "HTTP POST o.k");
>                         } else {
>                             ESP_LOGI ("system", "HTTP POST failed");
>                         }
> ```
> 
> How can i solve that, and as i have a cold and warm meter how can i monitor both values?

could you solve that? i have a similar problem trying to validate my wm-esp32.yaml..

`INFO ESPHome 2024.10.2
INFO Reading configuration /config/wm-esp32.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@3.2.3
Failed config

script: [source /config/wm-esp32.yaml:534]
  - id: set_status_message
    then: 
      - lambda: |-
          int msgcode = int(id(cc1101_state));
          std::string message = id(cc1101_state_message)[msgcode];
          ESP_LOGD("wmbus", "cc1101 state message: %s, error code: %d", message.c_str(), msgcode);
          id(watermeter_status_message).publish_state(message);
  - id: post_device_state
    then: 
      - if: 
          condition: 
            lambda: return "false =='true'";
          then: 
            - logger.log: 
                tag: system
                level: INFO
                format: Post new data to the 192.168.0.231water-meter-esp, service enabled: false
            - http_request.post: 
                
                Expected a file scheme or a URL scheme with host.
                url: 192.168.0.231/water-meter-esp
                
                verify_ssl has moved to the base component configuration.
                verify_ssl: False
                headers: 
                  Content-Type: application/json`

Does anybody have a clue what i am doing wrong? I am thankfully for every hint or help i can get...