zibous/ha-watermeter #29: Cannot compile

CoolyMcCool, 2023-11-24:
Hallo,
i cannot compile the version 2.0.7:
With the wm-esp32-test i have the following issues:
```
esphome: None
  name: water-meter-esp
  comment: |-
    Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) - all watermeters
  project: 
    name: Diehl_IZAR_RC_868.Watermeter
    version: 2.0.7
  build_path: ./build/water-meter-esp
  on_boot: 
    - priority: 200.0
      then: 
        - globals.set: 
            id: boot_counter
            value: !lambda |-
              return id(boot_counter)+=1;
        - globals.set: 
            
            Couldn't find ID 'send_millisecond'. Please check you have defined an ID with that name in your configuration.
```
Do i need to install the regular one (wm-esp32) before?
with this i have the following issue:
```
/config/esphome/wm-esp32.yaml: In lambda function:
/config/esphome/wm-esp32.yaml:700:14: error: '$' was not declared in this scope
                 if ( ${service_enabled} == true){
              ^
/config/esphome/wm-esp32.yaml:700:15: error: expected ')' before '{' token
                 if ( ${service_enabled} == true){
               ^
               )
Compiling .pioenvs/water-meter-esp/libb0f/WiFiClientSecure/WiFiClientSecure.cpp.o
Compiling .pioenvs/water-meter-esp/libb0f/WiFiClientSecure/esp_crt_bundle.c.o
Compiling .pioenvs/water-meter-esp/libb0f/WiFiClientSecure/ssl_client.cpp.o
Compiling .pioenvs/water-meter-esp/lib2f0/HTTPClient/HTTPClient.cpp.o
*** [.pioenvs/water-meter-esp/src/main.cpp.o] Error 1
```

And with the simple i have the following issue:
```
  build_path: ./build/water-meter-esp
  on_boot: 
    - priority: -100.0
      then: 
        - globals.set: 
            id: boot_counter
            value: !lambda |-
              return id(boot_counter)+=1;
        - globals.set: 
            id: send_millisecond
            value: !lambda |-
              return millis();
        - logger.log: 
            level: WARN
            tag: system
            format: BOOTMESSAGE:water-meter-esp API is connected, Device ready!
        - component.update: bootcounter
    - priority: 800
      then: 
        - logger.log: 
            level: WARN
            tag: system
            format: |-
              BOOTMESSAGE:water-meter-esp hardware initialization of vital components is executed 
        - 
          Unable to find action with the name 'text_sensor.template.publish'.
          text_sensor.template.publish: 
```
I added every some credential where necessary, commented out the SYSLOG SERVICE, i deleted the additional WLANs, removed the WIFI domain and updated the Secrets.yaml (with blank wmbusmeters data, service target, and syslog)

I am really new to this stuff, and i am totally clueless,..
Thanks i advance
wasty

CoolyMcCool, 2023-11-24:
At least the test.yaml i could fix by adding the the above named send_millisecond global and setting the IP manually

CoolyMcCool, 2023-11-24:
restarted by scratch and now it works...

closed 2023-11-24 by CoolyMcCool

pacraf, 2024-12-03:
I know that it was long time ago this issue, but how did you solved "Couldn't find ID 'send_millisecond'" compilation error?