zibous/ha-watermeter #30: ESP32 SX1276

PauloMPereira, 2023-11-29:
Hello ,

This project is compatibly with ESP32 SX1276?
Device link: [Aliexpress](https://pt.aliexpress.com/item/1005005967763162.html?spm=a2g0o.order_list.order_list_main.5.248ccaa4dAZqB9&gatewayAdapt=glo2bra)

Also is it possible to confirm that my water meter is compatible?
Water Meter Images:
[image]
[image]

Best regards,
Paulo

SzczepanLeon, 2023-11-29:
> This project is compatibly with ESP32 SX1276?

No, only CC1101 is supported.

> Also is it possible to confirm that my water meter is compatible?

Should be. Please include photo from other side.

PauloMPereira, 2023-11-29:
Ok I saw the reference and it looks like that it's the same:
[image]

zibous, 2023-12-06:
> This project is compatibly with ESP32 SX1276?
> Device link: [Aliexpress](https://pt.aliexpress.com/item/1005005967763162.html?spm=a2g0o.order_list.order_list_main.5.248ccaa4dAZqB9&gatewayAdapt=glo2bra)

Hi,
Negativ, you can't use ESP32 SX1276. Only ESP32 +  CC1101 is supported.

closed 2023-12-06 by zibous