zibous/ha-watermeter #31: Blank web page with "200 OK" return

mickaelBert, 2023-12-12:
HI, 

I'm not very familiar with ESPhome deployement. 
I have trie a lot of yaml (wm-esp32-simple, wm-esp32-test, ...) from your gihtub and i cannot get the webserver page to see the value or log information. I must be forgetting something :s

here is the url form the esp after deployment : 
[image]

here is the yaml used (but i have the same result after a lot of yaml test configuration)
[test.yaml.txt](https://github.com/zibous/ha-watermeter/files/13646246/test.yaml.txt)
i tried some configuration on webserver : local = true or false, etc. nothing help me. In all the case, i get the webserver with a blanck page and OK 200 text. 

here is the last deployment on esp32
[logs_test_run.txt](https://github.com/zibous/ha-watermeter/files/13646302/logs_test_run.txt)

Thanks for your help.

zibous, 2023-12-12:
hi,

You use webserver `version 2` but the js_url is `version 1` :
```
## ---------------------------------------------------
## COMPONENT WEBSERVER
## ---------------------------------------------------
web_server:
  port: 80
  version: 2
  js_url: https://esphome.io/_static/webserver-v1.min.js
```

see: 
  https://esphome.io/components/web_server.html?highlight=webserver
  https://github.com/zibous/ha-watermeter/tree/master/esphome/webserver

Try this for a simple test:
You use webserver `version 2` but the js_url is `version 1` :
```
## ---------------------------------------------------
## COMPONENT WEBSERVER
## ---------------------------------------------------
web_server:
  port: 80
  version: 1
```

mickaelBert, 2023-12-12:
Thanks for your fast answer. I think i have already tried, but here is a new test (with same result) : 
## ---------------------------------------------------
## COMPONENT WEBSERVER
## ---------------------------------------------------
web_server:
  port: 80
  version: 1

[image]
[testv1.yaml.txt](https://github.com/zibous/ha-watermeter/files/13647637/testv1.yaml.txt)
[logs_test1_run_v1.txt](https://github.com/zibous/ha-watermeter/files/13647638/logs_test1_run_v1.txt)

mickaelBert, 2023-12-12:
I made a new deployement (same configuration) to another wifi SSID (same vlan) and it's working now. sorry for the inconvenience and thanks for your fast answer.

mickaelBert, 2023-12-12:
network segmentation error with vlan rules

closed 2023-12-12 by mickaelBert

zibous, 2023-12-12:
> I made a new deployement (same configuration) to another wifi SSID (same vlan) and it's working now. sorry for the inconvenience and thanks for your fast answer.

o.k. if it works, you can close the issues.