zibous/ha-watermeter #32: Mqtt configuration example

mickaelBert, 2023-12-17:
Hi Zibous, 
Your program is really great! 
I get the information from sen REST correctly. 
However, I thought that it was possible to send  water-meter information with mqtt. Do you have a yaml configuration example for MQTT ? (i'm not using Home Assistant ;))

Mickaël

zibous, 2023-12-17:
> However, I thought that it was possible to send water-meter information with mqtt.

I think there is a problem when you activate MQTT.

see: https://github.com/SzczepanLeon/esphome-components/issues/60

But you can try it:

1. Disable the Home Assistant API COMPONENT:
```
## ---------------------------------------------------
## Home Assistant API COMPONENT
## ---------------------------------------------------
## api:
  ## id: espapi_wmbus_esp32
  ##  port: 6053
  ## reboot_timeout: 0s
```

2. Add the MQTT COMPONENT:
```
# ---------------------------------------
# COMPONENT MQTT Brocker
# ---------------------------------------
mqtt:
  broker: !secret mqtt_broker1
  port: !secret mqtt_port1
  username: !secret mqtt_username1
  password: !secret mqtt_password1
```

Please note that esphome need`s a lot of memory to publish MQTT Data.

mickaelBert, 2023-12-17:
OK thanks for your example, i will try  this.

mickaelBert, 2023-12-28:
Mqtt need too memory. I will use simple http request with Json

closed 2023-12-28 by mickaelBert