zibous/ha-watermeter #34: Wasseruhr Anzeige "NA"

vhamburger, 2023-12-28:
Hi, thanks for this project - it is amazing.
I managed to get my Diehl IZAR Watermeter ID and can successfully decode the telegrams. 
When I set up the device with the correct id (HEX) I get some data (battery life, alarm, etc..) but when I look at "Wasseruhr Anzeige" it says "NA". 
However, when I take the telegram and decode it, I get my current reading correctly decoded.
I am using the ESP Home and HA integration - I get NA on the internal webpage as well as in home assistant. 
Do I have to do some extra decoding steps to get those values?

zibous, 2023-12-28:
> I managed to get my Diehl IZAR Watermeter ID and can successfully decode the telegrams.

How is the result if you analyse the telegram with:
https://wmbusmeters.org/

vhamburger, 2023-12-28:
I can read everything - here is an example output:

>telegram=|1944A511780757550023A2_511000131FC3EC1F133D8818BBD961|

Auto driver  : izar
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 57550023 dll-id (23005557)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 511000131FC3EC1F133D8818BBD961 mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"23005557",
    "last_month_total_m3":175.379,
    "remaining_battery_life_y":8,
    "total_m3":177.81,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2023-12-19",
    "previous_alarms":"no_alarm",
    "timestamp":"2023-12-28T19:07:49Z"
}

Using: wmbusmeters: 1.14.0-48-gc4b9f0d
c4b9f0d10477cb6bb2472e1f2de5eb08ba42e7c4

zibous, 2023-12-28:
ok. so are you shure that you use the correct watermeter id.

Maybe Szczepan Leon can help you, because the error is based on the wmbus meter component.
see: https://github.com/SzczepanLeon/esphome-components

vhamburger, 2023-12-28:
Ok thanks - will open an issue on his project ;)

SzczepanLeon, 2023-12-28:
Logs - please include logs (VERY VERBOSE or DEBUG)

patmtp35, 2024-01-21:
hi
i have the same issue : i have an Izar RC 868 I G4  (french one).
when use wmbusmaters analyse tools  with izar error and with auto it runs ok
[image]

told me that the best driver is dme_07 but not working with esphome

zibous, 2024-01-21:
> told me that the best driver is dme_07 but not working with esphome

see: https://github.com/SzczepanLeon/esphome-components/issues/42
> added in 2.2.23 dme07.

Should actually work, how does the log look like if you set it to VERBOSE?

patmtp35, 2024-01-21:
i resume 

with izar driver i have the same probleme as told vhamburger =>
[image]

when i analyze telegram with izar => error  message ( i have take a photo of my watermeter it's an Izar RC 868 I G4 new one)

 if i analyse in auto it decode ok bu told me drivers dme07. 
 
if i use in esphome dme07 => verbose mode it decode nothing. 

perhaps as my izar was a G4 there is a driver problem

zibous, 2024-01-21:
Do you set the meter_id ?
w/o meter_id(meter_id:0) you can see only telegrams, no values are deccoded.

```yaml
- platform: wmbus
    meter_id: 0x14101780
    type: dme07
    lqi:
      name: "Lqi"
    rssi:
      name: "RSSI"
    total_water_m3:
      name: "water"
```

> perhaps as my izar was a G4 there is a driver problem

If no data is decoded, then it is the WMBUS component (driver) and then only  Szczepan Leon can help you.
It may well be that he would have to adjust these.

patmtp35, 2024-01-21:
yes of corse i set the id .
i use it with [wm-esp32-simple.yaml](https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-esp32-simple.yaml)

juste tO have the minimum : total M3 value and to be sure all is working basicly

SzczepanLeon, 2024-01-21:
@patmtp35 
As You can see data are encoded - AES_CBC_IV. Output from wmbusmeters can be missleading. dme07 as a driver should be OK, but you need to add key.

patmtp35, 2024-01-22:
ok thanks if i can't , i ll test like this solution :

https://github.com/SzczepanLeon/esphome-components/issues/64

patmtp35, 2024-01-22:
hi,

i m not a great developer  so i hope to not make some mistake , i apologize if i do some.

SzczepanLeon in his component WMBUS.CPP refer to driver   this->add_driver(new Dme07()); 
or in wmbusmeters depot the driver name is [driver_dme_07.cc] 

and dme_07 is the good ref for Izar 868 I G4. as the web analyse result confirm

zibous, 2024-01-22:
@SzczepanLeon 

> As You can see data are encoded - AES_CBC_IV.

Negativ can'nt see `AES_CBC_IV` and wmbus analyse is working w/o key

https://wmbusmeters.org/analyze/1E44A511801710147B077A610010052F2F04137ED5010002FD1700002F2F2F

[image]

SzczepanLeon, 2024-01-22:
In previous image You have AES

[image]

zibous, 2024-01-22:
@SzczepanLeon 

sorry, you're right. I overlooked line 013 in all the data.

patmtp35, 2024-01-22:
thanks for all :)

so there is what happen in verbose mode 
[image]

i have to pass the AES in the Script ?

SzczepanLeon, 2024-01-22:
Key - probably You need key in YAML.

patmtp35, 2024-01-24:
Thanks for your help ! insight always pays off!
so I started with this idea of ​​KEY and I found a post which talks about it throughout which I found a solution to find my key.

https://github.com/wmbusmeters/wmbusmeters/issues/281

the IZAR 868 I G4 therefore needs a 32 digit AES key :)

just have to add it under driver section => 

meter_id: !secret watermeterId
    type: dme07
    key: !secret key
    add_prefix: false

closed 2024-04-02 by zibous

balroX, 2024-05-12:
Hi,
I have the same problem and unfortunately I am not able to get values in HomeAssistant.

I have installed esphome/wm-esp32-simple.yaml using ESP Home.
The only data I have changed in the files are the WLAN data and the Meter_ID.

I get 3 messages from the logs. 45, 30 and 26 digits.
`[22:40:21][D][mbus:034]: Processing T1 A frame
[22:40:21][V][mbus:045]: Frame: 36571C99934D4EC5933A569C5933538F4CAC98E34D65659634B9A56728CD972D0BC6CD1A3A59A9C59D163B1D1A (45) [RAW]
[22:40:21][V][mbus:052]: Frame: 1944A511780729640717BCE8A211500013A95EB19EC3D8C629AFD5C02DC6 (30) [with CRC]
[22:40:21][V][mbus:095]: Validating CRC for Block1
[22:40:21][V][crc:031]:     calculated: 0xBCE8, read: 0xBCE8
[22:40:21][V][mbus:115]: Validating CRC for Block2
[22:40:21][V][crc:031]:     calculated: 0x2DC6, read: 0x2DC6
[22:40:21][V][mbus:062]: Frame: 1944A511780729640717A211500013A95EB19EC3D8C629AFD5C0 (26) [without CRC]
[22:40:21][V][sensor:043]: 'Watermeter CC1101 LQI': Received new state 128.000000
[22:40:21][D][sensor:094]: 'Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[22:40:21][V][sensor:043]: 'Watermeter CC1101 RSSI': Received new state -60.000000
[22:40:21][D][sensor:094]: 'Watermeter CC1101 RSSI': Sending state -60.00000 dbm with 0 decimals of accuracy
[22:40:21][V][sensor:043]: 'Watermeter current alarm': Received new state 0.000000
[22:40:21][D][sensor:094]: 'Watermeter current alarm': Sending state 0.00000  with 0 decimals of accuracy
[22:40:21][V][sensor:043]: 'Watermeter previous alarm': Received new state 4.000000
[22:40:21][D][sensor:094]: 'Watermeter previous alarm': Sending state 4.00000  with 0 decimals of accuracy
[22:40:21][V][sensor:043]: 'Watermeter Battery Life': Received new state 8.000000
[22:40:21][D][sensor:094]: 'Watermeter Battery Life': Sending state 8.00000  with 0 decimals of accuracy
[22:40:21][V][sensor:043]: 'Watermeter transmit periode': Received new state 8.000000
[22:40:21][D][sensor:094]: 'Watermeter transmit periode': Sending state 8.00000 s with 0 decimals of accuracy
[22:40:21][W][component:237]: Component wmbus took a long time for an operation (188 ms).
[22:40:21][W][component:238]: Components should block for at most 30 ms. `

For the last two, I can read the correct meter reading using https://wmbusmeters.org/analyze/1944a511780729640717a231500013a970ef0d959B619a19e7c5.
`telegram=|1944A511780729640717A2_31500013A970EF0D959B619A19E7C5|

Auto driver  : izar
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 29640717 dll-id (17076429)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 31500013A970EF0D959B619A19E7C5 mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"17076429",
    "last_month_total_m3":814.425,
    "remaining_battery_life_y":8,
    "total_m3":818.531,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-05-01",
    "previous_alarms":"leakage",
    "timestamp":"2024-05-12T20:51:19Z"
}

Using: wmbusmeters: 1.16.1-23-gefcc41d
efcc41d107eaea1f78388386d4004b25112b0c04

Do you know how to improve the decoding? [Open pre-filled issue ticket](https://github.com/weetmuts/wmbusmeters/issues/new?title=&body=Explain%20the%20problem.....%20%0A%0Ahttps://wmbusmeters.org/analyze/1944a511780729640717a231500013a970ef0d959B619a19e7c5:izar%20%0A%0A)`

Do I need to adjust anything else in the files?

zibous, 2024-05-13:
@balroX 

It looks like the ESP32+CC1101 is working without any problems. 
However, you only get some data decoded from the protocol, so I suspect that it is the wmbus component. 

- current alarm
- previous alarm
- Battery Life
- transmit periode

You can try to get more information by setting the log level to `VERY_VERBOSE`.

```
logger:
  id: appslogger
  level: ${log_level}
  baud_rate: ${log_baudrate}
  logs:
    wmbus: VERY_VERBOSE
    wMBus-lib: VERY_VERBOSE
```

```log
[22:40:21][D][mbus:034]:      Processing T1 A frame 
[22:40:21][V][mbus:045]:      Frame: 36571C99934D4EC5933A569C5933538F4CAC98E34D65659634B9A56728CD972D0BC6CD1A3A59A9C59D163B1D1A (45) [RAW] 
[22:40:21][V][mbus:052]:      Frame: 1944A511780729640717BCE8A211500013A95EB19EC3D8C629AFD5C02DC6 (30) [with CRC] 
[22:40:21][V][mbus:095]:      Validating CRC for Block1
[22:40:21][V][crc:031]:       calculated: 0xBCE8, read: 0xBCE8 
[22:40:21][V][mbus:115]:      Validating CRC for Block2 
[22:40:21][V][crc:031]:       calculated: 0x2DC6, read: 0x2DC6 
[22:40:21][V][mbus:062]:      Frame: 1944A511780729640717A211500013A95EB19EC3D8C629AFD5C0 (26) [without CRC] 
[22:40:21][V][sensor:043]:   'Watermeter CC1101 LQI': Received new state 128.000000 
[22:40:21][D][sensor:094]:   'Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy 
[22:40:21][V][sensor:043]:   'Watermeter CC1101 RSSI': Received new state -60.000000 
[22:40:21][D][sensor:094]:   'Watermeter CC1101 RSSI': Sending state -60.00000 dbm with 0 decimals of accuracy 
[22:40:21][V][sensor:043]:   'Watermeter current alarm': Received new state 0.000000 
[22:40:21][D][sensor:094]:   'Watermeter current alarm': Sending state 0.00000  with 0 decimals of accuracy 
[22:40:21][V][sensor:043]:   'Watermeter previous alarm': Received new state 4.000000 
[22:40:21][D][sensor:094]:   'Watermeter previous alarm': Sending state 4.00000  with 0 decimals of accuracy 
[22:40:21][V][sensor:043]:   'Watermeter Battery Life': Received new state 8.000000 
[22:40:21][D][sensor:094]:   'Watermeter Battery Life': Sending state 8.00000  with 0 decimals of accuracy 
[22:40:21][V][sensor:043]:   'Watermeter transmit periode': Received new state 8.000000 
[22:40:21][D][sensor:094]:   'Watermeter transmit periode': Sending state 8.00000 s with 0 decimals of accuracy 
[22:40:21][W][component:237]: Component wmbus took a long time for an operation (188 ms). 
[22:40:21][W][component:238]: Components should block for at most 30 ms. 
```
In this case, only Szczepan Leon can help you.

https://github.com/SzczepanLeon/esphome-components