zibous/ha-watermeter #37: Month and last month ?

patmtp35, 2024-02-02:
hi,

worki,g like a charm since 2 weeks, but some thing strange,  as we pass on ferbruary, watermonth , last month and water display last month dosen't change . any idea ?

[image]

zibous, 2024-02-03:
The following reasons could be: 

- You don't get any data from the water meter anymore. 
- The time component does not work properly because the history data (hour, day, monthly consumption) is created there.

see:

```yaml
## ---------------------------------------------------
## SNTP COMPONENT
## ---------------------------------------------------
time:
  - platform: sntp
    id: time_sntp
    timezone: Europe/Berlin
    servers:
      - !secret local_sntp
      - 0.at.pool.ntp.org
      - 0.pool.ntp.org
   ....... 
```

I would take a look at the log to see whether you receive data and how it is processed. 
```yaml
  ## logger settings
  log_level: "DEBUG"   # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "VERBOSE"  # Loglevel for wmbus meters component
  log_baudrate: "0"   # use 115200 to enable serial UART port logging.

```

The status "Waterdisplay lastmonth" comes from the water meter and if it is 0, 
then it doesn't have it or you don't get any data. 

```yaml
    # get the last month total watermter m3 from the wmbus telegram
    last_month_total_water_m3:
      name: "Wasseruhr Anzeige letzter Monat"
      id: "waterdisplay_lastmonth"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
```

patmtp35, 2024-02-04:
yes i could see : 
[image]

the only thing i modifiy is : timezone: Europe/Paris

zibous, 2024-02-04:
> the only thing i modifiy is : timezone: Europe/Paris

Ok. that is possible. So you get data from your watermeter and the SNTP COMPONENT scripts are working.
Calculation for history data:
```
   id(current_value) = float(id(waterdisplay).state-id(last_value)) * 1000.00;
   id(hour_value)+=id(current_value);
   id(daily_value)+=id(current_value);
   id(week_value)+=id(current_value);
   id(year_value)+=id(current_value);
```

The status `Waterdisplay lastmonth` and `current_month_total_water_l` comes from the water meter telegram. if this value is 0, the telegram do not has this.

```
izar
    total_water_m3
    last_month_total_water_m3
    current_month_total_water_l
    transmit_period_s
    remaining_battery_life_y
    current_alarms
    previous_alarms

```
see: https://github.com/SzczepanLeon/esphome-components
Is your Meter type `ìzar` ?

patmtp35, 2024-02-04:
very good that's what I thought my WATERMETER being different the telegram is not the same as yours, I'm going to do the calculation via home assistant.
Thank you again for your help !

zibous, 2024-02-04:
> very good that's what I thought my WATERMETER being different the telegram is not the same as yours, I'm going to do the calculation via home assistant.

Yes this is the best way 👍

closed 2024-02-04 by zibous