zibous/ha-watermeter #39: Water 5.5 Qundis

cool338, 2024-04-28:
> 
> https://wmbusmeters.org/analyze/49449344806281141a063246780dff5f350082c20035cdB226a05d9da9de22ecef0B3ec22a8381cd96234e7ca10a56c23e189dd3de4c7B1292161162681acB9559e456f314699da396d39f29046d320B0634675d
> 
> [12:50:51][V][rxLoop:167]: Have 86 bytes from CC1101 Rx, RSSI: -51 dBm LQI: 128 [12:50:51][D][mbus:013]: Processing C1 A frame [12:50:51][V][mbus:017]: Frame: 49449344806281141A063246780DFF5F350082C20035CDB226A05D9DA9DE22ECEF0B3EC22A8381CD96234E7CA10A56C23E189DD3DE4C7B1292161162681ACB9559E456F314699DA396D39F29046D320B0634675D (84) [with CRC] [12:50:51][V][mbus:095]: Validating CRC for Block1 [12:50:51][V][crc:031]: calculated: 0x3246, read: 0x3246 [12:50:51][V][mbus:115]: Validating CRC for Block2 [12:50:51][V][crc:031]: calculated: 0xA9DE, read: 0xA9DE [12:50:51][V][mbus:115]: Validating CRC for Block3 [12:50:51][V][crc:031]: calculated: 0x56C2, read: 0x56C2 [12:50:51][V][mbus:115]: Validating CRC for Block4 [12:50:51][V][crc:031]: calculated: 0x59E4, read: 0x59E4 [12:50:51][V][mbus:115]: Validating CRC for Block5 [12:50:51][V][crc:031]: calculated: 0x675D, read: 0x675D [12:50:51][V][mbus:062]: Frame: 49449344806281141A06780DFF5F350082C20035CDB226A05D9D22ECEF0B3EC22A8381CD96234E7CA10A3E189DD3DE4C7B1292161162681ACB9556F314699DA396D39F29046D320B0634 (74) [without CRC]
> 
> ![PXL_20240404_165711280](https://private-user-images.githubusercontent.com/122101055/320201311-478d2940-3f53-4d34-b894-ffec267da1ce.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzMjAyMzYsIm5iZiI6MTcxNDMxOTkzNiwicGF0aCI6Ii8xMjIxMDEwNTUvMzIwMjAxMzExLTQ3OGQyOTQwLTNmNTMtNGQzNC1iODk0LWZmZWMyNjdkYTFjZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyOFQxNTU4NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hOTYxZjkxYWVmZTBjNWM3NDkxMjc2OGVhZjE2MGQ3ZDk5YmZjMTNiMDdkYzc0ZTc2OGQzYTg3NDUwNTRjZWYxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.LRKaeYygCmvsLB6V44salnLdBDXkFyoGk2YFtXAhTBw)
> 
> https://wmbusmeters.org/analyze/49449344713180141A0763FB780DFF5F350082C30035AE4F225B383E1A19394980BB13B4CBE18D33E03772CD137E1AD00C95A74D9445050DD7D0F2263E25E46FF7B89F584E890CBEA985ED94046D320B0634798D
> 
> [12:50:47][V][rxLoop:167]: Have 86 bytes from CC1101 Rx, RSSI: -58 dBm LQI: 128 [12:50:47][D][mbus:013]: Processing C1 A frame [12:50:47][V][mbus:017]: Frame: 49449344713180141A0763FB780DFF5F350082C30035AE4F225B383E1A19394980BB13B4CBE18D33E03772CD137E1AD00C95A74D9445050DD7D0F2263E25E46FF7B89F584E890CBEA985ED94046D320B0634798D (84) [with CRC] [12:50:47][V][mbus:095]: Validating CRC for Block1 [12:50:47][V][crc:031]: calculated: 0x63FB, read: 0x63FB [12:50:47][V][mbus:115]: Validating CRC for Block2 [12:50:47][V][crc:031]: calculated: 0x1A19, read: 0x1A19 [12:50:47][V][mbus:115]: Validating CRC for Block3 [12:50:47][V][crc:031]: calculated: 0x1AD0, read: 0x1AD0 [12:50:47][V][mbus:115]: Validating CRC for Block4 [12:50:47][V][crc:031]: calculated: 0xF7B8, read: 0xF7B8 [12:50:47][V][mbus:115]: Validating CRC for Block5 [12:50:47][V][crc:031]: calculated: 0x798D, read: 0x798D [12:50:47][V][mbus:062]: Frame: 49449344713180141A07780DFF5F350082C30035AE4F225B383E394980BB13B4CBE18D33E03772CD137E0C95A74D9445050DD7D0F2263E25E46F9F584E890CBEA985ED94046D320B0634 (74) [without CRC]
> 
> ![PXL_20240402_090710000](https://private-user-images.githubusercontent.com/122101055/320201312-2247f699-438c-4c72-b1fe-d79d0ee628e1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzMjAyMzYsIm5iZiI6MTcxNDMxOTkzNiwicGF0aCI6Ii8xMjIxMDEwNTUvMzIwMjAxMzEyLTIyNDdmNjk5LTQzOGMtNGM3Mi1iMWZlLWQ3OWQwZWU2MjhlMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyOFQxNTU4NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05NzJiZDgwYzM2OGVmODE1ZGY3ZGVmOTJhMmU5ODYxZDNmODhiZjVhMGYwNjM3ZGIxZWVkYzA1ODM2Njg4MDE4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.D3Hm9gj8HN0OQkqoo2jdIoyUsiqE4wFMxY2zxLepY_0)

> Kann mir jemand erklären was ich falsch mache?
> 
> Und ist es möglich Rauchmelder mit zu intrigieren?
> 
> Can someone explain to me what I'm doing wrong?
> 
> And is it possible to integrate smoke detectors? https://wmbusmeters.org/analyze/58442515667383020C1AEFE47AA10000000BFD0F000102046D370A06710B3402FD17000082206CE828426C1033846B7B40FF2C000F11008250FD61000082506C232601018260FD6100008360FD31000000823706606C01018270FD61000082706C01013F9C

zibous, 2024-05-01:
It looks like, the ESP+CC1101 is working correctly, but the telegram is not being processed
by the WMBUS component. In this case, only Szczepan Leon can help you.

The same applies to the smoke detector, there is no implementation for this device.

see: https://github.com/SzczepanLeon/esphome-components

closed 2024-05-01 by zibous

zibous, 2024-05-13:
@cool338 

> Und ist es möglich Rauchmelder mit zu intrigieren?

Another option see: https://github.com/merbanan/rtl_433

sippeeey, 2024-06-07:
@cool338 ich habe genau das gleiche Problem, dass die Werte, die ausgelesen werden, nicht der Realität entsprechen.

Beispiel: https://wmbusmeters.org/analyze/49449344708828521a072198780dff5f350082a70035865B2883cf79596625201701faccdcedd4506930ee48B78676cf439BeB7e6054de780a19938a15235695B5c3e7af7140e85Bfaefe1dd046d011307365690

Tatsächlicher Verbrauch: 1885m³

Ich setze ebenfalls das ESP32 Projekt und habe diese Antenne: https://www.amazon.de/dp/B075PFQ57G?psc=1&ref=ppx_yo2ov_dt_b_product_details (Laqiya CC1101 868MHZ Funkübertragung Antenne Transceiver Modul)

Hast du es bereits lösen können?

cool338, 2024-06-08:
> @cool338 ich habe genau das gleiche Problem, dass die Werte, die ausgelesen werden, nicht der Realität entsprechen.
>
> Beispiel: [...]

Moin,

So wie es aussieht liegt es daran das die Daten verschlüsselt sind. Ohne das Passwort geht es nicht.

Ich hab bereits den Betreiber geschrieben der sagt natürlich nein (Kalo) ..

sippeeey, 2024-06-08:
> Moin,
>
> So wie es aussieht liegt es daran das die Daten verschlüsselt sind. Ohne das Passwort geht es nicht.
>
> Ich hab bereits den Betreiber geschrieben der sagt natürlich nein (Kalo) ..

Wie erkennt man, dass der Frame verschlüsselt ist?