zibous/ha-watermeter #41: Pin Belegung K0R41

AndiRued, 2024-05-28:
Hallo,
ich habe folgendes CC1101 bei A... gekauft: https://www.amazon.de/dp/B0C8RRG88V?ref=cm_sw_r_apin_dp_X5D4GFVWN8TVQAS84MBX&ref_=cm_sw_r_apin_dp_X5D4GFVWN8TVQAS84MBX&social_share=cm_sw_r_apin_dp_X5D4GFVWN8TVQAS84MBX&starsLeft=1&skipTwisterOG=1

Wie ist die Pinbelegung vom o.g. Modul zum ESP32?
In der Beschreibung ist zwar ein CC1101 beschrieben, jedoch sind die Namen der PINs ein wenig anders.

Beste Grüße und vielen Dank.

zibous, 2024-05-29:
I would try this

```
## ------------------------------------------------------------------
 ##           WMBUS K0R41 --> ESP32 az-delivery-devkit-v4
 ## ------------------------------------------------------------------
 ##
 ##
 ##                                                               o 2 (3.3V)
 ##                                                               |
 ##   ╭――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――o―╮
 ##   |                                                             |
 ## - | 5v               az-delivery-devkit-v4                      | -- ANT
 ##   |                                                             |
 ##   |                          16 17 5  18 19               23    |
 ##   ╰――x――x――x――x――x――x――x――x――o――x――o――o――o――o――o――o――o――o――o――o―╯
 ##                              |  |  |  |  |                 |   |
 ##                              o  |  |  o  |                 |   ╰-o - 1 (GND)
 ##                              3  o  |  5  o                 o
 ##                            GDO0 8  | CLK 7                 6
 ##                               GD02 o    MISO              M0SI
 ##                                    4
 ##                                   CSN
 ##
 ## ------------------------------------------------------------------
```

[image]

AndiRued, 2024-05-29:
Hi, thanks a lot. I try this tomorrow :-)

zibous, 2024-07-16:
This issue is stale because it has been open for 30 days with no activity.

closed 2024-07-16 by zibous