zibous/ha-watermeter #42: mehrere Devices

alkue1, 2024-06-05:
Hi,

gibt es eine einfache Möglichkeit, gleich mehrere Zähler in einer Installation einzufügen?

Vielen Dank!

zibous, 2024-06-05:
> gibt es eine einfache Möglichkeit, gleich mehrere Zähler in einer Installation einzufügen?
> 
> Vielen Dank!

Einfach pro Zähler `- platform: wmbus`einfügen und den `meter_id` und die benötigen Sensoren anfügen...

```
sensor:

  - platform: wmbus
     meter_id: !secret watermeterId1
     type: izar
     add_prefix: true
     rssi:
       name: "My RSSI"
     ... add more

  - platform: wmbus
     meter_id: !secret watermeterId2
     type: izar
     add_prefix: true
     rssi:
       name: "My RSSI"
     ... add more

- platform: wmbus
     meter_id: !secret watermeterId3
     type: izar
     add_prefix: true
     rssi:
       name: "My RSSI"
     ... add more
```

zibous, 2024-07-16:
This issue is stale because it has been open for 30 days with no activity.

closed 2024-07-16 by zibous