zibous/ha-watermeter #44: How to implement driver for QUALCOSONIC W1 meter

ZancaAngelo, 2024-06-15:
Hi at all,
I see at the follow link "https://github.com/wmbusmeters/wmbusmeters/tree/master/src"  it's be present a drive for the Qualcosonic "src/**driver_qualcosonic.cc**" ==>[(driver_qualcosonic.txt)](https://github.com/user-attachments/files/15847984/driver_qualcosonic.txt)  but it is written differently respect the drivers of other models, present in your folder "esphome/custom_components/wmbus"

Can you help me ?

zibous, 2024-06-15:
> present in your folder "esphome/custom_components/wmbus ..

Yes, that's right. Currently, only the water meters that Szczepan Leon has provided are supported. 
You can find more information here: https://github.com/SzczepanLeon/esphome-components

zibous, 2024-07-16:
This issue is stale because it has been open for 30 days with no activity.

closed 2024-07-16 by zibous