zibous/ha-watermeter #45: falt since last esphome update

patmtp35, 2024-06-21:
hi,

yesterday there is a esphome update, and now when i want to update my watermeter i have this issue : 

Failed config

http_request: [source /config/esphome/cpteauesp.yaml:523]
  
  ESPHome supports certificate verification only via ESP-IDF. Set 'verify_ssl: false' to skip certificate validation and allow less secure HTTPS connections.
  id: http_request_data
  useragent: izar.watermeter
  timeout: 10s

zibous, 2024-06-23:
Hi,

I don't have good experiences with the updates of ESPHome, they change too much and make everything unstable. 
Therefore I use the Docker variant and stay on the version where it works.

 The new ESHOME as changed al lot:
 - COMPONENT OTA
 - COMPONENT HTTP Request
  >ESPHome supports certificate verification only via ESP-IDF. Set 'verify_ssl: false' to skip certificate validation and allow less secure HTTPS connections.
see: https://esphome.io/components/http_request.html

[image]

 - COMPONENT BACKUP do not work anymore..

You can do
 - delete COMPONENT HTTP (optional) + SCRIPT and remove id(post_device_state).execute();
 - Fix OTA COMPONENT

patmtp35, 2024-06-24:
thanks for you 're fast answer i'll do that !

regards

zibous, 2024-07-01:
@patmtp35 

see: https://github.com/zibous/ha-watermeter/issues/46#issuecomment-2200675081

closed 2024-07-16 by zibous