zibous/ha-watermeter #46: fault since last esphome update

evilmumi, 2024-07-01:
some I could fix but now:

`INFO ESPHome 2024.6.4
INFO Reading configuration /config/esphome/testtestesp32.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Updating https://github.com/zdzichu6969/esphome-components.git@None
INFO Updating https://github.com/TheStaticTurtle/esphome_syslog.git@None
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Backup config will take: 5400 bytes
INFO Compiling app...
Processing water-meter-esp (board: az-delivery-devkit-v4; framework: arduino; platform: espressif32)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Syslog @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- SPI @ 2.0.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7+sha.b8c6af4
|-- wmbus-drivers @ 0.0.0+20240226111441.sha.1ad30e4
|-- noise-c @ 0.1.4
Compiling .pioenvs/water-meter-esp/src/main.cpp.o
Archiving .pioenvs/water-meter-esp/lib64d/libWiFi.a
Compiling .pioenvs/water-meter-esp/lib01c/FS/FS.cpp.o
Compiling .pioenvs/water-meter-esp/lib01c/FS/vfs_api.cpp.o
Compiling .pioenvs/water-meter-esp/libbc6/Update/HttpsOTAUpdate.cpp.o
Compiling .pioenvs/water-meter-esp/libbc6/Update/Updater.cpp.o
Compiling .pioenvs/water-meter-esp/libc0f/ESPAsyncWebServer-esphome/AsyncEventSource.cpp.o
Archiving .pioenvs/water-meter-esp/lib01c/libFS.a
Compiling .pioenvs/water-meter-esp/libc0f/ESPAsyncWebServer-esphome/AsyncWebSocket.cpp.o
Archiving .pioenvs/water-meter-esp/libbc6/libUpdate.a
Compiling .pioenvs/water-meter-esp/libc0f/ESPAsyncWebServer-esphome/WebAuthentication.cpp.o
/config/esphome/testtestesp32.yaml: In lambda function:
/config/esphome/testtestesp32.yaml:573:11: error: 'status_code' was not declared in this scope
                         if (status_code == 200) {
           ^          
/config/esphome/testtestesp32.yaml:573:11: note: suggested alternative: 'statusled'
                         if (status_code == 200) {
           ^          
           statusled
Compiling .pioenvs/water-meter-esp/libc0f/ESPAsyncWebServer-esphome/WebHandlers.cpp.o
Compiling .pioenvs/water-meter-esp/libc0f/ESPAsyncWebServer-esphome/WebRequest.cpp.o
*** [.pioenvs/water-meter-esp/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.48 seconds ==========================`

I think it is about:
#45 `## ---------------------------------------------------
## COMPONENT SCRIPTS
## ---------------------------------------------------
script:

  # id(set_status_message).execute();
  - id: set_status_message
    then:
        - lambda: |-
            int msgcode = int(id(cc1101_state));
            std::string message = id(cc1101_state_message)[msgcode];
            ESP_LOGD("wmbus", "cc1101 state message: %s, error code: %d", message.c_str(), msgcode);
            id(watermeter_status_message).publish_state(message);

  # id(post_device_state).execute();
  - id: post_device_state
    then:
      - if:
          condition:
            lambda: return "${service_enabled} =='true'";
          then:
            - logger.log:
                tag: "system"
                level: INFO
                format: "Post new data to the ${service_url}${device_name_short}, service enabled: ${service_enabled}"
            - http_request.post:
                url: "${service_url}/${device_name_short}"
                
                headers:
                  Content-Type: application/json
                json: |-
                  root["device"] = "${device_name_short}";
                  root["waterdisplay"] = id(waterdisplay).state;
                  root["current"] = id(id(watercurrent)).state;
                  root["hour"] = id(waterhour).state;
                  root["day"] = id(waterday).state;
                  root["yesterday"] = id(wateryesterday).state;
                  root["week"] = id(waterweek).state;
                  root["month"] = id(watermonth).state;
                  root["lastmonth"] = id(waterhour).state;
                  root["year"] = id(wateryear).state;
                  root["bootcounter"] = id(bootcounter).state;
                  root["version"] = "${device_name_short} ${appversion}";
                  root["time"] = id(time_sntp).now().strftime("%Y-%m-%d %H:%M:%S");
                on_response:
                  then:
                    - lambda: |-
                        if (status_code == 200) {
                           ESP_LOGI ("system", "HTTP POST o.k");
                        } else {
                            ESP_LOGI ("system", "HTTP POST failed");
                        }
`

zibous, 2024-07-01:
hi,

Try this:
```
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## this version shows all watermeters. Use this for the first test
##
##  1. uses the latest wmbusmeter component from github
##
## ---------------------------------------------------------------------------
substitutions:

  ## device settings
  device_name_short: "water-meter-esp"
  device_description: "Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) - all watermeters"
  projectname: "Diehl IZAR RC 868.Watermeter"
  appversion: "2.0.7"

  ## all watermeters   wmid: "0"
  ## your watermeter: wmid: !secret watermeterId
  wmid: "0"

  ## logger settings
  log_level: "VERBOSE"  # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "VERBOSE" # Loglevel for wmbus meters component
  log_baudrate: "0" # 0 disable uart logger messages

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}
  on_boot:
    priority: 200
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - logger.log:
          level: INFO
          tag: "system"
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
      - component.update: bootcounter
  on_shutdown:
    priority: 700
    then:
      - logger.log:
          level: ERROR
          tag: "system"
          format: "BOOTMESSAGE:${device_name_short} is down!"

## ----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4
##          ESP32 240MHz, 520KB RAM, 4MB Flash
## ----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino
[... 272 lines cut ...]
    update_interval: 60s
    internal: true
    disabled_by_default: true

  # Wifi quality RSSI in percentage
  # Received Signal Strength (RSSI) is a measure of incoherent
  ## (raw) RF power in a channel.
  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "Device WLAN Signal"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"

  # device boot counter
  - platform: template
    name: Device Boot counter
    id: bootcounter
    icon: mdi:counter
    accuracy_decimals: 0
    state_class: "measurement"
    entity_category: "diagnostic"
    lambda: return (id(boot_counter));
```

evilmumi, 2024-07-01:
found a solution
one line has to be changed
`if (response->status_code == 200) {`

closed 2024-07-01 by evilmumi

patmtp35, 2024-07-02:
hi thanks working without the http section , as i use it with HA it's ok