zibous/ha-watermeter #47: The data of the wmbus frame are not decoded

cdupaty, 2024-07-28:
Hello,
I made the same device with the same components, the ESP32 is programmed with "wm-esp32.yaml" and receives the WMBUS frames, I can decode them by hand with https://wmbusmeters.org/. but they are not decoded and visible on the WEB interface of the ESP32.
Can you help me?
Christian Dupaty
[image]

zibous, 2024-07-28:
Hi,

Did you enter the meter_id: !secret watermeterId correctly?
```
  # 1. Search Meter ID (wmid="0")
  #   watermeterId="0"  # all watermeters
  #   log_level: "VERBOSE"
  #   log_wmbus: "VERBOSE"
  #   save and compile and install to device

  #   see log to find the  Meter ID (wmid)
  #   [17:53:37][D][wmbus:182]: Meter ID [0x90620778] RSSI: -87 dBm LQI: 129 Frame: T1 A not found in configuration T:...

  # 2. Edit watermeterId (example)
  #  +  watermeterId="0x90620778"
  #  + Compile and install
  #    For produktion set:
  #    log_level: "WARN"
  #    log_wmbus: "WARN"

  ## your watermeter: wmid: !secret watermeterId
  ## watermeterId: "0" # 1. scanning all watermeters
  ## watermeterId: "0x90620778"  # 2. Test other watermeter
```

cdupaty, 2024-07-28:
Hi thanks for reply,
Yes watermeterId is correct and logs are VERBOSE
here is a picture of the decoding by https://wmbusmeters.org/
I just copied and pasted a WMBUS frame received by the ESP32
[image]

cdupaty, 2024-07-28:
0x01EE7B4B is the watermeterID in hex anwe see  32406347 in dec (same number)

zibous, 2024-07-28:
Strange,

[image]

But you can test the simple version:

```
---
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for dme07
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## this version shows all watermeters. Use this for the first test
##
##  1. uses the latest wmbusmeter component from github
##
## ---------------------------------------------------------------------------
substitutions:

  ## device settings
  hostname: "watermeter-test"
  friendly_name: "Watermeter"
  device_description: "Testcase Wasserzähler ESP32, CUL - CC1101, dme07 (SzczepanLeon) - all watermeters"

  # The format of the name should be author_name.project_name.
  projectname: "dme07.Watermeter"
  appversion: "2.0.7"
  hardware: "ESP32_Devkit_V4"
  location: "Labor"

  # wmid="0"
  wmid="0x01EE7B4B"
  
  ## logger settings
  log_level: "VERBOSE"  # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "VERBOSE"  # Loglevel for wmbus meters component
  log_baudrate: "0" # 0 disable uart logger messages

## ------------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4
## https://esphome.io/components/esp32
## ------------------------------------------------------------------
esp32:
  board: esp32dev
  framework:
    type: arduino

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${hostname}
  build_path: ./build/${hostname}
  comment: ${device_description}
  on_boot:
    priority: 600
    then:
      - logger.log:
          level: info
          format: "BOOTMESSAGE: device ${hostname} API is connected, Device ready!"
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - component.update: bootcounter

  on_shutdown:
    then:
      - logger.log:
          level: ERROR
          tag: "system"
[... 352 lines cut ...]

## ---------------------------------------------------
## TEXT SENSOREN
## https://esphome.io/components/text_sensor/
## ---------------------------------------------------
text_sensor:

  # waterdisplay timestamp last update (updated by id: "waterdisplay")
  - platform: template
    name: Timestamp
    id: watermeter_lastupdate
    icon: mdi:clock-start
    entity_category: "diagnostic"

  # installed esphome version
  - platform: version
    name: ESPHome Version
    id: eshome_version
    hide_timestamp: true
    disabled_by_default: true
    icon: mdi:information-box-outline

## E.O.F

```

cdupaty, 2024-07-28:
**First try, the result is** 
23:32:00
[V]
[rxLoop:167]
Have 56 bytes from CC1101 Rx, RSSI: -90 dBm LQI: 135
23:32:00
[D]
[mbus:034]
Processing T1 A frame
23:32:00
[V]
[mbus:045]
Frame: 37271C99934D71368B7162CE4E359359C6964E6B16596356599D0D68E64D2E3A664DC653A63A6C2CE70E4E98E3C99D0ED26D1C95AB2CD135 (56) [RAW]
23:32:00
[V]
[mbus:052]
Frame: 1E44A511476340327B0704607A80001005C162513BFA7457FBF832427FBBE5C2CAC49688C7 (37) [with CRC]
23:32:00
[V]
[mbus:095]
Validating CRC for Block1
23:32:00
[V]
[crc:031]
calculated: 0x0460, read: 0x0460
23:32:00
[V]
[mbus:115]
Validating CRC for Block2
23:32:00
[V]
[crc:031]
calculated: 0x7FBB, read: 0x7FBB
23:32:00
[V]
[mbus:115]
Validating CRC for Block3
23:32:00
[V]
[crc:031]
calculated: 0x88C7, read: 0x88C7
23:32:01
[V]
[mbus:062]
Frame: 1E44A511476340327B077A80001005C162513BFA7457FBF83242E5C2CAC496 (31) [without CRC]
23:32:01
[D]
[wmbus:182]
Meter ID [0x32406347] RSSI: -90 dBm LQI: 135 Frame: T1 A not found in configuration T: 1E44A511476340327B077A80001005C162513BFA7457FBF83242E5C2CAC496 (31)

**Then i change line 339 
type: dme07 by izar (my watermeter is izar type)
the result now is**

23:36:58
[V]
[rxLoop:167]
Have 56 bytes from CC1101 Rx, RSSI: -87 dBm LQI: 133
23:36:58
[D]
[mbus:034]
Processing T1 A frame
23:36:58
[V]
[mbus:045]
Frame: 37271C99934D71368B7162CE4E359359C6964E6B0D5963565995934CD669A7438EC4D36CC4D8D3D0B59C4D62D3D0B35CA6572570E9A5A635 (56) [RAW]
23:36:58
[V]
[mbus:052]
Frame: 1E44A511476340327B0704607A8100100507715FFC22D118D1B7C3047037C314F94942A9FB (37) [with CRC]
23:36:58
[V]
[mbus:095]
Validating CRC for Block1
23:36:58
[V]
[crc:031]
calculated: 0x0460, read: 0x0460
23:36:58
[V]
[mbus:115]
Validating CRC for Block2
23:36:58
[V]
[crc:031]
calculated: 0x7037, read: 0x7037
23:36:58
[V]
[mbus:115]
Validating CRC for Block3
23:36:58
[V]
[crc:031]
calculated: 0xA9FB, read: 0xA9FB
23:36:58
[V]
[mbus:062]
Frame: 1E44A511476340327B077A8100100507715FFC22D118D1B7C304C314F94942 (31) [without CRC]
23:36:58
[D]
[wmbus:182]
Meter ID [0x32406347] RSSI: -87 dBm LQI: 133 Frame: T1 A not found in configuration T: 1E44A511476340327B077A8100100507715FFC22D118D1B7C304C314F94942 (31)

[image]

zibous, 2024-07-29:
Hi,

> Then i change line 339
type: dme07 by izar (my watermeter is izar type)
the result now is

The logs are o.k:

1. ESP is working
2. CC1101 is working
3. But  when i try to decode the Telegram on wmbusmeters with izar driver i get errors,
    only with dme_07 or auto it is working.

 [Telegramm test](http://wmbusmeters.org/analyze/1E44A511476340327B077A8100100507715FFC22D118D1B7C304C314F94942)

```Auto driver    : dme_07
Similar driver : dme_07 06/06
Using driver   : izar 00/00
000   : 1e length (30 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 47634032 dll-id (32406347)
008   : 7b dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 81 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 13 vif (Volume l)
019 C?: 0E6F0200 <-- ERROR
023   : 02 dif (16 Bit Integer/Binary Instantaneous value)
024   : FD vif (Second extension FD of VIF-codes)
025   : 17 vife (Error flags (binary))
026 C?: 0000 <-- ERROR
028   : 2F skip
029   : 2F skip
030   : 2F skip

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"32406347",
    "timestamp":"2024-07-29T05:56:38Z"
}
```

> 23:36:58 [D] [wmbus:182] **Meter ID [0x32406347]** RSSI: -87 dBm LQI: 133 Frame: **T1 A** not found in configuration T: 1E44A511476340327B077A8100100507715FFC22D118D1B7C304C314F94942 (31)

`mode (Optional): Frame to process (T1, C1 or both). Defaults to T1`
I never saw "T1 A* , so you have to ask [Szczepan Leon](https://github.com/SzczepanLeon/esphome-components/commits?author=SzczepanLeon) how to do.

zibous, 2024-07-29:
But in my case it takes 3-4 minutes when starting the ESP until I can receive data. 
After that, it comes every 8 seconds. 

You can also try a "clean build" and the firmware will be rebuilt.

cdupaty, 2024-07-30:
Hello, I did a lot of tests and I arrived at this

**If the number on the meter is in decimal it gives**

wmid: "0x01EE7B4B"  #0x01EE7B4B in hexa is the value of 32406347 in decimal
type: dme07
[wmbus:182]: Meter ID [0x32406347] RSSI: -92 dBm LQI: 135 **Frame: T1 A not found in configuration T:** 1E44A511476340327B077A100010058550E8C9C56806F7FA0BC87AD33971DF (31)

**If the number on the meter is in hexadecimal it gives**

wmid: "843080519"   #843080519 in decimal is the value of 0x32406347 in hexa
type: dme07
[wmbus:097]: Using driver 'dme07' for ID [0x32406347] RSSI: -90 dBm LQI: 129 Frame: T1 A T: 1E44A511476340327B077A0F00100553676E63086DE6CC2AC04AB3337B3F08 (31)
[11:26:14][D]
[wmbus:161]: **Can't get value(s) from telegram for ID [0x32406347]**

**I think the meter number written on the label is in hexadecimal.
If I convert it to decimal the number is 843080519, this activates the driver, however there is still no decoding.
Here is the photo of the meter we see that the protocol is DM7B**

[image]

**With wmbusters.org**

Telegram:   1e44a511476340327B077a0f00100553676e63086de6cc2ac04aB3337B3f08
Driver: dme_07
Decryption key: 
Decryption key

telegram=|1E44A511476340327B077A0F0010052F2F_0413AA7E020002FD1700002F2F2F|

Auto driver    : dme_07
Similar driver : dme_07 06/06
Using driver   : dme_07 00/00
000   : 1e length (30 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 47634032 dll-id (32406347)
008   : 7b dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 0f tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 13 vif (Volume l)
019 C!: AA7E0200 ("total_m3":163.498)
023   : 02 dif (16 Bit Integer/Binary Instantaneous value)
024   : FD vif (Second extension FD of VIF-codes)
025   : 17 vife (Error flags (binary))
026 C!: 0000 ("status":"OK")
028   : 2F skip
029   : 2F skip
030   : 2F skip

{
    "media":"water",
    "meter":"dme_07",
    "name":"",
    "id":"32406347",
    "total_m3":163.498,
    "status":"OK",
    "timestamp":"2024-07-30T09:57:44Z"
}

gatismu, 2024-07-30:
Hello!
little oftopic, but how to get this to collect data from two water meters? Thanks in advance!

zibous, 2024-07-31:
Hi,

> Hello! little oftopic, but how to get this to collect data from two water meters? Thanks in advance!

see: https://github.com/zibous/ha-watermeter/issues/42

zibous, 2024-07-31:
@cdupaty 

> If the number on the meter is in decimal it gives

```
  # 1. Search Meter ID (wmid="0")
  #   wmid="0"  # all watermeters
  #   log_level: "VERBOSE"
  #   log_wmbus: "VERBOSE"
  #   save and compile and install to device

  #   see log to find the  Meter ID (wmid)
  #   [17:53:37][D][wmbus:182]: Meter ID [0x90620778] RSSI: -87 dBm LQI: 129 Frame: T1 A not found in configuration T:...

  # 2. Edit wmid
  #  +  wmid="0x90620778"
  #  + Compile and install
  #    For produktion set:
  #    log_level: "WARN"
  #    log_wmbus: "WARN"
```

Or use the test firmware from:
[esphome/testcases](https://github.com/zibous/ha-watermeter/tree/master/esphome/testcases)

gatismu, 2024-08-03:
> wmid="0x01EE7B4B"

Hello, thanks!

I see telegram and it decoved fine, correct:
[](https://wmbusmeters.org/analyze/2144B4098170320317067A5D0000000C1328020600046D1D3503380F030000000000)
I judge based on total value of m^3, my metr show only total value, could be it is fine to se N/a in web server, but still total display value shudl be there, ant find where I am wrong. After thsi will go with two meters, hot and cold water..

log from esphome
```

22:26:38][V][rxLoop:167]: Have 60 bytes from CC1101 Rx, RSSI: -74 dBm LQI: 129
[22:26:38][D][mbus:034]: Processing T1 A frame
[22:26:38][V][mbus:045]: Frame: 38D71C8DC5A5B0D4D62CE58B35359AC9AA4E4E66715965965965B434B3AC58E59A59659C6B13712D958B58B59A2EC5A958B596596596596596B1A959 (60) [RAW]
[22:26:38][V][mbus:052]: Frame: 2144B409817032031706E6F27A5D0000000C1328020600046D1D35030306380F0300000000008695 (40) [with CRC]
[22:26:38][V][mbus:095]: Validating CRC for Block1
[22:26:38][V][crc:031]:     calculated: 0xE6F2, read: 0xE6F2
[22:26:38][V][mbus:115]: Validating CRC for Block2
[22:26:38][V][crc:031]:     calculated: 0x0306, read: 0x0306
[22:26:38][V][mbus:115]: Validating CRC for Block3
[22:26:38][V][crc:031]:     calculated: 0x8695, read: 0x8695
[22:26:38][V][mbus:062]: Frame: 2144B4098170320317067A5D0000000C1328020600046D1D3503380F030000000000 (34) [without CRC]
[22:26:50][V][rxLoop:167]: Have 80 bytes from CC1101 Rx, RSSI: -83 dBm LQI: 132
```

in secrets I used decimal value

meter that I have is bmeters.

```
## SENSORS
## ---------------------------------------------------
sensor:

  - platform: wmbus

    # Meter ID (usually from sticker). Can be specified as decimal or hex.
    # only hex is working for my watermeter !
    # see: https://github.com/SzczepanLeon/esphome-components/issues/6
    # edit watermeterid in the secrets file
    ## add_prefix settings to add the watermeterid to the lqi, rssi and total_water_m3
    meter_id: !secret watermeterId
    type: bmeters
    add_prefix: false

    # The LQI value reported by the CC1101 is a 7 bit unsigned number with a range from 0 to 127.
    # Note that a lower value indicates a better link.
    # The LQI of a received packet will be bad (higher number) when there is lot of interference.
    lqi:
      id: wmbus_cc1101_lqi
      name: "Watermeter CC1101 LQI"
      entity_category: "diagnostic"
      unit_of_measurement: "lqi"
      state_class: "measurement"
```

[image]

gatismu, 2024-08-04:
Hello!
seems I finally solved issue, I dont know exact how, but I took sample code from here (above) it seems same as on repo, and water meter id I used decimal (same syntax/look as I see it in event log) *I tried this already with " simple esp32 code"  from repo, not sure where is difference, but now it works.
```
sensor:

  - platform: wmbus
    meter_id: 0x03327081
    type: bmeters
    add_prefix: false
```
next task it to add hot water meter to same " code" 

Thanks, good luck!

zibous, 2024-08-05:
> ... it seems same as on repo, and water meter id I used decimal
```
Meter ID (usually from sticker). Can be specified as decimal or hex.
# only HEX is working for my watermeter !
see: https://github.com/SzczepanLeon/esphome-components/issues/6
```
	
> 
next task it to add hot water meter to same " code"

```
sensor:
  - platform: wmbus
     meter_id: !secret watermeterId1
     type: izar
     add_prefix: true
     rssi:
       name: "My RSSI"
     ... add more

  - platform: wmbus
     meter_id: !secret watermeterId2
     type: izar
     add_prefix: true
     rssi:
       name: "My RSSI"
     ... add more

- platform: wmbus
     meter_id: !secret watermeterId3
     type: izar
     add_prefix: true
     rssi:
       name: "My RSSI"
     ... add more
```
 
But remember, ESP doesn't have much memory, so you can't process all the sensor data for 2 devices. 
You can try, but if you're lucky, it will work. It's safer to use 2 ESP32

gatismu, 2024-08-06:
> ```
> Meter ID (usually from sticker). Can be specified as decimal or hex.
> # only HEX is working for my watermeter !
> see: https://github.com/SzczepanLeon/esphome-components/issues/6
> ```
>
>
> next task it [...]

Hi, Zibous!

Thanks for replay!
I will run for test, let s see how stable.
Could be I can disable web server (it is now with local: false, but still..) let s see few days how it runs, it runs on Esp32 dev kit from Ali - https://www.aliexpress.com/item/1005006096709541.html?spm=a2g0o.order_list.order_list_main.266.63541802l0f0NI

[image]

+last update I will see in Homeassistant, but need to think on code how to take info fro both sensors

zibous, 2024-08-06:
> Could be I can disable web server (it is now with local: false, but still..) let s see few days how it runs
1. disable the section
   ```
   ## web_server: 
   ##  port: 80
   ##  version: 3
   ##  local: false
   ```
2.  Or use the local `www.js`, no internet required !
```
web_server:
  port: 80
  version: 3
  local: false
  # use the local www.js, no internet required !
  # Copy https://oi.esphome.io/v3/www.js to a v3 folder 
  # curl https://oi.esphome.io/v3/www.js > /var/www/esphome/v3/www.js
  # to your local webserver.
  js_url: "https://webserver.local/esphome/v3/www.js"
  ```

> +last update I will see in Homeassistant, but need to think on code how to take info fro both sensors
You need to text_sensors:
```
- platform: template
    name: Timestamp 
    id: watermeter_lastupdate_coldwater
    icon: mdi:clock-start
    entity_category: "diagnostic"
    web_server_sorting_weight: 10

  - platform: template
    name: Timestamp
    id: watermeter_lastupdate_warmwater
    icon: mdi:clock-start
    entity_category: "diagnostic"
    web_server_sorting_weight: 10
```

And for water meter 1 (coldwater) : 

```
total_water_m3:
 ....
  on_value:
        then:
          - text_sensor.template.publish:
                 id: watermeter_lastupdate_coldwater
                state: !lambda return id(time_sntp).now().strftime("%Y-%m-%dT%H:%M:%S %Z");

and water meter 2 (warmwater) the update instruction:
 ....
  on_value:
        then:
          - text_sensor.template.publish:
                 id: watermeter_lastupdate_warmwater
                state: !lambda return id(time_sntp).now().strftime("%Y-%m-%dT%H:%M:%S %Z");

```

gatismu, 2024-08-11:
Thanks!
seems 2 sensors work fine with this board-
[https://www.aliexpress.com/item/1005006096709541.html?spm=a2g0o.order_list.order_list_main.266.5e3718028LSARV](https://www.aliexpress.com/item/1005006096709541.html?spm=a2g0o.order_list.order_list_main.266.5e3718028LSARV)

[image]
*I reseted recentlry boot counter to see if it reboots, 66+h it is time sinece I added timestamp for both sensors. Seems stable.
In my case I have jsut " value from display", not match to loose it is reboots.

I will investigate local web server, I tried to place files on my external HA docker volume but did not managed to get work, it opens seems V2 and not V3 that I placed there.., need to read about this. Thanks.. this is then oftopic to this case. 

> > Could be I can disable web server (it is now with local: false, but still..) let s see few days how it runs
> 
> 1. disable the section
>    ```
>    ## web_server: 
>    ##  port: 80
>    ##  version: 3
>    ##  local: false
>    ```
> 2. Or use the local `www.js`, no internet required !
> 
> ```
> web_server:
>   port: 80
>   version: 3
>   local: false
>   # use the local www.js, no internet required !
>   # Copy https://oi.esphome.io/v3/www.js to a v3 folder 
>   # curl https://oi.esphome.io/v3/www.js > /var/www/esphome/v3/www.js
>   # to your local webserver.
>   js_url: "https://webserver.local/esphome/v3/www.js"
> ```
> 
> > +last update I will see in Homeassistant, but need to think on code how to take info fro both sensors
> > You need to text_sensors:
> 
> ```
> - platform: template
>     name: Timestamp  Coldwater
>     id: watermeter_lastupdate_coldwater
>     icon: mdi:clock-start
>     entity_category: "diagnostic"
>     web_server_sorting_weight: 10
> 
>   - platform: template
>     name: Timestamp Warmwater
>     id: watermeter_lastupdate_warmwater
>     icon: mdi:clock-start
>     entity_category: "diagnostic"
>     web_server_sorting_weight: 10
> ```
> 
> And for water meter 1 (coldwater) :
> 
> ```
> total_water_m3:
>  ....
>   on_value:
>         then:
>           - text_sensor.template.publish:
>                  id: watermeter_lastupdate_coldwater
>                 state: !lambda return id(time_sntp).now().strftime("%Y-%m-%dT%H:%M:%S %Z");
> 
> and water meter 2 (warmwater) the update instruction:
>  ....
>   on_value:
>         then:
>           - text_sensor.template.publish:
>                  id: watermeter_lastupdate_warmwater
>                 state: !lambda return id(time_sntp).now().strftime("%Y-%m-%dT%H:%M:%S %Z");
> ```

zibous, 2024-08-12:
Hi,

> Or use the local www.js, no internet required !

Try this:
  - create the bash file on your homeassistant server
  - edit HAHOST, HAWWWDIR
  - save the bash file
  - execute the bash file
  
```bash
#!/usr/bin/env bash

echo "Start get esphome www.js"
NEWLINE="\n\n"

# This is your homeassistant local url
HAHOST=**Homeassistant.host**:8123

# This is the homeassistant webserver url
HAWWWDIR="/home/homeassistant/.homeassistant/www/esp32/"

# Source fro ESP Home server
ESPHOMEJSURL="https://oi.esphome.io/v3/www.js"

echo -e "${NEWLINE}Create the target dir: ${HAWWWDIR}"
mkdir ${HAWWWDIR}  >/dev/null 2>&1

echo -e "${NEWLINE}Save content from ${ESPHOMEJSURL} to dir: ${HAWWWDIR}"
curl ${ESPHOMEJSURL} > ${HAWWWDIR}www.js

echo -e "${NEWLINE}Testing if all is done"
curl ${HAHOST}local/esp32/www.js

echo -e "${NEWLINE}Add this lines to your esp device template:"
str=$(cat <<END

web_server:
  port: 80
  version: 3
  local: false
  js_url: "${HAHOST}local/esp32/www.js"

END
)
echo "$str"
echo -e "${NEWLINE}"
```
  
[get_esphomejs.sh.zip](https://github.com/user-attachments/files/16578500/get_esphomejs.sh.zip)

gatismu, 2024-08-14:
Thanks, zibous!

I tried again by making folder on my external Docker volume (where I have nginx www files, I have twobasic pages looking to outside thru Cloudflare zero trust.. I found this easier then suffer with homeassitant, some day I will experiemnt with your script..)), and saved .js there, seems it works, not sure what I did wrong that night, must be made some mistake, thanks!
[image]

+140+ hours uptime, it works fine for me for two sensors.

closed 2024-08-29 by github-actions[bot]