zibous/ha-watermeter #50: How to read enrypted Sensus iPerl watermeter if I have the key?

B-Clever, 2024-09-30:
I have set up the C1101 with an ESP32 and ESPHOME. 
Everything works fine and I can receive the enrypted Sensus iPerl telegrams from my watermeter. I also got the AES key from my water supplier for my own watermeter. 
My question is: How can decrypt and read this watermeter with your ESPHOME code? I haven´t seen an option to provide the key inside the code? Where do I set up the key? 
Maybe someone can help?

SzczepanLeon, 2024-09-30:
If You can decrypt telegram on [wmbusmeterspawmbusmeters.org) page then You can use my [component](https://github.com/SzczepanLeon/esphome-components) - version 4.0

B-Clever, 2024-09-30:
Thank you @SzczepanLeon ! :-) Your code example worked fine for me. That is exactly what I needed.
Maybe there should be a separate topic in the readme how to deal with encrypted watermeters.

closed 2024-09-30 by B-Clever

SzczepanLeon, 2024-09-30:
@B-Clever Please share YAML here.
There is examples with key field.
[image]