zibous/ha-watermeter #52: Connection to CC1101 fails

stefanschramek, 2024-10-24:
Hi,
I am trying to set up a CC1101 with Adafruit ESP32 Feather V2.
For that I connected the pins in the following way:
- CC1101_3: GPIO19  # M0SI
- CC1101_4: GPIO5  # CLK
- CC1101_5: GPIO21  # MISO
- CC1101_6: GPIO22  # GD02
- CC1101_7: GPIO20  # GDO0
- CC1101_8: GPIO33  # CSN

This is the pinout schema for the board:
https://learn.adafruit.com/adafruit-esp32-feather-v2/pinouts

Unfortunately it is not working:
```
[15:50:49][C][wmbus:326]: wM-Bus v4.0.11-1.17.1-b8f4a945:
[15:50:49][C][wmbus:344]:   MAC: 37EA6BE8
[15:50:49][C][wmbus:346]:   CC1101 frequency: 868.950 MHz
[15:50:49][C][wmbus:347]:   CC1101 SPI bus:
[15:50:49][E][wmbus:349]:    Check connection to CC1101!
[15:50:49][C][wmbus:351]:     MOSI Pin: GPIO19
[15:50:49][C][wmbus:352]:     MISO Pin: GPIO21
[15:50:49][C][wmbus:353]:     CLK Pin:  GPIO5
[15:50:49][C][wmbus:354]:     CS Pin:   GPIO33
[15:50:49][C][wmbus:355]:     GDO0 Pin: GPIO20
[15:50:49][C][wmbus:356]:     GDO2 Pin: GPIO22
[15:50:49][C][wmbus:362]:   Available drivers: unknown, izar
[15:50:49][E][component:082]:   Component wmbus is marked FAILED
```
Do you have an idea what could cause this?

Is there an easy way to test if I maybe damaged the CC1101 when soldering the antenna and cables?

Thanks in advance!

WulleWulleWapWap, 2024-10-28:
Unfortunately I can't help you with your problem, but maybe you can help me. I've been trying to install the project using ESPHOME, but unfortunately I get an error every time I try. How did you get the export to work? Maybe we could get in touch somehow?

Ich kann ihnen mit ihrem Problem leider nicht helfen, aber eventuell können Sie mir helfen. Ich versuche das Projekt mittels ESPHOME zu installieren, leider kommt bei jedem versuch ein Fehler. Wie haben Sie es gemacht das der export funktioniert?
Vielleicht könnten wir irgendwie in Kontakt treten?

zibous, 2024-10-28:
Please find in YAML line external_components, and modyfi it to:

  - source: github://SzczepanLeon/esphome-components@3.2.3

stefanschramek, 2024-10-28:
> Please find in YAML line external_components, and modyfi it to:
> 
> * source: github://SzczepanLeon/esphome-components@3.2.3

Is this related to my issue or to the comment that seems to be removed?

zibous, 2024-10-31:
> Is there an easy way to test if I maybe damaged the CC1101 when soldering the antenna and cables?

Negativ, but wmbus component will check this, and reporting an error if failed.

> Error: Check connection to CC1101!

closed 2024-11-15 by github-actions[bot]