zibous/ha-watermeter #53: No water measurement only battery life and alarms?

WulleWulleWapWap, 2024-10-29:
Hello Guys, maybe someone can help me with my problems.

I have running wm-esp32.yaml and got my Diehl IZAR RC 868 I R4 PL running and also working in my HA.  Unfortunately, I only get information like battery life and alarm etc. Water values ​​are not displayed in the HA or in the log. Did I make a mistake in the config or does anyone have a tip for me?
When I translate the telegram I have water values...

Debug log:
```
[09:05:53][D][mbus:034]: Processing T1 A frame
[09:05:53][V][mbus:045]: Frame: 36571C99934D4EC5932D6B1358E39C2ECC9398D2CD34B59634B64D4CEA6CB1A3594F2A4D5A6C8DB0DA715B471C (45) [RAW]
[09:05:53][V][mbus:052]: Frame: 1944A51178073087022438E7A1311300135172F886157EF10AE181FD0C44 (30) [with CRC]
[09:05:53][V][mbus:095]: Validating CRC for Block1
[09:05:53][V][crc:031]:     calculated: 0x38E7, read: 0x38E7
[09:05:53][V][mbus:115]: Validating CRC for Block2
[09:05:53][V][crc:031]:     calculated: 0x0C44, read: 0x0C44
[09:05:53][V][mbus:062]: Frame: 1944A511780730870224A1311300135172F886157EF10AE181FD (26) [without CRC]
[09:05:53][D][wmbus:097]: Using driver 'izar' for ID [0x87300778] RSSI: -22 dBm LQI: 128 Frame: T1 A T: 1944A511780730870224A1311300135172F886157EF10AE181FD (26)
[09:05:53][V][wmbus:121]: Publishing 'current_alarms' = 0.0000
[09:05:53][D][wmbus:827]: Alarm message: \xdc$\xfb?, error code: 0
[09:05:53][V][wmbus:121]: Publishing 'previous_alarms' = 0.0000
[09:05:53][D][wmbus:857]: Alarm message: \xdc$\xfb?, error code: 0
[09:05:53][V][wmbus:121]: Publishing 'remaining_battery_life_y' = 9.5000
[09:05:53][V][wmbus:121]: Publishing 'transmit_period_s' = 8.0000
```

wmbusmeter online:
```
telegram=|1944A511780730870224A1_311300135172F886157EF10AE181FD|

Auto driver    : izar
Similar driver : unknown 00/00
Using driver   : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 30870224 dll-id (24028730)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a1 tpl-ci-field (Mfct specific)
011 C?: 311300135172F886157EF10AE181FD mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"24028730",
    "last_month_total_m3":0,
    "remaining_battery_life_y":9.5,
    "total_m3":15.517,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-12-01",
    "previous_alarms":"no_alarm",
    "timestamp":"2024-10-29T09:06:53Z"
}

Using: wmbusmeters: 1.17.1-18-g26760f3
26760f3838aaffe0830a73316fc6b0a36ec2300b
```

Here is my wm-esp32.yaml:
```
---
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## This version shows data from the defined watermeter and
## uses the watermeterID (HEX) from the secrets.yaml file.
##
## usage:  1. Using the last wM-Bus component
[... 1144 lines cut ...]
  # simple rest get message for all watermeter data
  # (disable this if you do not need rest service)
  - platform: template
    id: watermeterdata
    internal: true
    disabled_by_default: true
    lambda: |-
      char buf[128];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(watermonth).state,
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;
```

Thank you very much for every help i could receive.

zibous, 2024-10-29:
you can try this version

```yaml
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## This version shows data from the defined watermeter and
## uses the watermeterID (HEX) from the secrets.yaml file.
##
## Usage:
##     + Using Homassistant API (discovery + devices)
##     + Using the last wM-Bus component
##     + Using wmbusmeters i a pipe (optional)
##
##     + REST:  curl -i http://water-meter-esp.local/text_sensor/watermeterdata
##              curl -i curl -i http://water-meter-esp.local/sensor/wasseruhr_anzeige
##
##
## ---------------------------------------------------------------------------
substitutions:

  ## device settings
  hostname: "water-meter-esp"
  friendly_name: "Wasseruhr"
  device_description: "Production 2024: Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) my Watermeter"

  # The format of the name should be author_name.project_name.
  projectname: "Peter Siebler.Diehl IZAR RC 868"
  appversion: "Diehl IZAR RC 868 v2.1.2 (wm-esp32.yaml)"
  hardware: "ESP32_Devkit_V4"
  location: "Heizraum"

  ## logger settings
  log_level: "INFO" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "ERROR" # Loglevel for wmbus meters component
  log_baudrate: "0" # use 0 to disable serial UART port logging.

  # The built-in LED on the AZ-Delivery DevKit V4 is connected to GPIO 2.
  led_internal: "GPIO2"

## -----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4,ESP32 240MHz, 520KB RAM, 4MB Flash
## -----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Customize the frequency in which data is flushed to the flash.
preferences:
  flash_write_interval: 10min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${hostname}
  build_path: ./build/${hostname}
  comment: ${device_description}
  on_boot:
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - globals.set:
          id: send_millisecond
          value: !lambda "return millis();"
      - logger.log:
          level: WARN
[... 1187 lines cut ...]
  # you can get the sensor data with:
  #   curl -i http://water-meter-esp.local/text_sensor/watermeterdata
  - platform: template
    id: watermeterdata
    internal: true
    disabled_by_default: true
    lambda: |-
      char buf[128];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(watermonth).state,
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;
# e.o.f configuration esp32

```

WulleWulleWapWap, 2024-10-29:
i did, same log:

```
[15:33:33][V][rxLoop:167]: Have 45 bytes from CC1101 Rx, RSSI: -22 dBm LQI: 128
[15:33:33][D][mbus:034]: Processing T1 A frame
[15:33:33][V][mbus:045]: Frame: 36571C99934D4EC5932D6B1358E39C2ECC9398D4CD34B59634B64D58D7199968E5A658F26B1B31C99A5396969A (45) [RAW]
[15:33:33][V][mbus:052]: Frame: 1944A51178073087022438E7A171130013510145A0B9F9BE6D8DE5F79F66 (30) [with CRC]
[15:33:33][V][mbus:095]: Validating CRC for Block1
[15:33:33][V][crc:031]:     calculated: 0x38E7, read: 0x38E7
[15:33:33][V][mbus:115]: Validating CRC for Block2
[15:33:33][V][crc:031]:     calculated: 0x9F66, read: 0x9F66
[15:33:33][V][mbus:062]: Frame: 1944A511780730870224A171130013510145A0B9F9BE6D8DE5F7 (26) [without CRC]
[15:33:33][D][wmbus:097]: Using driver 'izar' for ID [0x87300778] RSSI: -22 dBm LQI: 128 Frame: T1 A T: 1944A511780730870224A171130013510145A0B9F9BE6D8DE5F7 (26)
[15:33:33][V][wmbus:121]: Publishing 'current_alarms' = 0.0000
[15:33:33][D][wmbus:925]: Alarm message: \xdc$\xfb?, error code: 0
[15:33:33][V][wmbus:121]: Publishing 'previous_alarms' = 0.0000
[15:33:33][D][wmbus:957]: Alarm message: \xdc$\xfb?, error code: 0
[15:33:33][V][wmbus:121]: Publishing 'remaining_battery_life_y' = 9.5000
[15:33:33][V][wmbus:121]: Publishing 'transmit_period_s' = 8.0000
```

zibous, 2024-10-30:
testcase see:
https://github.com/zibous/ha-watermeter/blob/master/esphome/water-meter-testcase.yaml
https://github.com/zibous/ha-watermeter/tree/master/esphome/webserver/v4

WulleWulleWapWap, 2024-10-30:
water-meter-testcase.yaml, same...

```
[08:30:00][V][rxLoop:167]: Have 45 bytes from CC1101 Rx, RSSI: -22 dBm LQI: 128
[08:30:00][D][mbus:034]: Processing T1 A frame
[08:30:00][V][mbus:045]: Frame: 36571C99934D4EC5932D6B1358E39C2ECC9398D58D34B59634B64D64DB2C674CAC374B19696B349A6A669A5366 (45) [RAW]
[08:30:00][V][mbus:052]: Frame: 1944A51178073087022438E7A1011300135151885CE81C85608CAAFAA91A (30) [with CRC]
[08:30:00][V][mbus:095]: Validating CRC for Block1
[08:30:00][V][crc:031]:     calculated: 0x38E7, read: 0x38E7
[08:30:00][V][mbus:115]: Validating CRC for Block2
[08:30:00][V][crc:031]:     calculated: 0xA91A, read: 0xA91A
[08:30:00][V][mbus:062]: Frame: 1944A511780730870224A1011300135151885CE81C85608CAAFA (26) [without CRC]
[08:30:00][D][wmbus:097]: Using driver 'izar' for ID [0x87300778] RSSI: -22 dBm LQI: 128 Frame: T1 A T: 1944A511780730870224A1011300135151885CE81C85608CAAFA (26)
[08:30:00][V][wmbus:121]: Publishing 'current_alarms' = 0.0000
[08:30:00][D][wmbus:911]: Alarm message: \xdc$\xfb?, error code: 0
[08:30:00][V][wmbus:121]: Publishing 'previous_alarms' = 0.0000
[08:30:00][D][wmbus:942]: Alarm message: \xdc$\xfb?, error code: 0
[08:30:00][V][wmbus:121]: Publishing 'remaining_battery_life_y' = 9.5000
[08:30:00][V][wmbus:121]: Publishing 'transmit_period_s' = 8.0000
```

zibous, 2024-10-30:
> Don't know why its not working with the old version, but with this one it looks good, now have to rebuild more of your work, to test it with HA.

Yes, i think this is based on the changes for the Szczepan Leon component v4. Maybe the version 3.0 do not work 
anymore with the wmbus driver.

WulleWulleWapWap, 2024-10-31:
Ok ty. at least all other components are working now. :) 

issue can be closed.

closed 2024-10-31 by zibous