zibous/ha-watermeter #54: can't run esp32.yaml

nevusZ, 2024-11-22:
Hy,
thank you for your work, I am trying to get it to work for me.
I can run the esp32-test.yaml and found my watermeter id

But I don't get the esp32.yaml to work?!

`INFO ESPHome 2024.10.3
INFO Reading configuration /config/esphome-web-9ea670.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@3.2.3
Failed config

script: [source /config/esphome-web-9ea670.yaml:537]
  - id: set_status_message
    then: 
      - lambda: |-
          int msgcode = int(id(cc1101_state));
          std::string message = id(cc1101_state_message)[msgcode];
          ESP_LOGD("wmbus", "cc1101 state message: %s, error code: %d", message.c_str(), msgcode);
          id(watermeter_status_message).publish_state(message);
  - id: post_device_state
    then: 
      - if: 
          condition: 
            lambda: return "false =='true'";
          then: 
            - logger.log: 
                tag: system
                level: INFO
                format: Post new data to the 192.168.11.55esphome-web-9ea670, service enabled: false
            - http_request.post: 
                
                Expected a file scheme or a URL scheme with host.
                url: 192.168.11.55/esphome-web-9ea670
                
                verify_ssl has moved to the base component configuration.
                verify_ssl: False
                headers: 
                  Content-Type: application/json`

i don't understand what are these secret definitions?

# UDP server wmbusmeter
wmbus_server:  192.168.11.55
wmbus_server_port: 7337

# Using wmbusmeters i a pipe
watermeterhost: 192.168.11.55
wmbusmeter_host: 192.168.11.55

# service target
service_host: 192.168.11.55
data_host: 192.168.11.55

I changed the webserver to my homeassistant, and i can access those files:

## webserver 3.0 change url to your webserver
webserver_jsurl:  "http://192.168.11.55:8123/local/www.js"
webserver_cssurl: "http://192.168.11.55:8123/local/webserver.css"

any help appreciated

nevusZ, 2024-11-22:
`[14:59:38][D][mbus:034]: Processing T1 A frame
[14:59:38][D][wmbus:097]: Using driver 'izar' for ID [0x33120778] RSSI: -57 dBm LQI: 128 Frame: T1 A T: 1944A511780712330024A201090013872BC0856039044E9293E6 (26)
[14:59:38][D][sensor:094]: '0x33120778 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[14:59:38][D][sensor:094]: '0x33120778 Watermeter CC1101 RSSI': Sending state -57.00000 % with 0 decimals of accuracy
[14:59:38][D][sensor:094]: '0x33120778 Watermeter current alarm code': Sending state 0.00000  with 0 decimals of accuracy
[14:59:38][D][sensor:094]: '0x33120778 Water current month': Sending state 6414.00000 L with 2 decimals of accuracy
[14:59:38][D][sensor:094]: '0x33120778 Water last month': Sending state 7.34800 m³ with 3 decimals of accuracy
[14:59:38][D][sensor:094]: '0x33120778 Watermeter pervious alarm code': Sending state 0.00000  with 0 decimals of accuracy
[14:59:38][D][sensor:094]: '0x33120778 Watermeter Battery Life': Sending state 4.50000 Years with 2 decimals of accuracy
[14:59:38][D][sensor:094]: '0x33120778 Watermeter Display': Sending state 13.76200 m³ with 3 decimals of accuracy
[14:59:38][I][wmbus:359]: Water Display value: 13.762, last value: 13.762
[14:59:38][D][sensor:094]: '0x33120778 Watermeter transmit periode': Sending state 8.00000 sec with 2 decimals of accuracy
[14:59:38][W][component:237]: Component wmbus took a long time for an operation (84 ms).
[14:59:38][W][component:238]: Components should block for at most 30 ms.
[14:59:47][D][esp32.preferences:114]: Saving 2 preferences to flash...
[14:59:47][D][esp32.preferences:143]: Saving 2 preferences to flash: 0 cached, 2 written, 0 failed`

nevusZ, 2024-11-25:
hy, i can run esp32-simple
but i think i have that issue
https://github.com/zibous/ha-watermeter/issues/34

nevusZ, 2024-11-25:
[logs_water-meter-esp_run.txt](https://github.com/user-attachments/files/17907785/logs_water-meter-esp_run.txt)
this is the "VERY_VERBOSE" log

nevusZ, 2024-11-25:
[logs_water-meter-esp_run (1).txt](https://github.com/user-attachments/files/17908093/logs_water-meter-esp_run.1.txt)
this is the esp32-test.yaml log

nevusZ, 2024-11-25:
my soldering skills/equipment is noob.
is this problem software related or hardware/wiring?

SzczepanLeon, 2024-11-25:
It is software/config problem. CC1101 is connected correctly.

nevusZ, 2024-11-26:
i made these modifications to original code:
source: github://SzczepanLeon/esphome-components@**3.2.3**
uncommented #safe_mode: false
remove       - !secret local_sntp

wifi settings adjusted
watermeterId: 0x33120778

nevusZ, 2024-11-30:
@zibous 
what am I doing wrong?

pacraf, 2024-12-03:
@nevusZ how are you able to compile this yaml... 
I think that I am close with  "wm-esp32-simple.yaml" where after similar changes like you (especially:source: github://SzczepanLeon/esphome-components@3.2.3) 
i am stuck now with:
```
INFO ESPHome 2024.9.2
INFO Reading configuration /root/config/watermeter.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@3.2.3
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config
esphome: None
  name: water-meter-esp
  comment: |-
    Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) - all watermeters
  project: 
    name: Diehl IZAR RC 868.Watermeter
    version: 2.0.7
  build_path: ./build/water-meter-esp
  on_boot: 
    - priority: 200.0
      then: 
        - globals.set: 
            id: boot_counter
            value: !lambda |-
              return id(boot_counter)+=1;
        - globals.set: 
            Couldn't find ID 'send_millisecond'. Please check you have defined an ID with that name in your configuration.
            id: send_millisecond
            value: !lambda |-
              return millis();
```

honestly to say, I have not tried proposed  in readme way with docker , but I am just using current version I have of ESPHome. 
Is it that if this release is from 2023 May, then I should downgrade also esphome to this period of time?

pacraf, 2024-12-03:
ok, so instead of taking send_millisecond from lambda function, I replaced it with empty string and test program is loaded, and frames from meter appeared also.  now have to figure out what next...

nevusZ, 2024-12-05:
great you figured it out
but i don't have that error
very strange mine has that problem

[09:47:06]	[W] [component:238]: Components should block for at most 30 ms.
[09:47:06]	[W] [component:237]: Component wmbus took a long time for an operation (118 ms).
[09:47:06]	[D] [sensor:094]: '0x33120778 Watermeter transmit periode': Sending state 8.00000 sec with 2 decimals of accuracy
[09:47:06]	[I] [wmbus:360]: Water Display value: 17.279, last value: 17.279
[09:47:06]	[D] [sensor:094]: '0x33120778 Watermeter Display': Sending state 17.27900 m³ with 3 decimals of accuracy
[09:47:06]	[D] [sensor:094]: '0x33120778 Watermeter Battery Life': Sending state 4.50000 Years with 2 decimals of accuracy
[09:47:06]	[D] [sensor:094]: '0x33120778 Watermeter pervious alarm code': Sending state 0.00000 with 0 decimals of accuracy
[09:47:06]	[D] [sensor:094]: '0x33120778 Water last month': Sending state 16.28000 m³ with 3 decimals of accuracy
[09:47:06]	[D] [sensor:094]: '0x33120778 Water current month': Sending state 999.00000 L with 2 decimals of accuracy
[09:47:06]	[D] [sensor:094]: '0x33120778 Watermeter current alarm code': Sending state 0.00000 with 0 decimals of accuracy
[09:47:06]	[D] [sensor:094]: '0x33120778 Watermeter CC1101 RSSI': Sending state -54.00000 % with 0 decimals of accuracy
[09:47:06]	[D] [sensor:094]: '0x33120778 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[09:47:06]	[D] [wmbus:097]: Using driver 'izar' for ID [0x33120778] RSSI: -54 dBm LQI: 128 Frame: T1 A T: 1944A511780712330024A25109001387DD24EA67B3BBF025EEE9 (26)
[09:47:06]	[D] [mbus:034]: Processing T1 A frame

pacraf, 2024-12-05:
what you mean by problem ? this warning about longer time of operation?
I also have this warnings (slightly lower time but anyway), but it is not a problem I think... rather only warning so you are aware in case of other tasks, if you would like to add there - to consider it, that this task blocks for ~0,1s your system.
but this is only my understanding - not sure tbh ;)

SzczepanLeon, 2024-12-05:
> ok, but how to get the values to home assistant?

[09:47:06] [D] [sensor:094]: '0x33120778 Watermeter CC1101 RSSI': Sending state -54.00000 % with 0 decimals of accuracy

It means that device is sending value to HA.
Can you see that ESPHome device in HA?

nevusZ, 2024-12-05:
it is all here :D
sorry, this is my first approach with esp32 device
thank you all

pacraf, 2024-12-05:
all what in your log have this : "[D] [sensor:094]" should appear in HA.
look carefully in settings >> devices and services>>esphome>>watermeter
there you can select what will go to the dashboard.
I have only two selected (current period cons. and total cons), but by
default I think it would all go to dashboard, so it is strange that you see
only two...

but as said - from log it looks that esphome is sending all of them.

pacraf, 2024-12-05:
probably I know what is your mistake, you look in energy dashboard >watersource data , and there indeed only two are visible (and this is correct) because only this two are valid source of water consumption for energy dashboard. 
I am pretty sure , as described above in setting, you will finad all of them, and you can pick (if you want) all of them to put in standard dashboard (not energy).

nevusZ, 2024-12-05:
when i press configure i can only select "Allow the device to perform Home Assistant actions."
[image]
these are the entities

pacraf, 2024-12-05:
Menu I am talking is one position above like this example graphic shows. Are you there ? 
Better look on YouTube is lot of examples…
I am also HA newbie in fact…
https://images.app.goo.gl/AajxNxX25invicVE8

nevusZ, 2024-12-05:
[image]

pacraf, 2024-12-05:
you got it :)

nevusZ, 2024-12-06:
no other values than current month and last month.
i use esp32-simple.yaml
i cannot run esp32.yaml

`INFO ESPHome 2024.11.2
INFO Reading configuration /config/esphome/water-meter-esp.yaml...
WARNING Found '${service}' (see http_request->useragent) which looks like a substitution, but 'service' was not declared
WARNING Found 'return "${service_enabled} =='true'";' (see script->1->then->0->if->condition->lambda) which looks like a substitution, but 'service_enabled' was not declared
WARNING Found 'Post new data to the ${service_url}${device_name_short}, service enabled: ${service_enabled}' (see script->1->then->0->if->then->0->logger.log->format) which looks like a substitution, but 'service_url' was not declared
WARNING Found 'Post new data to the ${service_url}${device_name_short}, service enabled: ${service_enabled}' (see script->1->then->0->if->then->0->logger.log->format) which looks like a substitution, but 'service_enabled' was not declared
WARNING Found '${service_url}/${device_name_short}' (see script->1->then->0->if->then->1->http_request.post->url) which looks like a substitution, but 'service_url' was not declared
WARNING Found 'if ((id(last_value) > 0.00) and (id(waterdisplay).state)>(id(last_value)) ) {
  id(cc1101_state) = 2;
  ESP_LOGI("wmbus", "Water Display value: %.3f, last value: %.3f", id(waterdisplay).state, id(last_value));
  id(current_value) = float(id(waterdisplay).state-id(last_value)) * 1000.00;
  id(watercurrent).publish_state(id(current_value));
  id(hour_value)+=id(current_value);
  id(waterhour).publish_state(id(hour_value));
  id(daily_value)+=id(current_value);
  id(waterday).publish_state(id(daily_value));
  id(week_value)+=id(current_value);
  id(waterweek).publish_state(id(week_value));
  id(year_value)+=id(current_value);
  id(wateryear).publish_state(id(year_value));
  ESP_LOGD("wmbus", "Set current value to %.3f litre and publish the data", id(current_value));
  if ( ${service_enabled} == true){
    id(post_device_state).execute();
  }else{
    ESP_LOGD("wmbus", "Data Service to host disabled");
  }
}else{
  id(current_value) = 0.00;
  id(cc1101_state) = 1;
  id(watercurrent).publish_state(id(current_value));
  ESP_LOGI("wmbus", "Reset current value to: %.3f, Waterdisplay value: %.3f, last value: %.3f", id(current_value), id(waterdisplay).state, id(last_value));
}
id(last_value)=id(waterdisplay).state;' (see sensor->0->total_water_m3->on_value->then->0->lambda) which looks like a substitution, but 'service_enabled' was not declared
WARNING Found '${service}' (see http_request->useragent) which looks like a substitution, but 'service' was not declared
WARNING Found 'return "${service_enabled} =='true'";' (see script->1->then->0->if->condition->lambda) which looks like a substitution, but 'service_enabled' was not declared
WARNING Found 'Post new data to the ${service_url}water-meter-esp, service enabled: ${service_enabled}' (see script->1->then->0->if->then->0->logger.log->format) which looks like a substitution, but 'service_url' was not declared
WARNING Found 'Post new data to the ${service_url}water-meter-esp, service enabled: ${service_enabled}' (see script->1->then->0->if->then->0->logger.log->format) which looks like a substitution, but 'service_enabled' was not declared
WARNING Found '${service_url}/water-meter-esp' (see script->1->then->0->if->then->1->http_request.post->url) which looks like a substitution, but 'service_url' was not declared
WARNING Found 'if ((id(last_value) > 0.00) and (id(waterdisplay).state)>(id(last_value)) ) {
  id(cc1101_state) = 2;
  ESP_LOGI("wmbus", "Water Display value: %.3f, last value: %.3f", id(waterdisplay).state, id(last_value));
  id(current_value) = float(id(waterdisplay).state-id(last_value)) * 1000.00;
  id(watercurrent).publish_state(id(current_value));
  id(hour_value)+=id(current_value);
  id(waterhour).publish_state(id(hour_value));
  id(daily_value)+=id(current_value);
  id(waterday).publish_state(id(daily_value));
  id(week_value)+=id(current_value);
  id(waterweek).publish_state(id(week_value));
  id(year_value)+=id(current_value);
  id(wateryear).publish_state(id(year_value));
  ESP_LOGD("wmbus", "Set current value to %.3f litre and publish the data", id(current_value));
  if ( ${service_enabled} == true){
    id(post_device_state).execute();
  }else{
    ESP_LOGD("wmbus", "Data Service to host disabled");
  }
}else{
  id(current_value) = 0.00;
  id(cc1101_state) = 1;
  id(watercurrent).publish_state(id(current_value));
  ESP_LOGI("wmbus", "Reset current value to: %.3f, Waterdisplay value: %.3f, last value: %.3f", id(current_value), id(waterdisplay).state, id(last_value));
}
id(last_value)=id(waterdisplay).state;' (see sensor->0->total_water_m3->on_value->then->0->lambda) which looks like a substitution, but 'service_enabled' was not declared
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@3.2.3
Failed config

http_request: [source /config/esphome/water-meter-esp.yaml:523]
  
  ESPHome supports certificate verification only via ESP-IDF. Set 'verify_ssl: false' to skip certificate validation and allow less secure HTTPS connections.
  id: http_request_data
  useragent: ${service}
  timeout: 10s`

pacraf, 2024-12-06:
I also can’t run other than “simple”, but this simple is fine for me.  I have current consumption of water in HA, last month, this month … what else are you missing? 
honestly I was so much surprised that every 8s it sends data , and battery last for so many years…

closed 2024-12-21 by github-actions[bot]