zibous/ha-watermeter #55: Compatibility with DIEHL LNE6250

Anoxym, 2024-11-26:
Hi,

Is this project could work with this water meter : 

[image]
[image]

Thanks

SzczepanLeon, 2024-11-26:
99% to yes

Anoxym, 2024-11-26:
Great thanks I'll try :)

g0mml, 2024-11-26:
I have the same water meter as in the picture. Using your 🛠 ESP32 Test config, I got the following results:

> [00:29:52][D][mbus:034]: Processing T1 A frame
> [00:29:52][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 138.00000 lqi with 0 decimals of accuracy
> [00:29:53][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -82.00000 % with 0 decimals of accuracy
> [00:29:54][D][wmbus:182]: Meter ID [0x92030778] RSSI: -82 dBm LQI: 138 Frame: T1 A not found in configuration T: 1944A511780703929021A2410F001328ACB3F245D8E43329C3F9 (26)
> [00:33:06][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 136.00000 lqi with 0 decimals of accuracy
> [00:33:06][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -82.00000 % with 0 decimals of accuracy
> [00:33:06][D][wmbus:182]: Meter ID [0x92030778] RSSI: -82 dBm LQI: 136 Frame: T1 A not found in configuration T: 1944A511780703929021A2310F001328CC7F0E143DDF3E288CF4 (26)
> [00:33:11][D][sensor:094]: 'Device Boot counter': Sending state 13.00000  with 0 decimals of accuracy
> [00:33:15][D][mbus:034]: Processing T1 A frame
> [00:33:15][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 142.00000 lqi with 0 decimals of accuracy
> [00:33:15][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -82.00000 % with 0 decimals of accuracy
> [00:33:15][D][wmbus:182]: Meter ID [0x92030778] RSSI: -82 dBm LQI: 142 Frame: T1 A not found in configuration T: 1944A511780703929021A2410F001328A9B3F245D8E43329C3F9 (26)
> [00:33:39][D][mbus:034]: Processing T1 A frame
> [00:33:39][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 135.00000 lqi with 0 decimals of accuracy
> [00:33:39][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -83.00000 % with 0 decimals of accuracy
> [00:33:39][D][wmbus:182]: Meter ID [0x92030778] RSSI: -83 dBm LQI: 135 Frame: T1 A not found in configuration T: 1944A511780703929021A2710F00132890C228B8BA905944E8FE (26)

Is there anything wrong/strange with the output? Is a LQI of 135 bad or good?

For the final config I would have to use the  "0x92030778" as meter ID right?

Thanks for your effort :-)

g0mml, 2024-11-26:
I have the same meter as you and I was able to get the data using the the 🛠 ESP32 Test configuration.

Anoxym, 2024-11-29:
Hi,

I try to deploy the esp32 test on my esp32 with ESPHome on Home Assistant but it won't work, I don't understand, can we have a minimalist code to get the ID ?

[image]

SzczepanLeon, 2024-11-29:
Please read informations on my main repo page.
Minimal YAwmbis described there.

Anoxym, 2024-11-29:
Did you have a working esphome configuration to get the ID ? 

with this : 

wmbus:
  all_drivers: True
  log_all: True

What did I need to get a working code ?

zibous, 2024-11-29:
for izar watermeter:

https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/wm-esp32-izar-test.yaml

Anoxym, 2024-11-29:
I have the same error : 

[image]

SzczepanLeon, 2024-11-29:
Userfriendly solution is ongoing:
https://www.olx.pl/d/oferta/zintegrowany-czytnik-wm-bus-ultimatereader-wodomierze-cieplomierze-CID99-ID13cBcJ.html
[image]

[image]

Anoxym, 2024-11-29:
It works adding "platform: esphome" : 

ota:
  platform: esphome
  password: "wmi01-testcase"
  on_begin:
    then:
      - logger.log:
          format: "OTA Start"
          tag: "OTA"
          level: WARN
  on_progress:
    then:
      - logger.log:
          level: WARN
          tag: "OTA"
          format: "OTA progress %0.1f%%"
          args: ["x"]
  on_end:
    then:
      - logger.log:
          format: "OTA End"
          tag: "OTA"
          level: WARN
  on_error:
    then:
      - logger.log:
          format: "OTA update error %d"
          tag: "OTA"
          level: ERROR
          args: ["x"]

We need to be close to the watermeter to get ID ?

Anoxym, 2024-11-29:
[17:17:14][V][rxLoop:167]: Have 56 bytes from CC1101 Rx, RSSI: -52 dBm LQI: 128
[17:17:14][D][mbus:034]: Processing T1 A frame
[17:17:14][V][mbus:045]: Frame: 37271C99934D70D38B39638D4E3593966A6C4E6A535963565994DAB255A39ACA65CA538E2E969964EB1CA4E5A6D199AC8D965A9A62D3A72A (56) [RAW]
[17:17:14][V][mbus:052]: Frame: 1E44A511412320217B079AF87AF700100576890BA8F9E9223F655284F20AC5A8B556AA37FE (37) [with CRC]
[17:17:14][V][mbus:095]: Validating CRC for Block1
[17:17:14][V][crc:031]:     calculated: 0x9AF8, read: 0x9AF8
[17:17:14][V][mbus:115]: Validating CRC for Block2
[17:17:14][V][crc:031]:     calculated: 0xF20A, read: 0xF20A
[17:17:14][V][mbus:115]: Validating CRC for Block3
[17:17:14][V][crc:031]:     calculated: 0x37FE, read: 0x37FE
[17:17:14][V][mbus:062]: Frame: 1E44A511412320217B077AF700100576890BA8F9E9223F655284C5A8B556AA (31) [without CRC]
[17:17:14][D][wmbus:182]: Meter ID [0x21202341] RSSI: -52 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077AF700100576890BA8F9E9223F655284C5A8B556AA (31)
[17:17:28][V][rxLoop:167]: Have 56 bytes from CC1101 Rx, RSSI: -49 dBm LQI: 128
[17:17:28][D][mbus:034]: Processing T1 A frame
[17:17:28][V][mbus:045]: Frame: 37271C99934D70D38B39638D4E3593966A6C4E6A535963565994DAB255A39ACA65CA538E2E969964EB1CA4E5A6D199AC8D965A9A62D3A72A (56) [RAW]
[17:17:28][V][mbus:052]: Frame: 1E44A511412320217B079AF87AF700100576890BA8F9E9223F655284F20AC5A8B556AA37FE (37) [with CRC]
[17:17:28][V][mbus:095]: Validating CRC for Block1
[17:17:28][V][crc:031]:     calculated: 0x9AF8, read: 0x9AF8
[17:17:28][V][mbus:115]: Validating CRC for Block2
[17:17:28][V][crc:031]:     calculated: 0xF20A, read: 0xF20A
[17:17:28][V][mbus:115]: Validating CRC for Block3
[17:17:28][V][crc:031]:     calculated: 0x37FE, read: 0x37FE
[17:17:28][V][mbus:062]: Frame: 1E44A511412320217B077AF700100576890BA8F9E9223F655284C5A8B556AA (31) [without CRC]
[17:17:28][D][wmbus:182]: Meter ID [0x21202341] RSSI: -49 dBm LQI: 128 Frame: T1 A not found in configuration T: 1E44A511412320217B077AF700100576890BA8F9E9223F655284C5A8B556AA (31)
[17:17:31][D][esp32.preferences:114]: Saving 1 preferences to flash...
[17:17:31][V][esp32.preferences:137]: NVS data not changed skipping 3484741779  len=7
[17:17:31][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed

Meter ID is 0x21202341 ? even if it says "T1 A not found in configuration T" ?

Anoxym, 2024-11-29:
Can someone help me with the code : 

`
substitutions:
  device_name_short: "watermeter"
  device_description: "Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) my Watermeter"
  projectname: "Diehl IZAR RC 868.Watermeter"
  appversion: "2.0.7"
  
  log_level: "WARN"
  log_wmbus: "ERROR"
  log_baudrate: "0"

esphome:
  name: ${device_name_short}
  comment: ${device_description}
  name_add_mac_suffix: false
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}
  on_boot:
    - priority: -100.00
      then:
        - globals.set:
            id: boot_counter
            value: !lambda "return id(boot_counter)+=1;"
        - logger.log:
            level: WARN
            tag: "system"
            format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
        - component.update: bootcounter
    - priority: 800
      then:
        - logger.log:
            level: WARN
            tag: "system"
            format: "BOOTMESSAGE:${device_name_short} hardware initialization of vital components is executed "

  on_shutdown:
    priority: 700
    then:
      - logger.log:
          level: ERROR
          tag: "system"
          format: "BOOTMESSAGE:${device_name_short} is down!"

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

  - source:
      type: local
      path: custom_components
    components: [backup]

backup:
  auth:
    username: "admin"
    password: "admin"
  force_update: false

globals:

[... 182 lines cut ...]
    name: "Device WLAN RSSI"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"

  - platform: template
    name: Device Boot counter
    id: bootcounter
    accuracy_decimals: 0
    state_class: "measurement"
    entity_category: "diagnostic"
    lambda: return (id(boot_counter));

  - platform: template
    name: "Device Last Boot"
    disabled_by_default: true
    id: device_lastBoot_time
    icon: mdi:clock-start

`

There is errors and I don't know why, maybe esphome can't get to wmbus ? : 

[image]

SzczepanLeon, 2024-11-29:
> source: github://SzczepanLeon/esphome-components@main

If you want to stay with this YAML format, then replace **main** with **version_3**

Anoxym, 2024-11-29:
Same errors but the url with @version_3 works

Anoxym, 2024-11-29:
I found the error and managed to upload the code to ESP. On Home Assistant I can show the Water  meter but entities are Unknown : 

[image]

closed 2024-12-14 by github-actions[bot]