zibous/ha-watermeter #57: Water cool and water hot

ediazrod, 2025-01-13:
I see the proyect, but I am interested on read water cool and water hot at the same time? this is possible? or must use two readers...

Regards!

SzczepanLeon, 2025-01-13:
> this is possible? 

Yes it is possible. Just add another sensor.
See example [here](https://github.com/SzczepanLeon/esphome-components).

ediazrod, 2025-01-23:
Thanks man, I will try to implement this module, look complex but it is great. at least you module works great!

ediazrod, 2025-01-29:
Hi  @ SzczepanLeon I am updating the code to adapt for hot and cold water reader, now is workiing the basic but I dont understand the secrets parameter
 service_url: !secret service_host
  service: "izar.watermeter"

What is service_host? the ip of the watermetter? the ip of home asistant what is the format http://mywatermeterip/

Sorry for ask this questions but I am very bad programer...

SzczepanLeon, 2025-01-29:
Ask @zibous , this is his YAML.

closed 2025-02-13 by github-actions[bot]