zibous/ha-watermeter #59: Does not work

Onepamopa, 2025-04-08:
Flashed the example esp32 with the module connected to the described GPIOs.

The ESP barely responds to ping, 85% timeout, no web interface, nothing.

zibous, 2025-04-09:
Try this:

```yaml
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## This version shows data from the defined watermeter and
## uses the watermeterID (HEX) from the secrets.yaml file.
##
## Usage:
##     + Using Homassistant API (discovery + devices)
##     + Using the last wM-Bus component
##
##     + REST:  curl -i http://water-meter-esp.local/text_sensor/watermeterdata
##              curl -i curl -i http://water-meter-esp.local/sensor/wasseruhr_anzeige
##
##
## ---------------------------------------------------------------------------
substitutions:

  ## device settings
  hostname: "water-meter-esp"
  friendly_name: "Wasseruhr"
  device_description: "Production 2024: Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) my Watermeter"

  # The format of the name should be author_name.project_name.
  projectname: "Peter Siebler.Diehl IZAR RC 868"
  appversion: "Diehl IZAR RC 868 v2.1.2 (wm-esp32.yaml)"
  hardware: "ESP32_Devkit_V4"
  location: "Heizraum"

  ## logger settings
  log_level: "INFO" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "ERROR" # Loglevel for wmbus meters component
  log_baudrate: "0" # use 0 to disable serial UART port logging.

  # The built-in LED on the AZ-Delivery DevKit V4 is connected to GPIO 2.
  led_internal: "GPIO2"

## -----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4,ESP32 240MHz, 520KB RAM, 4MB Flash
## -----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Customize the frequency in which data is flushed to the flash.
preferences:
  flash_write_interval: 10min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${hostname}
  build_path: ./build/${hostname}
  comment: ${device_description}
  on_boot:
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - globals.set:
          id: send_millisecond
          value: !lambda "return millis();"
      - logger.log:
          level: WARN
          tag: "system"
[... 1201 lines cut ...]
  # (disable this if you do not need rest service)
  # you can get the sensor data with:
  #   curl -i http://water-meter-esp.local/text_sensor/watermeterdata
  - platform: template
    id: watermeterdata
    internal: true
    disabled_by_default: true
    lambda: |-
      char buf[128];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(watermonth).state,
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;
# e.o.f configuration esp32
```

Onepamopa, 2025-04-09:
Now:
[09:22:33][I][app:100]: ESPHome version 2025.3.3 compiled on Apr  9 2025, 09:20:01
[09:22:33][I][app:102]: Project Peter Siebler.Diehl IZAR RC 868 version Diehl IZAR RC 868 v2.1.2 (wm-esp32.yaml)
[09:22:33][E][wmbus:409]:   Check connection to CC1101!
[09:22:33][E][component:082]:   Component wmbus is marked FAILED

zibous, 2025-04-09:
```yaml
09:22:33][E][wmbus:409]: Check connection to CC1101!
[09:22:33][E][component:082]: Component wmbus is marked FAILED
```

CC1101 connection is not valid or CC1101 is brocken.

```yaml
##
##                              ╭――――――――――――――――――――――――――――╮
##                              │                      CSN ■ │ 1  --> ■ GPIO21
##                              │                     GDO0 ■ │ 2  --> ■ GPIO16
##                              │ ■ GND               GDO2 ■ │ 3  --> ■ GPIO17
##                              │ ■ ANT      CC1101     SO ■ │ 4  --> ■ GPIO19
##                              │ ■ GND                CLK ■ │ 5  --> ■ GPIO18
##                              │                       SI ■ │ 6  --> ■ GPIO23
##                              │                      GND ■ │ 7  --> ■ GND
##                              │                     3.3V ■ │ 8  --> ■ 3.3V
##                              ╰――――――――――――――――――――――――――――╯
##
```

Onepamopa, 2025-04-12:
I connected https://www.tme.eu/Document/0ea266b82bfc27839fe7df92da7c351b/rcc1101-spi-smt.pdf 
It doesn't give an error, but it also doesn't see any data.

How do I make it log everything? I tried setting watermeterId 0 but it still doesn't show anything?

Onepamopa, 2025-04-12:
Also - I think it crashes several times and boots into safe mode.
When I power it on initially - it starts responding to ping, then drops, then starts, then drops (i.e. more than 70% packet loss).
After a while - it starts responding with 0% packet loss, but there is no web server, no wifi logs via esphome, nothing..

zibous, 2025-04-12:
> How do I make it log everything? 

Try:

```yaml
 ## logger settings
  log_level: "DEBUG" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "DEBUG" # Loglevel for wmbus meters component
  log_baudrate: "9600" # use 0 to disable serial UART port logging.
```

1. Disconnect CC1101
2. Connect the esp via usb cable
3. Clean build firmware and install this via usb cable

If the esp is running, the problem is the CC1101 otherwise the esp do not work

> I tried setting watermeterId 0 but it still doesn't show anything?

If it work go closer to your watermeter

Onepamopa, 2025-04-12:
Flashing via OTA (i.e. via homeassistant esphome builder) works normally, no issues there.
Issues start after flashing, I presume - after the CC1101 receives some data - the ESP starts to restart.

Here is log output:
```
INFO Starting log output from 192.168.254.25 using esphome API
INFO Successfully connected to water-meter-esp @ 192.168.254.25 in 11.235s
INFO Successful handshake with water-meter-esp @ 192.168.254.25 in 1.683s
[14:27:20][I][app:100]: ESPHome version 2025.3.3 compiled on Apr 12 2025, 14:25:34
[14:27:20][I][app:102]: Project Peter Siebler.Diehl IZAR RC 868 version Diehl IZAR RC 868 v2.1.2 (wm-esp32.yaml)
[14:27:20][C][logger:177]: Logger:
[14:27:20][C][logger:178]:   Max Level: DEBUG
[14:27:20][C][logger:179]:   Initial Level: DEBUG
[14:27:20][C][logger:181]:   Log Baud Rate: 9600
[14:27:21][C][logger:182]:   Hardware UART: UART0
[14:27:21][C][logger:186]:   Level for 'api': WARN
[14:27:21][C][logger:186]:   Level for 'api.service': ERROR
[14:27:21][C][logger:186]:   Level for 'component': ERROR
[14:27:21][C][logger:186]:   Level for 'internal_temperature': ERROR
[14:27:21][C][logger:186]:   Level for 'json': ERROR
[14:27:21][C][logger:186]:   Level for 'mqtt': WARN
[14:27:21][C][logger:186]:   Level for 'scheduler': ERROR
[14:27:21][C][logger:186]:   Level for 'sensor': WARN
[14:27:21][C][logger:186]:   Level for 'sensor.filter': WARN
[14:27:21][C][logger:186]:   Level for 'text_sensor': WARN
[14:27:21][C][logger:186]:   Level for 'wMBus-lib': DEBUG
[14:27:21][C][logger:186]:   Level for 'wifi': WARN
[14:27:21][C][logger:186]:   Level for 'wmbus': DEBUG
[14:27:21][C][gpio.output:010]: GPIO Binary Output:
[14:27:21][C][gpio.output:011]:   Pin: GPIO2
[14:27:22][C][template.binary_sensor:028]: Template Binary Sensor 'Wasseruhr Status Led'
[14:27:22][C][template.sensor:022]: Template Sensor 'Wasser Aktuell'
[14:27:22][C][template.sensor:022]:   Device Class: 'water'
[14:27:22][C][template.sensor:022]:   State Class: 'total_increasing'
[14:27:22][C][template.sensor:022]:   Unit of Measurement: 'L'
[14:27:22][C][template.sensor:022]:   Accuracy Decimals: 2
[14:27:22][C][template.sensor:022]:   Icon: 'mdi:water-circle'
[14:27:22][C][template.sensor:023]:   Update Interval: 60.0s
[14:27:22][C][template.sensor:022]: Template Sensor 'Wasser Stunde'
[14:27:22][C][template.sensor:022]:   Device Class: 'water'
[14:27:22][C][template.sensor:022]:   State Class: 'total_increasing'
[14:27:22][C][template.sensor:022]:   Unit of Measurement: 'L'
[14:27:22][C][template.sensor:022]:   Accuracy Decimals: 2
[14:27:22][C][template.sensor:022]:   Icon: 'mdi:water-circle'
[14:27:22][C][template.sensor:023]:   Update Interval: 60.0s
[14:27:23][C][template.sensor:022]: Template Sensor 'Wasser Tag'
[14:27:23][C][template.sensor:022]:   Device Class: 'water'
[14:27:23][C][template.sensor:022]:   State Class: 'total_increasing'
[14:27:23][C][template.sensor:022]:   Unit of Measurement: 'L'
[14:27:23][C][template.sensor:022]:   Accuracy Decimals: 2
[14:27:23][C][template.sensor:022]:   Icon: 'mdi:water-circle'
[14:27:23][C][template.sensor:023]:   Update Interval: 60.0s
[14:27:23][C][template.sensor:022]: Template Sensor 'Wasser Gestern'
[14:27:23][C][template.sensor:022]:   Device Class: 'water'
[14:27:23][C][template.sensor:022]:   State Class: 'total_increasing'
[14:27:23][C][template.sensor:022]:   Unit of Measurement: 'L'
[14:27:23][C][template.sensor:022]:   Accuracy Decimals: 2
[14:27:23][C][template.sensor:022]:   Icon: 'mdi:water-circle'
[14:27:23][C][template.sensor:023]:   Update Interval: 60.0s
[14:27:24][C][template.sensor:022]: Template Sensor 'Wasser Woche'
[14:27:24][C][template.sensor:022]:   Device Class: 'water'
[14:27:24][C][template.sensor:022]:   State Class: 'total_increasing'
[14:27:24][C][template.sensor:022]:   Unit of Measurement: 'L'
[14:27:24][C][template.sensor:022]:   Accuracy Decimals: 2
[14:27:24][C][template.sensor:022]:   Icon: 'mdi:water-circle'
[14:27:24][C][template.sensor:023]:   Update Interval: 60.0s
[14:27:24][C][template.sensor:022]: Template Sensor 'Wasser letzer Monat'
[14:27:24][C][template.sensor:022]:   Device Class: 'water'
[14:27:24][C][template.sensor:022]:   State Class: 'total_increasing'
[14:27:24][C][template.sensor:022]:   Unit of Measurement: 'L'
[... 144 lines cut ...]
[14:27:34][C][wmbus:427]:       Accuracy Decimals: 0
[14:27:34][C][wmbus:427]:       Icon: 'mdi:rss'
[14:27:34][C][wmbus:427]:     Sensor '0x00 Wasseruhr Anzeige'
[14:27:34][C][wmbus:427]:       Device Class: 'water'
[14:27:34][C][wmbus:427]:       State Class: 'total_increasing'
[14:27:34][C][wmbus:427]:       Unit of Measurement: 'm³'
[14:27:34][C][wmbus:427]:       Accuracy Decimals: 3
[14:27:34][C][wmbus:427]:       Icon: 'mdi:counter'
[14:27:34][C][wmbus:427]:     Sensor '0x00 Wasseruhr Update Intervall'
[14:27:34][C][wmbus:427]:       State Class: 'measurement'
[14:27:35][C][wmbus:427]:       Unit of Measurement: 'sec'
[14:27:35][C][wmbus:427]:       Accuracy Decimals: 2
[14:27:35][C][wmbus:427]:       Icon: 'mdi:timelapse'
WARNING water-meter-esp @ 192.168.254.25: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for water-meter-esp @ 192.168.254.25
WARNING Disconnected from API

. . . a few connect / disconnects

[14:30:31][D][wmbus:568]: cc1101 state message: Init, error code: 0
WARNING water-meter-esp @ 192.168.254.25: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for water-meter-esp @ 192.168.254.25
WARNING Disconnected from API

```

Onepamopa, 2025-04-12:
Also - when I move the ESP home (far away from any water meters, it remains stable, i.e. the web ui works, I can reset the boot count, etc.

BUT, if I move it closer to the water meter - i.e. when it has a chance to actually receive data - the crashing starts to happen, 6-7-8-10 restarts, then goes into safe mode.

Onepamopa, 2025-04-12:
Found the problem - 1 of the cables was losely connected to the ESP.

Now:
16:04:29 [D] [wmbus:568] cc1101 state message: Init, error code: 0
16:04:35 [D] [mbus:034] Processing T1 A frame
16:04:44 [D] [mbus:034] Processing T1 A frame
16:04:44 [D] [mbus:034] Processing T1 A frame
16:04:53 [D] [mbus:034] Processing T1 A frame

But... I don't see any additional data, such as the meter ID and so forth?

Onepamopa, 2025-04-12:
Figured it out, it's working normally.

closed 2025-04-12 by Onepamopa

Onepamopa, 2025-08-10:
Can you please help adjusting the old v3 config to v5?
This is the device I purchased: https://www.aliexpress.com/item/1005006860809926.html

Here is the pinout:

[image]

```
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## This version shows data from the defined watermeter and
## uses the watermeterID (HEX) from the secrets.yaml file.
##
## Usage:
##     + Using Homassistant API (discovery + devices)
##     + Using the last wM-Bus component
##
##     + REST:  curl -i http://water-meter-esp.local/text_sensor/watermeterdata
##              curl -i curl -i http://water-meter-esp.local/sensor/wasseruhr_anzeige
##
##
## ---------------------------------------------------------------------------
substitutions:

  ## device settings
  # water-meter-esp.local
  hostname: "water-meter-esp"
  friendly_name: "Къща"
  friendly_name2: "Помещ."
  device_description: "Water meter ESP32, RF - CC1101, IZAR module (Diehl IZAR RC 868 H)"

  # The format of the name should be author_name.project_name.
  projectname: "Dobri Dobrev.Diehl IZAR RC 868 H"
  appversion: "Diehl IZAR RC 868 H"
  hardware: "ESP32_Devkit_V4"
  location: "Tsanko Tserkovski 7"

  ## logger settings
  log_level: "INFO" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "INFO" # Loglevel for wmbus meters component
  log_baudrate: "115200" # use 0 to disable serial UART port logging. 115200

  # The built-in LED on the AZ-Delivery DevKit V4 is connected to GPIO 2.
  led_internal: "GPIO2"

## -----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4,ESP32 240MHz, 520KB RAM, 4MB Flash
## -----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  #board: esp32dev
  framework:
    type: arduino

# Customize the frequency in which data is flushed to the flash.
preferences:
  flash_write_interval: 10min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${hostname}
  build_path: ./build/${hostname}
  comment: ${device_description}
  on_boot:
    then:
      - globals.set:
[... 1863 lines cut ...]
      return s;

  # Kyshtata
  - platform: template
    id: watermeterdata
    internal: true
    disabled_by_default: true
    lambda: |-
      char buf[128];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(watermonth).state,
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;
# e.o.f configuration esp32
```

zibous, 2025-08-10:
> Can you please help adjusting the old v3 config to v5?
> This is the device I purchased: https://www.aliexpress.com/item/1005006860809926.html

If the device is v2 version you can use this:

[wm_heltec.zip](https://github.com/user-attachments/files/21705388/wm_heltec.zip)

Onepamopa, 2025-08-10:
@zibous the new hardware is for V5 SXxxxx
The old configuration was for V3 (C1101). I'm trying to retain as much of the existing values / sensors / etc as possible, but nomatter how I edit - It always gives errors....

Onepamopa, 2025-08-21:
@zibous after updating to latest ha, now getting an error in
## ----------------------------------------------------------------
## Board Heltec wifi lora 32 V2
## ----------------------------------------------------------------
esp32:
  board: heltec_wifi_lora_32_V2
  flash_size: 8MB
  framework:
    type: esp-idf
    version: latest

ESP-IDF latest may be supported by platformio/espressif32; please specify 'platform_version'.

Any ideas ?!

SzczepanLeon, 2025-08-21:
Remove **version**

Onepamopa, 2025-09-30:
Hey guys..
So, the water company replaced the meters with IPERL.
How to adjust the current izar yaml for the new meters?

[image]

[image]

[image]

zibous, 2025-09-30:
> So, the water company replaced the meters with IPERL.
> How to adjust the current izar yaml for the new meters?
You can try:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@version_3

sensor:
  - platform: wmbus
    meter_id: your meter id
    type: iperl
    key: "E6C88800DEB868C0D6A84880CE982840"
    
```
      
Maybe you need a key:
see: https://github.com/wmbusmeters/wmbusmeters/issues/877#issuecomment-1442513839  
I managed to decrypt the encrypted part with the magic key : E6C88800DEB868C0D6A84880CE982840