zibous/ha-watermeter #60: Issues compiling

adverserath, 2025-04-19:
I had my sensor up and running, a few days later I wanted to rename the host and tried to reimage. But now I'm getting compile errors from c++ headers.
Can you give any explanations of what I could be doing wrong? The simple config works fine, but the esp32.yaml doesnt

`Processing water-meter-esp (board: az-delivery-devkit-v4; framework: arduino; platform: espressif32)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - framework-arduinoespressif32-libs @ 5.1.0+sha.632e0c2a9f 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- SPI @ 2.0.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7+sha.b8c6af4
|-- wmbus-drivers @ 0.0.0+20250419021412.sha.c856f7d
Compiling .pioenvs/water-meter-esp/src/esphome/components/http_request/http_request.cpp.o
Compiling .pioenvs/water-meter-esp/src/esphome/components/http_request/http_request_idf.cpp.o
Compiling .pioenvs/water-meter-esp/src/esphome/components/internal_temperature/internal_temperature.cpp.o
Compiling .pioenvs/water-meter-esp/src/esphome/components/logger/logger.cpp.o
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/portable.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:63,
                 from src/esphome/core/helpers.h:27,
                 from src/esphome/components/json/json_util.h:5,
                 from src/esphome/components/http_request/http_request.h:9,
                 from src/esphome/components/http_request/http_request.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:613:48: warning: ignoring attribute 'section (".iram1.24")' because it conflicts with previous 'section (".iram1.22")' [-Wattributes]
  613 | static inline bool IRAM_ATTR xPortCanYield(void)
      |                                                ^
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:343:30: note: previous declaration here
  343 | static inline bool IRAM_ATTR xPortCanYield(void);
      |                              ^~~~~~~~~~~~~
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:632:55: warning: ignoring attribute 'section (".iram1.25")' because it conflicts with previous 'section (".iram1.23")' [-Wattributes]
  632 | static inline BaseType_t IRAM_ATTR xPortGetCoreID(void)
      |                                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:385:36: note: previous declaration here
  385 | static inline BaseType_t IRAM_ATTR xPortGetCoreID(void);
      |                                    ^~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/portable.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:63,
                 from src/esphome/core/helpers.h:27,
                 from src/esphome/components/json/json_util.h:5,
                 from src/esphome/components/http_request/http_request.h:9,
                 from src/esphome/components/http_request/http_request_idf.h:3,
                 from src/esphome/components/http_request/http_request_idf.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:613:48: warning: ignoring attribute 'section (".iram1.24")' because it conflicts with previous 'section (".iram1.22")' [-Wattributes]
  613 | static inline bool IRAM_ATTR xPortCanYield(void)
      |                                                ^
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:343:30: note: previous declaration here
  343 | static inline bool IRAM_ATTR xPortCanYield(void);
      |                              ^~~~~~~~~~~~~
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:632:55: warning: ignoring attribute 'section (".iram1.25")' because it conflicts with previous 'section (".iram1.23")' [-Wattributes]
  632 | static inline BaseType_t IRAM_ATTR xPortGetCoreID(void)
      |                                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:385:36: note: previous declaration here
  385 | static inline BaseType_t IRAM_ATTR xPortGetCoreID(void);
      |                                    ^~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/portable.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:63,
                 from src/esphome/core/helpers.h:27,
[... 190 lines cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_vector.h:730:15:   required from 'std::vector<_Tp, _Alloc>::~vector() [with _Tp = std::array<float, 3>; _Alloc = std::allocator<std::array<float, 3> >]'
src/esphome/components/sensor/filter.h:397:9:   required from here
**/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/type_traits:910:52: error: static assertion failed: template argument must be a complete class or an unbounded array**
  910 |       static_assert(std::__is_complete_or_unbounded(__type_identity<_Tp>{}),
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/type_traits:910:52: note: 'std::__is_complete_or_unbounded<__type_identity<array<float, 3> > >((std::__type_identity<std::array<float, 3> >(), std::__type_identity<std::array<float, 3> >()))' evaluates to false
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/alloc_traits.h:33,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/ext/alloc_traits.h:34,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_tree.h:67:
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_construct.h: In instantiation of 'void std::_Destroy(_ForwardIterator, _ForwardIterator) [with _ForwardIterator = array<float, 3>*]':
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/alloc_traits.h:850:15:   required from 'void std::_Destroy(_ForwardIterator, _ForwardIterator, allocator<_T2>&) [with _ForwardIterator = array<float, 3>*; _Tp = array<float, 3>]'
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_vector.h:730:15:   required from 'std::vector<_Tp, _Alloc>::~vector() [with _Tp = std::array<float, 3>; _Alloc = std::allocator<std::array<float, 3> >]'
src/esphome/components/sensor/filter.h:397:9:   required from here
**/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_construct.h:188:51: error: static assertion failed: value type is destructible**
  188 |       static_assert(is_destructible<_Value_type>::value,
      |                                                   ^~~~~
[... cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/tuple:1595:45: note: declaration of 'std::iterator_traits<std::array<float, 3>*>::value_type' {aka 'struct std::array<float, 3>'}
 1595 |   template<typename _Tp, size_t _Nm> struct array;
      |                                             ^~~~~
*** [.pioenvs/water-meter-esp/src/esphome/components/logger/logger.cpp.o] Error 1
========================== [FAILED] Took 4.08 seconds ==========================`

zibous, 2025-04-20:
Hi,

It looks like that you have not the version 3 wmbus-drivers ?

Check this section:

```
## ------------------------------------------------------------------
## EXTERNAL COMPONENTS
## https://esphome.io/components/external_components.html
## ------------------------------------------------------------------
external_components:
  # uses the version 3 from SzczepanLeon (github)
  # https://github.com/SzczepanLeon/esphome-components
  # https://github.com/SzczepanLeon/esphome-components/issues/189#issuecomment-2679873456
  - source: github://SzczepanLeon/esphome-components@version_3
    components: [ wmbus ]
```

> Clean Build Files

```
INFO ESPHome 2025.4.0
INFO Reading configuration /config/watermeter.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing water-meter-esp (board: az-delivery-devkit-v4; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- SPI @ 2.0.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7+sha.b8c6af4
|-- wmbus-drivers @ 0.0.0+20240828185136.sha.c856f7d
|-- noise-c @ 0.1.6
Compiling .pioenvs/water-meter-esp/src/main.cpp.o
Linking .pioenvs/water-meter-esp/firmware.elf
RAM:   [=         ]   8.1% (used 43276 bytes from 532480 bytes)
Flash: [======    ]  59.4% (used 1090409 bytes from 1835008 bytes)
Building .pioenvs/water-meter-esp/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
```

zibous, 2025-04-20:
You can try this version, but first check the  **WMBUS CC1101 --> ESP32 az-delivery-devkit-v4** connections:

```yaml
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## This version shows data from the defined watermeter and
## uses the watermeterID (HEX) from the secrets.yaml file.
##
## Usage:
##     + Using Homassistant API (discovery + devices)
##     + Using the last wM-Bus component
##     + Using wmbusmeters i a pipe (optional)
##
##     + REST:  curl -i http://water-meter-esp.local/text_sensor/watermeterdata
##              curl -i curl -i http://water-meter-esp.local/sensor/wasseruhr_anzeige
##
## ---------------------------------------------------------------------------
substitutions:

  ## device settings
  hostname: "water-meter-esp"
  friendly_name: "Wasseruhr"
  device_description: "Production 2024: Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon) my Watermeter"

  # The format of the name should be author_name.project_name.
  projectname: "Peter Siebler.Diehl IZAR RC 868"
  appversion: "Diehl IZAR RC 868 v2.1.2 (wm-esp32.yaml)"
  hardware: "ESP32_Devkit_V4"
  location: "Heizraum"

  ## logger settings
  log_level: "INFO" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "ERROR" # Loglevel for wmbus meters component
  log_baudrate: "0" # use 0 to disable serial UART port logging.

  # The built-in LED on the AZ-Delivery DevKit V4 is connected to GPIO 2.
  led_internal: "GPIO2"

## -----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4,ESP32 240MHz, 520KB RAM, 4MB Flash
## -----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Customize the frequency in which data is flushed to the flash.
preferences:
  flash_write_interval: 10min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${hostname}
  build_path: ./build/${hostname}
  comment: ${device_description}
  on_boot:
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - globals.set:
          id: send_millisecond
          value: !lambda "return millis();"
      - logger.log:
          level: WARN
          tag: "system"
[... 1201 lines cut ...]
  # (disable this if you do not need rest service)
  # you can get the sensor data with:
  #   curl -i http://water-meter-esp.local/text_sensor/watermeterdata
  - platform: template
    id: watermeterdata
    internal: true
    disabled_by_default: true
    lambda: |-
      char buf[128];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(watermonth).state,
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;
# e.o.f configuration esp32
```

adverserath, 2025-04-20:
That worked perfectly for me. 
It seemed when i had the source set to "- source: github://SzczepanLeon/esphome-components@version_3", and cleaned the build files, esphome is pulling a specific commit for wmbus. I didnt set up esphome through docker directly though, I use hassOS to pull in the addon

closed 2025-04-24 by zibous