zibous/ha-watermeter #61: C1 A not found in configuration T

llaumgui, 2025-04-22:
Hi,

I'm using your project with an ESP32 Dev Kit C V4 NodeMCU and a CC1101 868MHz module.

I can see frames appearing in the logs, but I believe they aren't being correctly decoded. The water meter reading remains at 0L, and both the CC1101 LQI and RSSI values are still showing as N/A, as shown in this screenshot:

[image]

This is my configuration for a itron watermeter:

```yaml
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## This version shows data from the defined watermeter and
## uses the watermeterID (HEX) from the secrets.yaml file.
## ---------------------------------------------------------------------------
substitutions:
  ## Device settings
  device_name_short: "watermeter-esp"
  device_description: "Easy Watermeter integration for Home Assistant"
  projectname: "Itron 868MHz.Watermeter"
  friendly_name: "Watermeter-esp" # device name
  appversion: "3.0.0"

  # Watermeter information
  watermeter_type: "itron"

  ## logger settings
  #log_level: "WARN" # not that logging need memory, so in production mode use "WARN"
  #log_wmbus: "ERROR" # Loglevel for wmbus meters component
  log_baudrate: "115200"
  log_level: "VERBOSE" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "VERBOSE" # Loglevel for wmbus meters component

  ## data service (optional, default: False)
  ## use only if you have a service host present
  service_enabled: "false"
  service_url: !secret watermeter_service_host
  service: "itron.watermeter"

## -----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4,ESP32 240MHz, 520KB RAM, 4MB Flash
## -----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Customize the frequency in which data is flushed to the flash.
preferences:
  flash_write_interval: 10min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false
  # additiona compile otions
  platformio_options:
    platform: espressif32
    board: az-delivery-devkit-v4
    ## larger stack size required with all registers enable_load_test
    build_flags:
      - "-DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768"
[... 1022 lines cut ...]
12:55:16	[V]	[crc:031]	
    calculated: 0x7906, read: 0x7906
12:55:16	[V]	[mbus:062]	
Frame: 0C4795260644240242378CC000 (13) [without CRC]
12:55:16	[D]	[wmbus:182]	
Meter ID [0x02244406] RSSI: -88 dBm LQI: 160 Frame: C1 A not found in configuration T: 0C4795260644240242378CC000 (13)
12:55:21	[V]	[rxLoop:167]	
Have 19 bytes from CC1101 Rx, RSSI: -84 dBm LQI: 139
12:55:21	[D]	[mbus:013]	
Processing C1 A frame
12:55:21	[V]	[mbus:017]	
Frame: 0C479526634324024237E54A8CC0007906 (17) [with CRC]
12:55:21	[V]	[mbus:095]	
Validating CRC for Block1
12:55:21	[V]	[crc:031]	
    calculated: 0xE54A, read: 0xE54A
12:55:21	[V]	[mbus:115]	
Validating CRC for Block2
12:55:21	[V]	[crc:031]	
    calculated: 0x7906, read: 0x7906
12:55:21	[V]	[mbus:062]	
Frame: 0C4795266343240242378CC000 (13) [without CRC]
12:55:30	[D]	[wmbus:377]	
cc1101 state message: Init, error code: 0
```

zibous, 2025-04-22:
Warning

If you enable MQTT and you do not use the [Native API Component](https://esphome.io/components/api), you must remove the api: configuration or set reboot_timeout: 0s, otherwise the ESP will reboot every 15 minutes because no client connected to the native API.

see: https://esphome.io/components/mqtt.html

try:
https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-esp32-simple.yaml

or
https://github.com/zibous/ha-watermeter/blob/master/esphome/wm-esp32-test.yaml

If you need API and MQTT, you could try version 4 see https://github.com/SzczepanLeon/esphome-components. However, I didn't have any success with that.

zibous, 2025-04-22:
It could also be that you are too far away from the water meter with the ESP. 
Basically, the APPLICATION works because you receive data from other water meters.

> Meter ID [0x02244385] RSSI: -80 dBm LQI: 137 Frame: C1 A not found in configuration T: 0C4795268543240242378CC000 (13)

llaumgui, 2025-04-22:
Hi,

Thx for your help !

> If you enable MQTT and you do not use the [Native API Component](https://esphome.io/components/api), you must remove the api: configuration or set reboot_timeout: 0s, otherwise the ESP will reboot every 15 minutes because no client connected to the native API.

OK, I'veremove it:

```yaml
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
## Hardware: ESP32 240MHz, 520KB RAM, 4MB Flash
##
## This version shows data from the defined watermeter and
## uses the watermeterID (HEX) from the secrets.yaml file.
## ---------------------------------------------------------------------------
substitutions:
  ## Device settings
  device_name_short: "watermeter-esp"
  device_description: "Easy Watermeter integration for Home Assistant"
  projectname: "Itron 868MHz.Watermeter"
  friendly_name: "Watermeter-esp" # device name
  appversion: "3.0.0"

  # Watermeter information
  watermeter_type: "itron"
  watermeter_mode: "T1C1"
  watermeter_frequency: "868.950"

  ## logger settings
  #log_level: "WARN" # not that logging need memory, so in production mode use "WARN"
  #log_wmbus: "ERROR" # Loglevel for wmbus meters component
  log_level: "VERBOSE" # not that logging need memory, so in production mode use "WARN"
  log_wmbus: "VERBOSE" # Loglevel for wmbus meters component
  log_baudrate: "115200" # use 0 to disable serial UART port logging.

  ## data service (optional, default: False)
  ## use only if you have a service host present
  service_enabled: "false"
  service_url: !secret watermeter_service_host
  service: "itron.watermeter"

## -----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4,ESP32 240MHz, 520KB RAM, 4MB Flash
## -----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Customize the frequency in which data is flushed to the flash.
preferences:
  flash_write_interval: 10min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false
  # additiona compile otions
  platformio_options:
    platform: espressif32
    board: az-delivery-devkit-v4
    ## larger stack size required with all registers enable_load_test
    build_flags:
      - "-DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768"
[... 932 lines cut ...]
    lambda: |-
      char buf[128];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(watermonth).state,
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;
```

> It could also be that you are too far away from the water meter with the ESP

It's the closest watermeter of my esp32. But, I live in a new neighborhood and all the water meters are the same model, is it normal the telegram error ?

> or you are using the wrong watermeterid ?

If I put my watermeterid to 0 I can see the watermeterid and this watermeterid is on mon physical water meter.

zibous, 2025-04-22:
Can you see data with watermeterId: 0x02244385 or 0x02244406

```yaml

 - platform: wmbus
    meter_id: 0x02244385
    type: "${watermeter_type}"
    # mode: "${watermeter_mode}"
    # add_prefix: true

```

zibous, 2025-04-23:
1. The Telegram data is too small, you may need a key because the data is encrypted.
    see: https://github.com/wmbusmeters/wmbusmeters/issues/346#issuecomment-931732495

   key: "???": This is the encryption key that is used to decode the data transmitted by the radio overlay ?

2. Or there is a problem with the wmbus driver, only Szczepan Leon could help you. see:  https://github.com/SzczepanLeon/esphome-components

I get no data for the analyze: https://wmbusmeters.org/analyze/0c47952606442402423737098cc0007906

llaumgui, 2025-04-23:
> The Telegram data is too small, you may need a key because the data is encrypted.
>see: https://github.com/wmbusmeters/wmbusmeters/issues/346#issuecomment-931732495
> key: "???": This is the encryption key that is used to decode the data transmitted by the radio overlay ?

Nice ! Like your link, I have an Itron Cybl5 !

Just question how to use WMBusmeter on ha-watermeter and enable a full telegtram debug like that:

```
Received telegram from: 13010647
manufacturer: (ITW) ITRON Water, France (0x2697)
type: Water meter (0x07)
ver: 0x02
Concerning meter: 20529976
manufacturer: (ITW) Unknown (0xa697)
type: Reserved for system devices (0x30)
ver: 0x00
device: rtl433[00000001]
rssi: 999 dBm
driver: unknown!
telegram=||4E449726470601130207727699522097A60030330030A5E704DFE72E18E4893BB5197252E4322EFC590269C9979B060CFE2DBA87EE138CF08D448F4E350372BB874F539DFCF04B0E79755447020000|+79
```

llaumgui, 2025-04-23:
Other question where to find my key ? On the qrcode ?

[image]

zibous, 2025-04-23:
> Other question where to find my key ?

Normally you get it from the water supplier.

> Just question how to use WMBusmeter on ha-watermeter and enable a full telegram debug like that...

If the ESP works with the C1101, then you get the whole telegram and you can analyze it with https://wmbusmeters.org

Testversion:

```yaml
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 
##
## ---------------------------------------------------------------------------
substitutions:
  device_name_short: "watermeter-esp"
  device_description: "Easy Watermeter integration for Home Assistant"
  projectname: "Itron 868MHz.Watermeter"
  friendly_name: "Watermeter-esp" # device name
  appversion: "3.0.0"

  # Watermeter information
  watermeter_type: "itron"
  watermeter_mode: "T1C1"
  watermeter_frequency: "868.950"
  watermeter_key: ???? add the key here

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  name_add_mac_suffix: false
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}
  on_boot:
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - logger.log:
          level: WARN
          tag: "system"
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
      - component.update: bootcounter
  on_shutdown:
    then:
      - logger.log:
          level: ERROR
          tag: "system"
          format: "BOOTMESSAGE:${device_name_short} is down!"

## ----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4
##          ESP32 240MHz, 520KB RAM, 4MB Flash
## ----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

## ---------------------------------------------------
## External components
## ---------------------------------------------------
external_components:
  # uses the version 3 from SzczepanLeon
  # https://github.com/SzczepanLeon/esphome-components
  # You can make ESPHome check the repository every time by setting this option to 0s
  - source: github://SzczepanLeon/esphome-components@version_3
    refresh: 0s
    components: [wmbus]

## ----------------------------------------------------------------
## GLOBALS VARIABLES
## ----------------------------------------------------------------
globals:
  - id: boot_counter
    type: int
    restore_value: yes
    initial_value: "0"

## ---------------------------------------------------
## WIFI Settings 3 wifis
## ---------------------------------------------------
wifi:
  networks:
    - ssid: !secret ssid_name
      password: !secret ssid_pswd
      priority: 0
  domain: !secret domain

## ---------------------------------------------------
## The captive portal component in ESPHome is a
## fallback mechanism for when connecting to the
## configured WiFi fails.
## ---------------------------------------------------
captive_portal:

## ---------------------------------------------------
## LOGGER COMPONENT
## ---------------------------------------------------
logger:
  id: appslogger
  level: VERBOSE
  baud_rate: 9600

## ---------------------------------------------------
## OTA COMPONENT
## ---------------------------------------------------
ota:
  password: !secret ota_pswd
  platform: esphome

## ---------------------------------------------------
## COMPONENT WEBSERVER
## ---------------------------------------------------
web_server:
  port: 80
  version: 3
  local: false
  enable_private_network_access: true
  include_internal: false
  ota: true
  log: true
  js_url: https://oi.esphome.io/v3/www.js
  # js_url: http://homassistant.local:8123/www/ephome/v3/www.js

## ---------------------------------------------------
## Home Assistant API COMPONENT
## ---------------------------------------------------
api:
  id: espapi_wmbus_esp32
  port: 6053
  reboot_timeout: 0s

## ---------------------------------------------------
## SNTP Time server
## Change this to your timezone and ntp server
## ---------------------------------------------------
time:
  - platform: sntp
    id: time_sntp
    timezone: Europe/Paris
    servers:
     - 0.pool.ntp.org
     - 1.pool.ntp.org
     - 2.pool.ntp.org

wmbus:
  mosi_pin: GPIO23 ## SI:   braun   3: MOSI Attached to Hardware SPI controller MOSI SPI Interface
  miso_pin: GPIO19 ## SO:   grün    5: MISO Attached to Hardware SPI controller MISO SPI Interface
  clk_pin: GPIO18 ## SCLK: violett 4: SCK  Attached to Hardware SPI controller CLK
  cs_pin: GPIO05 ## CSN:  orange  8: CSN  Attached to Hardware SPI controller
  gdo0_pin: GPIO16 ## GD00: gelb    7: RX Clock output. High Impedance !
  gdo2_pin: GPIO17 ## GD02: weiss   6: TX FIFO status signals. High Impedance !
  log_unknown: False

## ---------------------------------------------------
## SENSORS
## ---------------------------------------------------
sensor:
  - platform: wmbus
    meter_id: #secret watermeter_watermeterId
    type: "${watermeter_type}"
    key: "${watermeter_key}"

    lqi:
      id: wmbus_cc1101_lqi
      name: "Watermeter CC1101 LQI"
      entity_category: "diagnostic"
      unit_of_measurement: "lqi"
      state_class: "measurement"

    total_water_m3:
      name: "Watermeter display"
      id: "waterdisplay"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
```

it could also be that version 3 doesn't work, in which case you would have to test version 4.
see:   https://github.com/SzczepanLeon/esphome-components

llaumgui, 2025-04-23:
OK, I request key to my water suplier... Fingercross.

With v4, frame not working...

```log
[20:23:34]	[V] [3of6:083]: Decode 3 out of 6 failed..
[20:23:34]	[V] [mbus:043]: 055E6DDBF7B97ECFBA4EF9B74FB6FD1A56F9FC4139DF7536D80884AF28069F07CD691880F147A95685CDCF927DF89DDC892D39865604B3D66BBFEFBF5DFBBB6FBBCBB4DA8B22D04848FEF9875AB5E26E42AF57C9F96DBA99DE58D1635C2000961030994F7BE381DE6D1167FFF9FEFFCE5748FFB74516D4B1371960EF777C8F6C66E4C6F5AA7EF5A53EB15600812DE7A712F9DC502A2E76C071B6F5EFEFD67DFBB7D59C6089421812D0842FAD10F7B4BB7EF91BE68C611E (383) [RAW]
[20:23:34]	[V] [mbus:041]: Frame: C9668D06E04507563BD6AF331C3E1089E7E0142C53FA3A552E52309AF94FE0EC2A4D7440958886CE1083FFA0529F1D20E063509EC650F51E47B97FDFFFFFFFFF77FB27EC72A834F26BCCB166E6BB592B1929BAF70F42B21651BD6DF950DE9D50503C1A68595E3C3B231B7ECF8FF64DB6DDFBFCC2C22F93C7B0C992DBB41E42EFE101444E7F157866F6684D05CB6DF6A3C8074C565B9A8F7E48A3A7FAF5DBB7CD7E3A30B50849518947DF731428563CDDAE0EA42E32CFA823E02F19C75338D474F6A297999430B814 [RAW]
[20:23:33]	[D] [mbus:035]: Received T1 A frame
[20:23:33]	[V] [rxLoop:167]: Have 383 bytes from CC1101 Rx, RSSI: -98 dBm LQI: 21
```

zibous, 2025-04-24:
> With v4, frame not working...

This looks like that the wmbus component do not work.
Szczepan Leon could help you. see: https://github.com/SzczepanLeon/esphome-components

closed 2025-05-09 by github-actions[bot]