zibous/ha-watermeter #62: ESP32 & CC1101 : no data

Anoxym, 2025-04-26:
Hi,

I successfully compiled the program on ESP32 and communication with CC1101 works. I first compiled the test program and get the watermeter address (0x21202341) and then, complied this program : https://github.com/zibous/ha-watermeter/issues/60#issuecomment-2817070527 (I am on ESPHome via Home assistant)

But I didn't get any data, here is what I have in log : 

```
[00:17:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:18:06][D][mbus:034]: Processing T1 A frame
[00:18:06][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -66 dBm LQI: 129 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:18:07][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:18:07][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
[00:18:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:18:34][D][mbus:034]: Processing T1 A frame
[00:19:23][D][mbus:034]: Processing T1 A frame
[00:19:24][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -67 dBm LQI: 140 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:19:24][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:19:24][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
[00:19:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:19:36][D][mbus:034]: Processing T1 A frame
[00:19:36][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -68 dBm LQI: 130 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:19:37][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:19:37][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
[00:19:49][D][mbus:034]: Processing T1 A frame
[00:19:49][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -65 dBm LQI: 128 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:19:50][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:19:50][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
[00:20:03][D][mbus:034]: Processing T1 A frame
[00:20:03][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -65 dBm LQI: 133 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:20:03][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:20:03][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
[00:20:07][D][sntp:067]: Synchronized time: 2025-04-27 00:20:07
[00:20:07][W][system:471]: Synchronized sntp clock
[00:20:07][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:20:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:20:42][D][mbus:034]: Processing T1 A frame
[00:20:54][D][mbus:034]: Processing T1 A frame
[00:20:54][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -63 dBm LQI: 132 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:20:54][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:20:54][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
[00:21:06][D][mbus:034]: Processing T1 A frame
[00:21:06][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -68 dBm LQI: 130 Frame: T1 A T: 1E44A511412320217B077A8F0010054B59096DB937F54B8E362B048AABED11 (31)
[00:21:07][W][wmbus:923]: WARNING Alarm message: Fehler, Überschwemung, error code: 130
[00:21:07][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
[00:21:19][D][mbus:034]: Processing T1 A frame
[00:21:30][D][wmbus:549]: cc1101 state message: Init, error code: 0
[00:21:32][D][mbus:034]: Processing T1 A frame
[00:21:32][D][wmbus:097]: Using driver 'izar' for ID [0x21202341] RSSI: -67 dBm LQI: 130 Frame: T1 A T: 3E44A511412320217B077A8F003005BB15441123859046292A3987BBF052222ACF4FF3B5BEB88F0C490633D033F7BBB4D2513D71D38ABA6E7472DB54CC9F84 (63)
[00:21:32][W][wmbus:923]: WARNING Alarm message: Fehler, Wasser Überlauf, Überschwemung, error code: 194
[00:21:33][D][wmbus:943]: Alarm message: \xec$\xfb?, error code: 0
```

Can someone help me?
Thanks

zibous, 2025-04-27:
Hi,
It looks like that the wmbusmeter component do not work:

Analyze:[3E44A511412320217B077A8F003005BB15441123859046292A3987BBF052222ACF4FF3B5BEB88F0C490633D033F7BBB4D2513D71D38ABA6E7472DB54CC9F84](https://wmbusmeters.org/analyze/3E44A511412320217B077A8F003005BB15441123859046292A3987BBF052222ACF4FF3B5BEB88F0C490633D033F7BBB4D2513D71D38ABA6E7472DB54CC9F84)

Maybe Szczepan Leon could help you. see: https://github.com/SzczepanLeon/esphome-components

Anoxym, 2025-04-27:
Thanks for the answer. Or maybe I need to choose dme_07 instead of izar as driver ?

zibous, 2025-04-28:
> Or maybe I need to choose dme_07 instead of izar as driver ?

Maybe, if i analyse the telelgram with driver dme_07 i can see total_m3, but no more data.

[image]

closed 2025-05-13 by github-actions[bot]