zibous/ha-watermeter #63: Usage with LilyGo T-Internet-POE possible?

Eagle3386, 2025-05-01:
Hi there,

I'm about to get a [Diehl Hydrus 2.0](https://www.diehl.com/metering/en/products-solutions/products-services/water-metering/hydrus-20-de/) water meter which offers wmBus, including 2 AES keys (1 general key for the water supplier, 1 individual for each household).

For my electricity meter, I'm using a [LilyGo T-Internet-POE](https://lilygo.cc/products/t-internet-poe) for powering an IR reading head which - thanks to ESP Home - reads the meter periodically & passes the data along.

The board's pin layout looks like this:
[image]

Bluetooth & WiFi are deliberately turned off (I prefer cables & especially PoE-based sensors wherever possible), so the question is:

While both meters should be read every minute (alternatingly due to possible usage constraints), can I connect the "typical" CC1101 for the actual wireless connection to the remaining pins?

How many does the CC1101 actually need? Because pins IO34/35/36/39 can only be used as inputs, hence - depending on the CC1101 - I've got only 4 configurable pins left, if there's no use for the aforementioned input-only pins..

SzczepanLeon, 2025-05-01:
Maybe this hint will help you:
GDOx pins are output from CC1101 side, so on ESP they are input.

Eagle3386, 2025-05-01:
Yes, that helped - thanks a lot!

Additionally, for future reference, here's an example of an Arduino pinout which further clarifies that there are 7 pins needed in total:
[image]

closed 2025-05-01 by Eagle3386