zibous/ha-watermeter #65: wmbus break since esphome 2025.7.1 ?

cyrinux, 2025-07-18:
Hi 😃 
First thanks for this cool project ! I use it for the moment and no issue, but, since yesterday, after I update all, esphome in  2025.7.1,  looks like I receive no data anymore, wmbus in DEBUG with my watercounter counter id or 0 see nothing, no logs from watermeters.

Any idea how to fix this ?

SzczepanLeon, 2025-07-18:
Stay on 2025.6.x

cyrinux, 2025-07-19:
Ha, got it, https://github.com/SzczepanLeon/esphome-components/tree/main dont have yet support for CC1101, and version_4 branch doesnt have support for 2025.7.0.

cyrinux, 2025-07-19:
https://github.com/SzczepanLeon/esphome-components/issues/239

SzczepanLeon, 2025-07-26:
Everything about YAML is on my github page.

v1-v4 are for arduino
v5 is for esp-idf

v1-v4 are for CC1101 only
v5 is for SX only

v1-v4 are based on Satan CC1101 radio lib
v5 is based on custom code

Because this is unofficial ESPHome components sometimes it needs to align/update when new ESPHome is released. And I don't have time to maintain all branches.

You can always stay on ESPHome 2025.6.3

Onepamopa, 2025-07-26:
Why not make it "official" ?

SzczepanLeon, 2025-07-26:
Maybe in future. There is couple thinks that are not in line with ESPHome pholosophy.

mrlordziemniak, 2025-07-27:
> Everything about YAML is on my github page.
>
> v1-v4 are for arduino v5 is for esp-idf
>
> v1-v4 are for CC1101 only v5 is for SX only
>
> v1-v4 are based on Satan CC1101 radio lib v5 is based on custom [...]

will this radio work? https://pl.aliexpress.com/item/1005005291351005.html?gatewayAdapt=glo2pol

Onepamopa, 2025-07-27:
Or this: https://pl.aliexpress.com/item/1005006860809926.html  ??

SzczepanLeon, 2025-07-27:
It should be OK.